openrazer/openrazer #5: New Daemon Testers

terrycain, 2016-07-10:
Sign up here for beta testers of #NewDaemon :wink:

geordieF, 2016-07-11:
Sure, you know my gear.

terrycain, 2016-07-11:
Well good news, there shouldn't be any manual compiling at all ;). What distibution and version you running? Basically the new daemon takes in a block of code (about 4ish lines) which defines a device, so I need to add a section for the BW X Chroma but hopefully should have a test and some instructions out by the end of the week.

geordieF, 2016-07-12:
Ubuntu 16.04.
This is my current 'about device' screen.
[image]

lah7, 2016-07-13:
* **Device:** BlackWidow Chroma
* **ID:** 1532:0203
* **Works?** Yep, works great... except the service needs restarting after sleeping the system. :pensive: 
* **Distro:** Ubuntu MATE 16.04
 * 4.4.0-28-generic x86_64

Hurray for packages! I'll be swapping between them often so I can still use the old daemon service and work on [Polychromatic](https://github.com/lah7/polychromatic). :wink:

GabMus, 2016-07-13:
@lah7 You know, [razerCommander](https://github.com/gabmus/razercommander) works independently from the daemon, so you could try that :wink: 

Also, I need testers :confused: 

I only have a BlackWidow Ultimate 2016, it has no RGB so I couldn't verify that my code works for RGB keyboards.

terrycain, 2016-07-13:
We might have razer kraken support soon :smile::smile::smile:

@geordieF output of `ls /dev/input/by-id` need some more info before I can add the chromaX to this.

---

Onto the task at hand. Eventually I'll put these in a repo (not launchpad as it has pissed me off one too many times + I cba to swap between bzr and git or try and wedge git into launchpad).

**[Here](https://drive.google.com/open?id=0BydYgD2LzDUCeHR4SnJFN25BeFU)** are some beautiful .deb file. So beautiful that `lintian` barely complains about them. `razer-doc` is empty and useless so you can leave that one there. Also I cba to increment version numbers at the moment :wink: so if I say there's new debs just take it at that.

### Method:
Ok the fun bit :wink:. Remove any razer crap from your system, that means all driver stuff, kernel modules, daemon, daemon libs etc... **AND ESPECIALLY!! the udev rules!!!!**, check in `/etc/udev` and `/lib/udev`

To install normally you'd use a repo so `apt` would pull down any dependencies, we are too lazy for that so get the debs in a folder. Run as sudo/root `dpkg -i *.deb`. It WILL complain, probably about packages it depends on. Now run as sudo/root `apt install -f` that should fix things by downloading the packages it depended on. If the `apt` command removed the razer stuff then that not good lol.

Now reboot. I know rebooting is a pain but its kernel stuff so its always best to. Theoretically once you've rebooted all of your supported razer stuff should be automagically binded to the driver. Refer to [this](https://github.com/pez2001/razer_chroma_drivers/wiki/Setting-up-the-keyboard-driver) page if things arn't binded to the driver / if you want to check that.

Thats the hard part. (plus you can now use apt to remove razer stuff with `apt purge razer*`)

Also ensure you user is a member of the **plugdev** group. If it doesn't exist then create it. Might look as feasibly automating that on install.

#### Testing:
I've purposly not set the new daemon to autostart as I'm looking into the best way to do that. For now just live with it, plus I want it in a terminal so you can easily get text from it if it dies :wink: oh and it will die ha.

This daemon is new. Its written in python so no more compiling, plus theoretical 32bit support. It is multithreadded so theres already 100x more ways for things to go wrong. If you see it hogging a core on your CPU then well, I'll discuss that another time.

 To run the daemon. Run in a terminal as your user `/usr/sbin/razer-service --verbose --foreground` It wont go away and it will spam stuff.

Once were all here I'll go on to how to manually prod the daemon, but technically if your here and not running some dodgy desktop envionment (*cough* @geordieF kde *cough*) If you lock/let screensaver completely appear your devices might just turn off and then reappear when its gone.

Sound good?

GabMus, 2016-07-13:
> there. Also I cba to increment version numbers at the moment 😉 so if I
> Ok the fun bit 😉. Remove any razer crap from your system, that means all
> terminal so you can easily get text from it if it dies 😉 oh and it will

I'm running Arch, could you provide some distro-agnostic install
instructions? Of course I mean only about the deb part, the rest seems
pretty good.

terrycain, 2016-07-13:
Ugh you one of those people :wink: .

`make install` should work, but you'll need to install DKMS and any other dependencies listed in the debs :wink:) (open them up ha, easiest thing would be to extract the at `/` then its all "installed"). Once `make install`ed. Make sure the driver is build by DKMS, (that's out of the scope here) (the debs would do it automagically lol).

GabMus, 2016-07-13:
> dependencies listed in the debs 😉) (open them up ha, easiest thing would

Ugh. That's not very helpful. I'd try to make a PKGBUILD for Arch myself,
but I have no idea how dkms really works. I mean I know what it does, but
not how to make a package that uses it. If you know a good reference that'd
be great.

terrycain, 2016-07-13:
Tbf I just read the scripts which other dkms deb files ran. I know its not too helpful. Make install should do most of it then look at the first part of this script `install_files/DEBIAN_ubuntu/postinst`. There should be scripts with the same or similar names on your system. As `make install` will put files in the right place for dpkg and then the deb would call something to register it, all you'd need to do is register it and all should be good. (plus registering it also triggers the compliation). Does that help?

terrycain, 2016-07-13:
Worst case `make driver_install` will install the driver straight into the driver dir but will die if you change kernel as they wont be there.

terrycain, 2016-07-13:
Actually make install will not use dkms, and will just install the driver straight.

GabMus, 2016-07-13:
Yup, thanks. I'll spend some time on this when I'm done with finals (Friday
HOPEFULLY).

PS: I'm glad to be collaborating to this project, even the slightest. You
guys are cool 😉

terrycain, 2016-07-13:
Ha that's how I felt when I first joined the repo (though there was one other person on the repo then).
Finals -- University, School, College?? I just finished up at Uni, moved house and now have finally got set up again.

GabMus, 2016-07-13:
University. Not really "finals" more like the last ones of the summer
session. I'm studying computer science.

That said, hopefully Friday I'll be fee to tinker with PKGBUILDs and the
new daemon, and probably breaking my Arch system. If everything goes
smoothly I can start a branch of razerCommander for the new daemon (if you
don't plan to change the API ofc).

GabMus, 2016-07-13:
While we're at it, don't you think it would be a good idea to make a
telegram group or something similar to coordinate development? It'd also be
easier to send log files back and forth

terrycain, 2016-07-13:
Not your last year then? and +1 for #CompSci

And the interface on DBus wont change but the python "client" library which you can use if you don't want to DBus will probably change along until I'm happy with it. Don't worry about this until its running ha.

And if you've got any ideas for chat thingy then I'm all ears. Just not slack - another thing I'm not a fan off.

GabMus, 2016-07-13:
No, I'm finishing my second year :)
Yeah, if the python lib is changing I'm just gonna have a good look at it
for now.
As for the chat, if you don't know telegram, it's basically "whatsapp meets
irc" in a (partially) open source fashion. It's got desktop and web
clients, as well as mobile. You sign up with your phone number but you
don't have to share it, you can use usernames. Have a look at it, if you're
ok with it we can make a nice group.

terrycain, 2016-07-13:
Ok that telegram thing looks alright, thoses of you that wish to use it make an account and drop your username here and ill eventually work out how to use this thing. @GabMus  add me "terrycain"

geordieF, 2016-07-19:
Telegram looks good, its just we have to use phone and that wouldn't be great to put on aa github issue.

GabMus, 2016-07-19:
You can use the desktop app or the web app

geordieF, 2016-07-19:
@GabMus Added ya.

apol1, 2016-08-02:
What is the status on this driver? I feel a little lost sorry

GabMus, 2016-08-02:
Right now we're fixing some bugs and soon we'll start the work on the python library. As now the daemon seems to be working good. The development is currently going on in terrycain's fork of this repo (github.com/terrycain/razer_drivers)

terrycain, 2016-08-03:
Thanks @gabmus yeah there's an issue on my fork with instructions on how to test out the new stuff

claell, 2016-08-03:
Actually this seems to be the mentioned issue :)

terrycain, 2016-08-03:
Oh yeah. Didnt scroll up ha

apol1, 2016-08-07:
Sounds great, I bought a Blackwidow Chroma X keyboard and would like to be a tester for you! Soon I'll be purchasing a Mamba and a firefly to go along with my build. Do you have an email or should I stay on this thread? Email is the best way to contact me.

Im running Ubuntu LTS 16.04

terrycain, 2016-08-09:
@apol1 Join the telegram group would be the best way to stay up to date otherwise when I'll keep this issue up to date when I start working

apol1, 2016-08-09:
thx but im not much of a chatter, I like to stay with email

terrycain, 2016-08-09:
no problem

closed 2016-09-03 by terrycain