openrazer/openrazer #16: Fedora 24 Installation Error

JKenyon1, 2016-08-11:
Unable to install on Fedora 24 getting the following:

make -C /lib/modules/4.5.5-300.fc24.x86_64/build SUBDIRS=/home/jasonk/pez2001-razer_chroma_drivers-88eaeed/driver modules > /dev/null 2>&1
Makefile:19: recipe for target 'driver' failed
make: *** [driver] Error 2

terrycain, 2016-08-11:
What does `make driver_verbose` output

JKenyon1, 2016-08-12:
Hey thanks for the reply.  I'm currently not at work which is where it is, I will be going back in sat night. But I was using the other repo I commented on, I will try this one also and provide u the information u requested

JKenyon1, 2016-08-12:
Okay started up a VM, with the same installation and everything.  As it would have been a new installation, heres the information you requested. 

http://pastebin.com/zsD4ix6M

I'm thinking of because this would be a new installation just jumping to linux mint, and trying my luck with that.

terrycain, 2016-08-12:
Did you do a dnf update? I remember having issues with fedora as it installs headers for the latest kernel version installed not the on currently running kernel. Try a reboot of the vm and run those commands again.

As you can see you have headers for 4.6.5 but it's looking in 4.5.5 as that's the one running.

JKenyon1, 2016-08-12:
Indeed you were correct, getting another error will post that also soon.  Linux mint it appears all went well, but note able to run the service:

Failed to start razer_bcd.service: Unit razer_bcd.service not found.

latest linux mint install; cinnamon as the desktop env

terrycain, 2016-08-12:
Ahh different service name (different service entirely :wink:). Still a bit experimental. Should have some app support for it soon, will update read me/wiki sometime this weekend

terrycain, 2016-08-29:
New service is out :smile:

closed 2016-08-29 by terrycain