openrazer/openrazer #20: Demon support for Razer Tartarus Chroma

Denlior, 2016-08-24:
Demon support for Razer Tartarus Chroma would be great.
Especially the keycode customisation feature. 
Ready and willing to test it, byt need instructions how to do it.

Now it produces these keycodes:
01 23
02 24
03 25
04 26
05 27
06 66
07 38
08 39
09 40
10 41
11 50
12 52
13 58
14 54
15 55
Mode Switch 64
Up (hat switch) 111
Down (hat switch) 116
Left (hat switch) 113
Right (hat switch) 114
Bottom Thumb Button (below hat switch) 65

terrycain, 2016-08-24:
:smile: Am currently battling launchturd for our v1.0.0 release. Once I've won I'll get to work

terrycain, 2016-08-28:
Have made a start, What distro you using? this is probably gonna involve alot of testing but shouldn't be toooo bad.

In the mean time have a look over DBus and get a rough idea of what it does / how it works. Then acquire the program `d-feet`

terrycain, 2016-08-28:
Also the output of `ls -lah /dev/input/by-id/` and `ls -lah /dev/input`

Denlior, 2016-08-29:
I'm using xubuntu.
OK, now I have d-feet and vague understanding about D-Bus.

`root@OnyxBlue:/home/danid# ls -lah /dev/input/by-id/
total 0
drwxr-xr-x 2 root root 220 авг 29 22:36 .
drwxr-xr-x 4 root root 480 авг 29 22:36 ..
lrwxrwxrwx 1 root root   9 авг 29 20:35 usb-Logitech_Gaming_Mouse_G300-event-mouse -> ../event7
lrwxrwxrwx 1 root root   9 авг 29 20:35 usb-Logitech_Gaming_Mouse_G300-if01-event-kbd -> ../event8
lrwxrwxrwx 1 root root   9 авг 29 20:35 usb-Logitech_Gaming_Mouse_G300-mouse -> ../mouse1
lrwxrwxrwx 1 root root   9 авг 29 22:36 usb-Razer_Razer_Tartarus_Chroma-event-kbd -> ../event4
lrwxrwxrwx 1 root root   9 авг 29 20:35 usb-Razer_Razer_Tartarus_Chroma-if01-event-kbd -> ../event5
lrwxrwxrwx 1 root root   9 авг 29 20:35 usb-Razer_Razer_Tartarus_Chroma-if02-event-mouse -> ../event6
lrwxrwxrwx 1 root root   9 авг 29 20:35 usb-Razer_Razer_Tartarus_Chroma-if02-mouse -> ../mouse0
lrwxrwxrwx 1 root root   9 авг 29 20:35 usb-SEMICO_USB_Keyboard-event-kbd -> ../event2
lrwxrwxrwx 1 root root   9 авг 29 20:35 usb-SEMICO_USB_Keyboard-if01-event-kbd -> ../event3
root@OnyxBlue:/home/danid# ls -lah /dev/input
total 0
drwxr-xr-x  4 root root       480 авг 29 22:36 .
drwxr-xr-x 20 root root      4,6K авг 29 22:36 ..
drwxr-xr-x  2 root root       220 авг 29 22:36 by-id
drwxr-xr-x  2 root root       220 авг 29 22:36 by-path
crw-rw----  1 root plugdev 13, 64 авг 29 20:35 event0
crw-rw----  1 root plugdev 13, 65 авг 29 20:35 event1
crw-rw----  1 root plugdev 13, 74 авг 29 20:35 event10
crw-rw----  1 root plugdev 13, 75 авг 29 20:35 event11
crw-rw----  1 root plugdev 13, 76 авг 29 20:35 event12
crw-rw----  1 root plugdev 13, 77 авг 29 20:35 event13
crw-rw----  1 root plugdev 13, 78 авг 29 20:35 event14
crw-rw----  1 root plugdev 13, 79 авг 29 20:35 event15
crw-rw----  1 root plugdev 13, 80 авг 29 20:35 event16
crw-rw----  1 root plugdev 13, 66 авг 29 20:35 event2
crw-rw----  1 root plugdev 13, 67 авг 29 20:35 event3
crw-rw----  1 root plugdev 13, 68 авг 29 22:36 event4
crw-rw----  1 root plugdev 13, 69 авг 29 20:35 event5
crw-rw----  1 root plugdev 13, 70 авг 29 20:35 event6
crw-rw----  1 root plugdev 13, 71 авг 29 20:35 event7
crw-rw----  1 root plugdev 13, 72 авг 29 20:35 event8
crw-rw----  1 root plugdev 13, 73 авг 29 20:35 event9
crw-rw----  1 root input   13, 63 авг 29 20:35 mice
crw-rw----  1 root input   13, 32 авг 29 20:35 mouse0
crw-rw----  1 root input   13, 33 авг 29 20:35 mouse1
`

terrycain, 2016-08-30:
:smile: Ta. I have some stuff for you to try out, will give you some .deb's and some instructions when I finish work.

terrycain, 2016-08-30:
Ok here are some debs, https://drive.google.com/open?id=0BydYgD2LzDUCeHR4SnJFN25BeFU
Remove any razer things already installed. Then install all of the debs bar the razer-doc one as that's empty atm.

Then reboot. The Tartarus "should" hopefully be automatically bound to the razerkbd driver then. If we get to that point we'll take it from there. Tbf if it does, when you log in the daemon "should" automatically start (but it might throw a fit and complain :wink:).

Denlior, 2016-08-31:
Well, all done.
Tartarus binds to the driver, but with a strange result - first few keys pressed work ok, but later it either sticks perpetualy pressed on some key, or goes silent.
Deamon loads - I see a razer-deamon process, but later it dies.

terrycain, 2016-08-31:
Is there a log in ~/.razer-service

Denlior, 2016-09-01:
Yes, there is.
[razer.log.tar.gz](https://github.com/terrycain/razer-drivers/files/448872/razer.log.tar.gz)

terrycain, 2016-09-01:
Well good news is the daemon recognises the Tartarus. It'll probably take a bit of trial and error.

If all works well the tartarus should look like it doesnt work as i've told the daemon to hide all of its keypresses from everything else. So when you press keys it doesnt type anything as it should run macros.
Though I've just got it to output to log file for now until is working. Will get some new debs tonight,

terrycain, 2016-09-01:
Try [this](https://drive.google.com/open?id=0BydYgD2LzDUCeHR4SnJFN25BeFU). Not much will happen. Can you run `tail -f ~/.razer-service/logs/razer.log`, it will constantly keep an eye on the output.

I have a feeling the numbers it return wont match up with whats at the top of the issue, can you recheck the mapping. It'll either say Got Key: somekey or it will say could not convert key id someid.

Denlior, 2016-09-02:
Ok, now it is silent on keypress.
In log it now goes:

Key 06: "2016-09-02 21:34:31 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't store in bucket
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer_daemon/misc/key_event_management.py", line 688, in key_action
    self._stats[storage_bucket][key_name] += 1
KeyError: '13'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer_daemon/misc/key_event_management.py", line 695, in key_action
    self._stats[storage_bucket][key_name] += 1
KeyError: '13'
2016-09-02 21:34:31 | razer.device0.keymanager       | INFO     | Got Key: 13
"

Key 01: "2016-09-02 21:47:22 | razer.device0.keymanager       | ERROR    | Could not convert key id 15 to a corrosponding name"
Key 02: same error with id 16
Key 03: same error with id 17
Key 04: same error with id 18
Key 05: same error with id 19

Key 07: same error with id 30
Key 08: same error with id 31
Key 09: same error with id 32
Key 10: same error with id 33

Key 11: same error with id 42
Key 12: same error with id 44
Key 13: same error with id 45
Key 14: same error with id 46
Key 15: same error with id 47

Mode Key: same error with id 56
Bottom Thumb Button: same error with id 57

Hat switch up: same error with id 103
Hat switch left: same error with id 105
Hat switch right: same error with id 106
Hat switch down: same error with id 108

terrycain, 2016-09-02:
Yeah I didnt bother fixing the errors as once i add the id's to the file they "should" go away as the device has a finite set of key IDs it can emit.

Once again [here](https://drive.google.com/open?id=0BydYgD2LzDUCeHR4SnJFN25BeFU) is another daemon deb. In theory if I've got it right the errors should go away (key 6 is questionable as its not on the list). So test the keys again and all should work.

Aside from that. If you run the daemon:
1. whilst its running open d-feet
2. go to the session bus tab
3. find and click on the entry `org.razer`
4. it should should show some object paths like `/org/razer` and `/org/razer/device/SERIALNUMBER`
5. expand the entry for your tartarus
6. there should be some options for you to mess around with like brightness / effects / profile led colours (ignore macro bit for now).

Denlior, 2016-09-02:
Well now it goes "INFO Got Key" with proper numbers for keys 1-5 and 7-9 and "ERROR    | Could not convert key id 58 to a corrosponding name" kor key 6
For all other keys it is now like this:
"2016-09-02 23:18:15 | razer.device0.keymanager       | ERROR    | Got key error. Couldn't store in bucket
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer_daemon/misc/key_event_management.py", line 688, in key_action
    self._stats[storage_bucket][key_name] += 1
KeyError: 'MODE_SWITCH'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer_daemon/misc/key_event_management.py", line 695, in key_action
    self._stats[storage_bucket][key_name] += 1
KeyError: 'MODE_SWITCH'
2016-09-02 23:18:15 | razer.device0.keymanager       | INFO     | Got Key: MODE_SWITCH"
with proper names for all keys and switches.

D-feet works ok with colour and brightness settings, but IMHO setting them from terminal (bar superuser privileges) is much more convenient :)

terrycain, 2016-09-02:
Yeah the daemon will just sit between the kernel driver and the user and performs all the high level logic. There is a python api forming which will interface with it and once thats good ill probaby make some script examples for people.

terrycain, 2016-09-02:
If you can convince @lah7 or @gabmus, they might add it to their app thingy. And yeah I forgot about that other error.

terrycain, 2016-09-02:
[Here](https://drive.google.com/open?id=0BydYgD2LzDUCeHR4SnJFN25BeFU) is a daemon that should eliminate all errors. Mind testing all dbus methods to change the rgb profile led's just to test they work.

Have broken the mamba driver so am going back to fix that. Then ill crack on with whats left.

Denlior, 2016-09-03:
Yep, now all the keys interact ok with the deamon.
With profile leds I was totally unsuccessful, for there is no corresponding method visible in D-feet.

terrycain, 2016-09-03:
Could you send over the log

Denlior, 2016-09-03:
Here it is 
[razer.log.tar.gz](https://github.com/terrycain/razer-drivers/files/453490/razer.log.tar.gz)

terrycain, 2016-09-03:
Sorry my bad, added the methods to the daemon just didn't add them to the Tartarus

[Here](https://drive.google.com/open?id=0BydYgD2LzDUCeHR4SnJFN25BeFU) should do it.

Denlior, 2016-09-03:
OK, now I can change profile leds as well.

terrycain, 2016-09-03:
:smile: Great. Ok all that's left is getting macros working then a cli type interface is needed. You want a cli or would an app work better?

Denlior, 2016-09-03:
I'm ok with command line, so we can go on without apps.

terrycain, 2016-09-03:
Ok, in terms of macros, what sorta features you need?

Denlior, 2016-09-03:
Well, primarily I need not macroses, but custom keycodes. Something like key 11 > 65, key 07 > 84, key 08 > 71 and key 09 > 67. (I'm a geneticist, so I need  to type in DNA/RNA alphabets a lot)

Denlior, 2016-09-03:
Complex commands or even simple scripts would be nice too, but mostly just to learn how to, for further use.

terrycain, 2016-09-03:
Ok well rebinding keys would simply be a macro of one key :wink:. The daemon already lets you do single or even a string of keys. It also lets you bind a key to open a webpage so extending that to run arbitrary scripts would be easy.

The main bit of work would be exposing this functionality in a nice easy to use manner. But it needs to be done anyway so will start to work on that, it'll even benefit the gamers too :smile:.

Denlior, 2016-09-04:
And one more thing that would be really useful to me. A mode when with, say, mode button pressed other keypresses would be collected and summed up to a single key (or other action maybe) according to an algorithm (written by end-user in python or whatever) on mode key release. (in my case to type [IUPAC ambiguity codes](http://droog.gs.washington.edu/parc/images/iupac.html))

Denlior, 2016-09-04:
By the way, what is the language of macro actions of the deamon? I.E. is it possible to implement actions from biopython, bioperl e.t.c so I (and maybe other fellow scientists) would be able to bind keys to specific scripted actions with, say, selected string of text?

terrycain, 2016-09-04:
Ok last night I added the ability to:
* Bind one or more keystrokes to a macro key.
* Open URLs via a macro key
* Run arbitrary scripts via a macro key.

Have made a simple script to manage things via the command line. Is a work in progress at the moment. Will get you to try it out soonish.

---

> And one more thing that would be really useful to me. A mode when with, say, mode button pressed other keypresses would be collected and summed up to a single key (or other action maybe) according [...]

I can get the mode button to trigger "on-the-fly" macro recording though the keys it would see would be limited to those of the Tartarus as I think its a bit out of scope to start snooping keypresses of random devices (unless you have a razer keyboard as it'll could be made to see those keys). I don't get what you mean by "according to an algorithm" bit, can you elaborate.

> By the way, what is the language of macro actions of the deamon? I.E. is it possible to implement actions from biopython, bioperl e.t.c so I (and maybe other fellow scientists) would be able to bind keys to specific scripted actions with, say, selected string of text?

Macro actions are just a simple class (KeyPress, URL, Script) which perform actions. Arbitary scripts can be bound to keys so anything you could do on the command line could be bound to a key. Would that do or are you after something specific?

Anything I missed?

Denlior, 2016-09-05:
Sorry, my bad. What I want in general is to type in DNA/RNA alphabets.
There are four letters used mostly - A, T/U, G and C corresponding to four
nucleotide types in DNA and RNA genetic code. It's bloody inconvenient to
type them on a normal keyboard. There are also single letter codes for
variable or ill determined position. I would like to press mode key and
then keys corresponding to T and C to get Y (meaning it can be T or C in
this position), or all four to get N (meaning there is some nucleotide
there, we just do not know what it is) and so on. Like a lot of hotkey
combos but bound to this specific controller. No snooping on other devices.
Even Razer ones - if I had a Razer keyboard I would not like it to give me
Y on Alt+T+C.

Yes, command line is good. I believe anything I would like done could be
done that way.

terrycain, 2016-09-05:
Ok so if I understand correctly you want something like
Key 1 => A
Key 2 => T
Key 3 => G
Key 4 => C

And then something like
Mode Key + Key 2 (T) + Key 4 (C) => Y
Mode Key + Key 1 (A) + Key 2 (T) + Key 3 (G) + Key 4 (C) => N

Am I on the right track?

Denlior, 2016-09-05:
Yes, that's it.

terrycain, 2016-09-05:
Ok cool. I'll work on that tonight. Shouldn't take too long to do.

terrycain, 2016-09-05:
Am creating some test stuff so I can simulate a Tartarus (and all the other supported devices) so I can test the daemon macro things.

GabMus, 2016-09-06:
@Denlior Hey there I just released an implementation of macro assignation for the Tartarus in my frontend for this driver, [razerCommander](https://github.com/GabMus/razerCommander) and it'd be really nice if you could test it. Just make sure to get both the driver and razercommander from master, and not by using debs/ppas.

terrycain, 2016-09-06:
[These](https://drive.google.com/open?id=0BydYgD2LzDUCeHR4SnJFN25BeFU) debs are the latest :wink:.

Ok I have tested this my side (as created a simple device simulator), but it might totally fail :wink:.
Run the daemon. Tail -f the log as that's the output we want. Go on d-feet and you want to set `razer.device.macro.setModeModifier` to `True` (case sensitive). Now the Mode Switch should allow you to perform key combos, 

So once `setModeModifier` is true, typing keys should result in some log messages. Then hold down mode switch and push a few keys and then release mode switch, hopefully some log like 1+2+3 if you pressed 1 and 2 and 3

Make sense?

Denlior, 2016-09-06:
Ok, it works, as I presume it should - I get a log of mode key pressed/unpressed and other keys summed up to a macro string.

terrycain, 2016-09-06:
Yes please

Denlior, 2016-09-06:
@GabMus Will be glad to test it. Yet I have a problem with getting this driver from it's main not as a deb - current instructions for manual installation are I believe outdated, and I failed installing a made driver.

GabMus, 2016-09-06:
If you want I can better help you out on telegram, write me there, my nick
is @gabmus

terrycain, 2016-09-07:
Got the log. All looks good, just need to go through the code setting the macros to make sure I havent limited what keys could be submitted.

terrycain, 2016-09-18:
@Denlior is everything good?

Denlior, 2016-09-18:
Well, deamon works and I can assign macroses to keypresses through razerComander.

There is other stuff I believe it can do, but I don't know how to configure it to. Like using mode and/or tumb buttons as alt/shift/ctrl. As we discussed priorly on this thread.

And, by the way, is it possible to interact with the deamon by command line or a config file?

terrycain, 2016-09-20:
> And, by the way, is it possible to interact with the deamon by command line or a config file?

Config file - not yet, is on the todo list. Command line - there is a proof of concept script in the scripts folder somewhere which will eventaully get expanded.

mode/thumb cant be used as modifiers just yet, as you would need a key macro for when the key is pressed and released. Once I've done some more driver work I do intend to do a fair bit of documentation to allow people to do more themselves.

terrycain, 2016-09-27:
Ok been thinking about it, what sort of options would be useful to you in a config file?

Denlior, 2016-09-27:
Key settings the deamon would use when loaded, so there is no need to set
them on each boot.

On a bigger scale, maybe even different profiles of key setups with
corresponding led schemes. But that implies a way to switch between them,
and, I believe, it makes it bigger than just "config file".

Color, brightness and other visual stuff would be good too, I guess. Just
to make it complete.

P.S.: Let me know if I can help in any way, other than testing stuff. I
know python a little bit - to the extent of being capable to write and use
biopython scripts.

terrycain, 2016-09-28:
> Key settings the deamon would use when loaded

As soon as you mentioned that I thought of a few things :smile: 

> On a bigger scale, maybe even different profiles of key setups...

Exactly, where do you draw the line. I do want a concept of profiles eventually but as this is like my 3rd attempt at the daemon, any new big features are gonna be thought out. (we even have a class diagram for the daemon somewhere)

> Color, brightness and other visual stuff would be good too, I guess. Just
> to make it complete.

I'm torn between adding this and just saying that an app should do it as I have to draw the line of a daemon somewhere. I think a startup brightness and static colour will suffice, if people want anything extra they can use/make an app.

> P.S.: Let me know if I can help in any way, other than testing stuff....

That's the thing, I have no clue what things people can do to help ha.

duapruis, 2016-10-18:
Hi, I just recently bought a Tartarus Chroma because I found out this project out.

First of all congratulations on this amazing project and thanks a lot for all the effort you put into it, I was a bit turned off out of my hype when I found out that the macro keys and profile options were greyed out, but HEY, my fault for not checking github before buying it, right? Not a big issue since I gladly see here that you are already working on it (yay!  :D). Once that being said, if you need an extra tester ,some output/input information that I can give you, etc, just tell me , since I have the physical device maybe I can help you out even if I'm a dummy hehe.

Once again, amazing project, you literally made my day when I found out this project.

Ps: Excuse my English...

terrycain, 2016-10-18:
What's greyed out on where?

duapruis, 2016-10-18:
Wow, that was quick. On the tray applet: Game Mode, Profiles, Enable Macro Keys, Change Primary/Secondary colors and on the top the first option Tartarus Chroma is also greyes out. Ofc this options aren't displayed on the configuration utility.

terrycain, 2016-10-18:
Pretty sure @lah7 took over the tray applet. Probs best to go to his repo and raise issue there

duapruis, 2016-10-18:
So, the options shouldn't be greyed out then? How about the polychromatic controller?

terrycain, 2016-10-18:
Np

terrycain, 2016-12-31:
Aint got time to work on extra stuff right now

closed 2016-12-31 by terrycain