openrazer/openrazer #21: Better communication? I can help with stuff.

Maffblaster, 2016-08-24:
Hi @terrycain, I started a channel on Freenode channel for better communication with this project. I still want to be involved with this project in little ways since I have a Razer Chroma keyboard now. Channel name is: #razer-drivers

You can join using by connecting any IRC client to the Freenode network or by using [Freenode webchat](https://webchat.freenode.net/). I can give you OP privileges, etc. 

A few things:

1. Can you rename the project to "razer-drivers" before lots of people start using it? Looks better and is more consistent with GitHub project naming.
2. I'd love to help work on packaging/testing. Join the channel and we can discuss in real time.
3. If you don't want the channel I can destroy it.

Kind regards!

terrycain, 2016-08-24:
@GabMus has beaten you to it. Currently we're all on this telegram chat thing. Its essentially IRC with the added benefit of sending photos/media, plus it has a nice mobile client which is ideal for me. If you wanna join search @terrycain or @GabMus and get us to add you.

> I still want to be involved with this project in little ways since I have a Razer Chroma keyboard now...I'd love to help work on packaging/testing. Join the channel and we can discuss in real time.

YES we need more people willin. Am currently busy killing launchpad, for release 1.0.0, once its done will put up some docs.

terrycain, 2016-08-24:
Renamed ;) - Closing this to lower teh issue count but feel free to reply.

closed 2016-08-24 by terrycain

GabMus, 2016-08-25:
@Maffblaster if you write me an email I can send you the link to the telegram chat, or if @terrycain is ok with it, I can share the link here in a comment

terrycain, 2016-08-25:
Yeah share teh link

GabMus, 2016-08-25:
There you go: https://telegram.me/joinchat/A1E4RT9TxhnxyetdQCSj7Q

Maffblaster, 2016-08-25:
That works. Gives me an excuse to use Telegram. Thanks, guys!