openrazer/openrazer #22: Orbeweaver Support

Gorta, 2016-08-25:
Hi,

For information about the Orbweaver support, I have an issue with VirtualBox (Orbweaver is always in busy status so I can't connect it to the vm).
I actually try to solve that for the next step.

terrycain, 2016-08-25:
That's most odd. You got any 3rd party software trying to use it?

Gorta, 2016-08-29:
Hi, 
Sorry for this late reply. Now, my orbeweaver is properly connect to the vm (on an over pc ^^)
I have capture the num button 1 to 20. Do you more than that ? (maybe colors and effect ? stick / top and bottom button [I think they are de same than Tartarus]) 

The lsusb for Orbweaver :+1: 
`Bus 002 Device 006: ID 1532:0207 Razer USA, Ltd` 

[1-20.pcapng.gz](https://github.com/terrycain/razer-drivers/files/443115/1-20.pcapng.gz)

terrycain, 2016-08-30:
No problem, I'm not in any rush, got lots of other stuff to keep be busy :wink:.  Yeah captures of effects (especially changing from 1 effect to the other) would be good. 

Is it an orbweaver chroma? As I have spectrum static and breath from the tartarus so any others would be good. (I know exactly what the packet looks like as I know the protocol but I like to see evidence before I implement things as then theres none of "You bricked my device" rubbish). Also if you can assign colours to the led's individually examples of that would be good.

terrycain, 2016-08-30:
Also `ls -vvv -d 1532:0207` would be good. Also Im pretty sure its the same but if you run xev and get me a mapping of what button points to what keycode sort of like whats [here](https://github.com/pez2001/razer_chroma_drivers/issues/131). If its the same then just say that obviously ;)

Gorta, 2016-08-30:
Hi, 

I don't have exactly the same keycode (key are different for the num pad).

`lsusb -vvv -d 1532:0207

Bus 002 Device 002: ID 1532:0207 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0207 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
`

`1 49
2 10
3 11
4 12
5 13
6 23
7 24
8 25
9 26
10 27
11 66
12 38
13 39
14 40
15 41
16 50
17 52
18 53
19 54
20 55
Mode Switch 
Up (hat switch) 111
Down (hat switch) 116
Left (hat switch) 113
Right (hat switch) 114
Bottom Thumb Button (below hat switch) 65`

For the changing effect, Orbweaver chroma has more effect than Tartarus chroma (I have custom, Breathing, Reactive, Spectrum cycling, Static, Wave, Ripple and "none" ^^ ). So, it's better if I create 1 file per switching effect (like "Breathing green > Reactive red", "Reactive red > Spectrum blue" ) ?

terrycain, 2016-08-30:
Wicked :)

> Breathing green > Reactive red", "Reactive red > Spectrum blue" ) ?

YES That's perfect :smiley:. And for the custom ones could you do a couple extra :wink:. Preferably name the files or add a readme saying what they do though I've seen the effects on other devices it _should_ be obvious.

terrycain, 2016-08-30:
Ahh of course the orbweaver has an extra row of keys

terrycain, 2016-09-20:
Any luck on those effects captures?

madbobmcjim, 2016-09-22:
I've got the non-chroma orbweaver, will this still support keymaps? Is there any information required about this device?

terrycain, 2016-09-22:
I think the keymaps are different from above anyway, as I was looking as the wrong numbers. some work is still requried for the non orbweaver, though significantly less

terrycain, 2016-09-22:
Gimme till the weekend then ill look into it. And if I forget, ping me

madbobmcjim, 2016-09-30:
Any luck? I'm happy to get the keymaps from my orbweaver if that would help (and you could explain the process)

terrycain, 2016-09-30:
Completely forgot about this ha. I will answer this weekend. Theres a script that gets this information but it'll need a bit of modifying for the tartarus. A list of what the orbweaver features would be good. Plus output from
`lsusb`
`ls -l /dev/input/by-id`

madbobmcjim, 2016-09-30:
Sure, here you go:
`lsusb
Bus 003 Device 007: ID 1532:0113 Razer USA, Ltd RZ07-0074 Gaming Keypad [Orbweaver]

ls -l /dev/input/by-id
lrwxrwxrwx 1 root root 9 Sep 30 22:29 usb-Razer_Razer_Orbweaver-event-kbd -> ../event6
lrwxrwxrwx 1 root root 9 Sep 30 22:29 usb-Razer_Razer_Orbweaver-if01-event-kbd -> ../event7
lrwxrwxrwx 1 root root 9 Sep 30 22:29 usb-Razer_Razer_Orbweaver-if02-event-mouse -> ../event8
lrwxrwxrwx 1 root root 9 Sep 30 22:29 usb-Razer_Razer_Orbweaver-if02-mouse -> ../mouse1
`

The Orbweaver has a 5x4 keypad, a directional pad with 2 buttons, 3 LEDs used for indicating key mappings, and the keypad only glows green. 

Cheers :-)

terrycain, 2016-10-01:
Can the green glow's brightness be changed? and can to do any effects?

terrycain, 2016-10-01:
@madbobmcjim what OS you running? If its a deb based one, add the PPA, install the daemon etc... Then clone this repo and there will be a script `REPO_DIR/scripts/misc/event_reader.py`, run that with the path to the orbweavers event file `/dev/input/by-id/usb-Razer_Razer_Orbweaver-event-kbd` and the argument `--tartarus` and then press keys on the orbweaver, it should output some stuff.

You with me so far?

madbobmcjim, 2016-10-01:
Ok, I've installed the PPA, cloned the repo, modified the event_reader file to point at the Orbweaver-event file, but it's giving me an error when I run it:
'python razer-drivers/scripts/misc/event_reader.py '
'Traceback (most recent call last):'
'  File "razer-drivers/scripts/misc/event_reader.py", line 2, in <module>'
'    from razer_daemon.keyboard import EVENT_MAPPING'
'ImportError: No module named razer_daemon.keyboard'

I'm running ubuntu 16.04. I've also got a dual boot into Win 10 and I think there was a 'pulse' option in the windows driver. I'll double check in a little while.

terrycain, 2016-10-01:
did you install daemon? it wants a file in that otherwise you have to mess around with pythonpaths which is too much hassle

madbobmcjim, 2016-10-01:
I went through the process on the website, so I ran:
 `sudo apt install python3-razer razer-kernel-modules-dkms razer-daemon razer-doc`
I also modprobed razerkbd and I've tried a reboot too.

terrycain, 2016-10-01:
use python3

wrohdewald, 2016-10-01:
Out of curiosity I tried event_reader.py today. I had the same error message until I used python3. Maybe you could add some #!/usr/bin/env python3 ?

terrycain, 2016-10-01:
Yeah it was only a misc script that I used to run in my IDE. I'll do that :wink:

madbobmcjim, 2016-10-01:
Doh! :-)

Here's the output, it is outputting a backtick, which was a little fiddly to input into the markdown code block. The first block of text is the keypad, the second is the D-pad and it's two buttons:

`python3 event_reader.py --tartarus`
`Type: EV_KEY, Code: BACKTICK, Value: DOWN`
```Type: EV_KEY, Code: BACKTICK, Value: UP``
`Type: EV_KEY, Code: 1, Value: DOWN`
`1Type: EV_KEY, Code: 1, Value: UP`
`Type: EV_KEY, Code: 2, Value: DOWN`
`2Type: EV_KEY, Code: 2, Value: UP`
`Type: EV_KEY, Code: 3, Value: DOWN`
`3Type: EV_KEY, Code: 3, Value: UP`
`Type: EV_KEY, Code: 4, Value: DOWN`
`4Type: EV_KEY, Code: 4, Value: UP`
`Type: EV_KEY, Code: TAB, Value: DOWN`
`	Type: EV_KEY, Code: TAB, Value: UP`
`Type: EV_KEY, Code: Q, Value: DOWN`
`QType: EV_KEY, Code: Q, Value: UP`
`Type: EV_KEY, Code: W, Value: DOWN`
`WType: EV_KEY, Code: W, Value: UP`
`Type: EV_KEY, Code: E, Value: DOWN`
`EType: EV_KEY, Code: E, Value: UP`
`Type: EV_KEY, Code: R, Value: DOWN`
`RType: EV_KEY, Code: R, Value: UP`
`Type: EV_KEY, Code: CAPSLK, Value: DOWN`
`Type: EV_KEY, Code: CAPSLK, Value: UP`
`Type: 17, Code: 1, Value: 0`
`Type: EV_KEY, Code: A, Value: DOWN`
`aType: EV_KEY, Code: A, Value: UP`
`Type: EV_KEY, Code: S, Value: DOWN`
`sType: EV_KEY, Code: S, Value: UP`
`Type: EV_KEY, Code: D, Value: DOWN`
`dType: EV_KEY, Code: D, Value: UP`
`Type: EV_KEY, Code: F, Value: DOWN`
`fType: EV_KEY, Code: F, Value: UP`
`Type: EV_KEY, Code: LEFTSHIFT, Value: DOWN`
`Type: EV_KEY, Code: LEFTSHIFT, Value: UP`
`Type: EV_KEY, Code: Z, Value: DOWN`
`zType: EV_KEY, Code: Z, Value: UP`
`Type: EV_KEY, Code: X, Value: DOWN`
`xType: EV_KEY, Code: X, Value: UP`
`Type: EV_KEY, Code: C, Value: DOWN`
`cType: EV_KEY, Code: C, Value: UP`
`Type: EV_KEY, Code: V, Value: DOWN`
`vType: EV_KEY, Code: V, Value: UP`

`Type: EV_KEY, Code: LEFTALT, Value: DOWN`
`Type: EV_KEY, Code: LEFTALT, Value: UP`
`Type: EV_KEY, Code: SPACE, Value: DOWN`
` Type: EV_KEY, Code: SPACE, Value: UP`
`Type: EV_KEY, Code: DOWNARROW, Value: DOWN`
`^[[BType: EV_KEY, Code: DOWNARROW, Value: UP`
`Type: EV_KEY, Code: UPARROW, Value: DOWN`
`^[[AType: EV_KEY, Code: UPARROW, Value: UP`
`Type: EV_KEY, Code: LEFTARROW, Value: DOWN`
`^[[DType: EV_KEY, Code: LEFTARROW, Value: UP`
`Type: EV_KEY, Code: RIGHTARROW, Value: DOWN`
`^[[CType: EV_KEY, Code: RIGHTARROW, Value: UP`

Also, I checked the Windows driver and it's got options to change the brightness and to pulsate:
[image]

terrycain, 2016-10-01:
Would you mind editing the above to say what key you pressed compared to what key the program thinks it is ;)

madbobmcjim, 2016-10-01:
Ok, edited. Let me know if it still doesn't make any sense :-)

terrycain, 2016-10-01:
Ok install daemon deb from [here](https://drive.google.com/open?id=0BydYgD2LzDUCeHR4SnJFN25BeFU), then re-clone the repository and change to the `feature_orbweaver_0x0113` branch, then run that script and get the output again.

[Heres](https://github.com/terrycain/razer-drivers/blob/feature_orbweaver_0x0113/scripts/misc/event_reader.py) a direct link to that script.

Run script like
```
./event_reader.py --orbweaver /dev/input/by-id/usb-Razer_Razer_Orbweaver-event-kbd
```
That should be the correct event file.

In theory it should recognise all the keys and no longer think keys like 15 mean F ha. Once that's all good, I can whip up a driver for you to test out and add daemon support. Shouldn't be too difficult.

madbobmcjim, 2016-10-04:
Yep that seems to work, it's now outputting the correct code values to the corresponding key number:

./event_reader.py --orbweaver /dev/input/by-id/usb-Razer_Razer_Orbweaver-event-kbd
Starting. Press keys
Type: EV_KEY, Code: '1', Value: DOWN
`Type: EV_KEY, Code: '1', Value: UP
Type: EV_KEY, Code: '2', Value: DOWN
1Type: EV_KEY, Code: '2', Value: UP
Type: EV_KEY, Code: '3', Value: DOWN
2Type: EV_KEY, Code: '3', Value: UP
Type: EV_KEY, Code: '4', Value: DOWN
3Type: EV_KEY, Code: '4', Value: UP
Type: EV_KEY, Code: '5', Value: DOWN
4Type: EV_KEY, Code: '5', Value: UP
Type: EV_KEY, Code: '6', Value: DOWN
	Type: EV_KEY, Code: '6', Value: UP
Type: EV_KEY, Code: '7', Value: DOWN
QType: EV_KEY, Code: '7', Value: UP
Type: EV_KEY, Code: '8', Value: DOWN
WType: EV_KEY, Code: '8', Value: UP
Type: EV_KEY, Code: '9', Value: DOWN
EType: EV_KEY, Code: '9', Value: UP
Type: EV_KEY, Code: '10', Value: DOWN
RType: EV_KEY, Code: '10', Value: UP
Type: EV_KEY, Code: '11', Value: DOWN
Type: EV_KEY, Code: '11', Value: UP
Type: 17, Code: 1, Value: 0
Type: EV_KEY, Code: '12', Value: DOWN
aType: EV_KEY, Code: '12', Value: UP
Type: EV_KEY, Code: '13', Value: DOWN
sType: EV_KEY, Code: '13', Value: UP
Type: EV_KEY, Code: '14', Value: DOWN
dType: EV_KEY, Code: '14', Value: UP
Type: EV_KEY, Code: '15', Value: DOWN
fType: EV_KEY, Code: '15', Value: UP
Type: EV_KEY, Code: '16', Value: DOWN
Type: EV_KEY, Code: '16', Value: UP
Type: EV_KEY, Code: '17', Value: DOWN
zType: EV_KEY, Code: '17', Value: UP
Type: EV_KEY, Code: '18', Value: DOWN
xType: EV_KEY, Code: '18', Value: UP
Type: EV_KEY, Code: '19', Value: DOWN
cType: EV_KEY, Code: '19', Value: UP
Type: EV_KEY, Code: '20', Value: DOWN
vType: EV_KEY, Code: '20', Value: UP
Type: EV_KEY, Code: 'MODE_SWITCH', Value: DOWN
Type: EV_KEY, Code: 'MODE_SWITCH', Value: UP
Type: EV_KEY, Code: 'THUMB', Value: DOWN
 Type: EV_KEY, Code: 'THUMB', Value: UP
Type: EV_KEY, Code: 'LEFT', Value: DOWN
^[[DType: EV_KEY, Code: 'LEFT', Value: UP
Type: EV_KEY, Code: 'UP', Value: DOWN
^[[AType: EV_KEY, Code: 'UP', Value: UP
Type: EV_KEY, Code: 'RIGHT', Value: DOWN
^[[CType: EV_KEY, Code: 'RIGHT', Value: UP
Type: EV_KEY, Code: 'DOWN', Value: DOWN
^[[BType: EV_KEY, Code: 'DOWN', Value: UP

terrycain, 2016-10-04:
OK armed with that, on the weekend I'll whip up a driver for you to try and were go from there.

madbobmcjim, 2016-10-04:
Sweet, thanks :-)

terrycain, 2016-10-04:
Yeah hopefully the setting of the "profile" led's is the same as the tartarus, and if the pulsate is the same as my bw2013 then technically we have everything we need.

madbobmcjim, 2016-10-11:
Any luck with that driver? :-)

terrycain, 2016-10-11:
:-( I totally forgot. I have something somewhere just need to see if it's ready. Prod me on Saturday if nothings here as it should be done by then.

GRANITsalvo, 2016-10-12:
Happy to see work for the older orbweaver, but I'm less confident with linux and I think I'm not following one of the steps above.  Would it be a better idea for me to just wait for an official update for support?  I'm not in a rush, but pretty excited to have one less reason to boot back into Windows.

RD777, 2016-10-16:
Hey there, just thought I'd ask if you had had any time to come up with a driver.  Thanks in advance

terrycain, 2016-10-17:
Sorry guys was super busy this weekend but there is a driver I created at some point in time I'll make it a priority to go over it to ensure it "should" work and then I'll drop some details on how to install it.

What distribution you guys using as if it's ubuntu I could just roll out some .debs

madbobmcjim, 2016-10-17:
I'm on Ubuntu, so feel free to throw me a deb when you're ready.

terrycain, 2016-10-17:
https://drive.google.com/open?id=0BydYgD2LzDUCeHR4SnJFN25BeFU

Install the debs and then restart. Then run `tree /sys/bus/hid/drivers/`

madbobmcjim, 2016-12-22:
Hi, sorry for the delay on this. I've removed the old installs, installed the new debs and I get:
```
tree /sys/bus/hid/drivers/
/sys/bus/hid/drivers/
├── hid-generic
│   ├── 0003:046D:C249.0001 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9.2/3-9.2:1.0/0003:046D:C249.0001
│   ├── 0003:046D:C249.0002 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9.2/3-9.2:1.1/0003:046D:C249.0002
│   ├── 0003:1532:0113.0005 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9.4/3-9.4.1/3-9.4.1:1.0/0003:1532:0113.0005
│   ├── 0003:1532:0113.0006 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9.4/3-9.4.1/3-9.4.1:1.1/0003:1532:0113.0006
│   ├── 0003:1532:0113.0007 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9.4/3-9.4.1/3-9.4.1:1.2/0003:1532:0113.0007
│   ├── bind
│   ├── module -> ../../../../module/hid_generic
│   ├── new_id
│   ├── uevent
│   └── unbind
└── microsoft
    ├── 0003:045E:00DB.0003 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9.3/3-9.3:1.0/0003:045E:00DB.0003
    ├── 0003:045E:00DB.0004 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9.3/3-9.3:1.1/0003:045E:00DB.0004
    ├── bind
    ├── module -> ../../../../module/hid_microsoft
    ├── new_id
    ├── uevent
    └── unbind

11 directories, 8 files
```
The Microsoft entry is probably my keyboard.

terrycain, 2016-12-22:
That link probably links to some debs that arnt relevant. I'll get a link for the orbweaver ones tonight

terrycain, 2016-12-22:
Ok looking back at what I did, I didn't even make a driver for it lol. Gimme a few days and I should have something to try out.

terrycain, 2016-12-22:
https://drive.google.com/drive/folders/0BydYgD2LzDUCeDZoTmNqWFdXSXc?usp=sharing

Install debs then reboot. Join telegram chat (add me `@terrycain`)  if you want. Lots of us are on there.

madbobmcjim, 2016-12-22:
That looks better, here's my tree:

```
tree /sys/bus/hid/drivers/
/sys/bus/hid/drivers/
├── hid-generic
│   ├── 0003:046D:C249.0001 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9.2/3-9.2:1.0/0003:046D:C249.0001
│   ├── 0003:046D:C249.0002 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9.2/3-9.2:1.1/0003:046D:C249.0002
│   ├── bind
│   ├── module -> ../../../../module/hid_generic
│   ├── new_id
│   ├── uevent
│   └── unbind
├── microsoft
│   ├── 0003:045E:00DB.0003 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9.3/3-9.3:1.0/0003:045E:00DB.0003
│   ├── 0003:045E:00DB.0004 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9.3/3-9.3:1.1/0003:045E:00DB.0004
│   ├── bind
│   ├── module -> ../../../../module/hid_microsoft
│   ├── new_id
│   ├── uevent
│   └── unbind
└── razerkbd
    ├── 0003:1532:0113.0005 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9.4/3-9.4.1/3-9.4.1:1.0/0003:1532:0113.0005
    ├── 0003:1532:0113.0006 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9.4/3-9.4.1/3-9.4.1:1.1/0003:1532:0113.0006
    ├── 0003:1532:0113.0007 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9.4/3-9.4.1/3-9.4.1:1.2/0003:1532:0113.0007
    ├── bind
    ├── module -> ../../../../module/razerkbd
    ├── new_id
    ├── uevent
    └── unbind
```
I'm not on telegram yet, I'll take a look when I get a moment.

terrycain, 2016-12-22:
`ls -lah /sys/bus/hid/drivers/razerkbd/0003:1532:0113.0007/`

madbobmcjim, 2016-12-22:
```
ls -lah /sys/bus/hid/drivers/razerkbd/0003:1532:0113.0007/
total 0
drwxr-xr-x 5 root plugdev    0 Dec 22 20:14 .
drwxr-xr-x 5 root root       0 Dec 22 20:14 ..
-r--r--r-- 1 root plugdev 4.0K Dec 22 20:14 country
-rw-rw---- 1 root plugdev 4.0K Dec 22 20:14 device_mode
-r--r----- 1 root plugdev 4.0K Dec 22 20:14 device_serial
-r--r----- 1 root plugdev 4.0K Dec 22 20:14 device_type
lrwxrwxrwx 1 root plugdev    0 Dec 22 20:14 driver -> ../../../../../../../../../bus/hid/drivers/razerkbd
-r--r----- 1 root plugdev 4.0K Dec 22 20:14 firmware_version
drwxr-xr-x 3 root plugdev    0 Dec 22 20:14 hidraw
drwxr-xr-x 3 root plugdev    0 Dec 22 20:14 input
-rw-rw---- 1 root plugdev 4.0K Dec 22 20:14 matrix_brightness
--w--w---- 1 root plugdev 4.0K Dec 22 20:14 matrix_effect_none
-rw-rw---- 1 root plugdev 4.0K Dec 22 20:14 matrix_effect_pulsate
--w--w---- 1 root plugdev 4.0K Dec 22 20:14 matrix_effect_static
-r--r--r-- 1 root plugdev 4.0K Dec 22 20:14 modalias
drwxr-xr-x 2 root plugdev    0 Dec 22 20:14 power
-rw-rw---- 1 root plugdev 4.0K Dec 22 20:14 profile_led_blue
-rw-rw---- 1 root plugdev 4.0K Dec 22 20:14 profile_led_green
-rw-rw---- 1 root plugdev 4.0K Dec 22 20:14 profile_led_red
-r--r--r-- 1 root plugdev 4.0K Dec 22 20:14 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Dec 22 20:14 subsystem -> ../../../../../../../../../bus/hid
-rw-rw---- 1 root plugdev 4.0K Dec 22 20:14 test
-rw-rw---- 1 root plugdev 4.0K Dec 22 20:14 test2
-rw-r--r-- 1 root plugdev 4.0K Dec 22 20:14 uevent
-r--r----- 1 root plugdev 4.0K Dec 22 20:14 version
```

terrycain, 2016-12-22:
Ok  as root not sudo from inside that folder

```
cat device_serial
cat device_type
cat firmware_version

# Should change brightness
echo -n '255' > matrix_brightness
echo -n '128' > matrix_brightness
echo -n '64' > matrix_brightness
echo -n '255' > matrix_brightness

# "Should" "turn off" light
echo -n '1' > matrix_effect_none
# "Should" set a breathing/pulsing effect
echo -n '1' > matrix_effect_pulsate
# "Should" set static effect
echo -n '1' > matrix_effect_static

# Should toggle the "profile" leds
echo -n '1' > profile_led_blue
echo -n '0' > profile_led_blue

echo -n '1' > profile_led_green
echo -n '0' > profile_led_green

echo -n '1' > profile_led_red
echo -n '0' > profile_led_red
```

madbobmcjim, 2016-12-22:
```
root@desktop:/sys/devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9.4/3-9.4.1/3-9.4.1:1.2/0003:1532:0113.0007# cat device_serial 
PM1417074403268
root@desktop:/sys/devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9.4/3-9.4.1/3-9.4.1:1.2/0003:1532:0113.0007# cat device_type 
Razer Orbweaver
root@desktop:/sys/devices/pci0000:00/0000:00:14.0/usb3/3-9/3-9.4/3-9.4.1/3-9.4.1:1.2/0003:1532:0113.0007# cat firmware_version 
v1.1
```
The serial matches that of my Orbweaver.
```
# Should change brightness
echo -n '255' > matrix_brightness
echo -n '128' > matrix_brightness
echo -n '64' > matrix_brightness
echo -n '255' > matrix_brightness
```
I couldn't notice any change in brightness, until I set it to '0' then it went off and I couldn't get it back again.
```
# "Should" "turn off" light
echo -n '1' > matrix_effect_none
# "Should" set a breathing/pulsing effect
echo -n '1' > matrix_effect_pulsate
# "Should" set static effect
echo -n '1' > matrix_effect_static
```
These didn't do anything, I'll reboot and check if setting the last one to '0' killed these too.
```
# Should toggle the "profile" leds
echo -n '1' > profile_led_blue
echo -n '0' > profile_led_blue

echo -n '1' > profile_led_green
echo -n '0' > profile_led_green

echo -n '1' > profile_led_red
echo -n '0' > profile_led_red
```
These worked fine. Although on this model the LEDs are yellow, green, blue. Toggling profile_led_red toggled the yellow LED.

madbobmcjim, 2016-12-22:
Yep still the same after a reboot (keyboard LEDs are off) unplugging and replugging the Orbweaver brought them back, but the pulsate command didn't have any effect, nor changing the brightness.

terrycain, 2016-12-22:
ok try the following, one could be static and one could be pulsate
```
echo -n '1' > test
echo -n '1' > tes2
```

madbobmcjim, 2016-12-22:
That got it.
test2 = pulsate
test = static

terrycain, 2016-12-22:
Yeah the old bw keyboards refered to the backlight as the "logo_led" but it seems the orbweaver uses the proper "backlight_led" ID. I'll adjust brightness to also use backlight. That red,green,blue profile files map to the proper colours?

madbobmcjim, 2016-12-22:
Mostly:
Red = Yellow
Green = Green
Blue = Blue

This thing doesn't have a red LED.

terrycain, 2016-12-22:
Ok I'll leave the name as red as thats what it says in the spec I have, but whatever client app you use can say yellow :wink:

Try [this](https://drive.google.com/open?id=0BydYgD2LzDUCeDZoTmNqWFdXSXc) Brightness "should" work now as well as static/pulsate.

madbobmcjim, 2016-12-23:
That fixed it. The matrix_brightness now works, as does the matrix_effect_pulsate and matrix_effect_static. :-)

terrycain, 2016-12-23:
Will whip up daemon support later, can you install `d-feet` that'll help with some debugging

terrycain, 2016-12-23:
New debs https://drive.google.com/open?id=0BydYgD2LzDUCeDZoTmNqWFdXSXc

mjtorn, 2017-01-04:
I happen to own an Orbweaver for my Steam Machine (running SteamOS). So far I have made a repository, which is not public yet, with the dependencies available in Debian but not available for SteamOS and would like to give my Orbweaver a test.

Actually what I need to do is be able to remap some keys, as ESC is not directly available on it :(
This is a problem for most games that use a keyboard and are not by Valve.

@terrycain should I run those 0.1.13 debs from above or wait until Orbweaver gets merged/synced? Is that far on the roadmap?

Thanks!

terrycain, 2017-01-04:
@mjtorn fancy finishing off the testing? You'd need the application `d-feet`

mjtorn, 2017-01-05:
@terrycain what do you need? So far things look good on the razer-driver front.

It's mainly Polychromatic that I'm not entirely sure about, as I understood it would allow me to remap keys. Judging by https://github.com/lah7/polychromatic/issues/47#issuecomment-270502825 it's a question of the daemon supporting a matrix for keypads, is that what `d-feet` is good for debugging?

But yeah, let me know what you need and I'll get back to you when I can :)

mjtorn, 2017-01-05:
> And the orbweaver doesnt have a matrix, the new one does though.

Could you elaborate on what having a matrix means? AFAIU it's something related to mapping keys, right? Is this on a hardware level?

Also, what's the new one? Nostromo?

> Can you join the telegram chat

That's a bit fishier, because I'm not on Telegram (I use Signal for most of my communication) and there are a couple of other points. I have all this on the Steam Machine, which is hooked to the living room TV, which is often not available... I'd have to have the missus do something else, hoard the TV, be on Telegram _and_ hope someone (you, anyone) is also online at the same time. That's a lot of moving parts!

Besides, if the steps get walked through on GitHub, they turn into documentation ;)

Thanks!

terrycain, 2017-01-06:
> Could you elaborate on what having a matrix means

Sorry by matrix I generally mean if the LEDs are individually programmable, the Orbweaver Chroma can do this (hence the ability to perform the wave function). 

And I'll get some instructions for you in a bit. All it'll be is running the daemon (which i think should already be working) and then clicking on a few things.

[11 more replies]

commit 7bda56ba (2017-01-16): Added Orbweaver to README

Closes #22
closed 2017-01-16 by terrycain