openrazer/openrazer #23: Razer DeathAdder Chroma - support

presianbg, 2016-08-26:
Hi,

Opening this issue, regarding pez2001/razer_chroma_drivers#126.

I will drop some dumps when I have the time for it :) Thank you.

presianbg, 2016-08-28:
@terrycain  

I do not think I will be able to do the reverse engineering on the Death Adders soon enough. Can you please look at mbuesch/razer#38 ?

I believe information from this PR will do the trick :)

Let me know if I'm right/wrong...

terrycain, 2016-08-28:
No rush mate. Tbf I could create the deathadder driver without any reference or dumps. The only reason I want the dumps is to verify what I've created as I want to be as confident as possible when releasing a driver so I don't get people moaning about bricked devices.

Gimme a week or two as busy at the moment as just released v1.0.0. I can make a driver at some point, just beware it would be more volatile and experimental than the normal test drivers I release.

terrycain, 2016-09-20:
Have a few ideas for this but will probably wait until I tidy up the mouse driver a bit.

plumpudding, 2016-09-26:
I will get mine in a few days. What details can I contribute?

terrycain, 2016-09-27:
Something like [this](https://github.com/terrycain/razer-drivers/issues/51), and [here](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol)

plumpudding, 2016-09-27:
Omg that means I will have to install *Windows* bah

terrycain, 2016-09-27:
Is a temporary evil :wink:

z3ntu, 2016-10-10:
@terrycain do you need traces for this to progress? I would be happy to help if they are necessary.

terrycain, 2016-10-10:
Screenshots like in #59 would help :)

z3ntu, 2016-10-10:
[image]
[image]
[image]
[image]
[image]
[image]

terrycain, 2016-10-10:
Good :) I just love the deathadder and naga, they just ruin my naming scheme :wink:, will have a think about how I want to go about this

z3ntu, 2016-10-10:
Hmm what naming scheme?

terrycain, 2016-10-10:
Well currently like [this](https://github.com/terrycain/razer-drivers/wiki/Using-the-keyboard-driver). I have an endpoint like `mode_static`, `mode_spectrum` etc... And now the deathadder as 2 LEDs that can have seperate effects and the Naga has 3

plumpudding, 2016-10-11:
Anything else you need? I have mine now too.

terrycain, 2016-10-11:
Lotsa packet captures like mentioned [here](https://github.com/terrycain/razer-drivers/issues/23#issuecomment-249841813) would be nice :wink: other than that familarise yourselves with the wiki as I'll need testers

z3ntu, 2016-10-11:
So... Have a [zip file with a big capture](https://github.com/terrycain/razer-drivers/files/522612/deathadder_razer.zip) :) if you need explanations what exactly happened, I have recorded (nearly) everything I did: https://youtu.be/FfG-0dQac7Q . Before recording, I wrote the following down:

*All numbers are from the packet nrs the last packet after I did something! (also in the video)*

> 4022 -> plugged in
9548 -> moved it a bit around (incl scrollwheel)

terrycain, 2016-10-11:
Lol nice. Moving it around adds packet spam that's the only bad thing but I can filter that out :smile: will have a glance at it tomorrow but probs won't get to have a good look till the weekend

z3ntu, 2016-10-12:
@terrycain I just moved it around a bit at the beginning and then never again, and the last packet after I have moved it was `9548` so I hope that helps.

terrycain, 2016-10-12:
Ahh that's alright then np will have a look soon :-)

z3ntu, 2016-10-29:
bumpy bump

terrycain, 2016-10-30:
Sorry guys been super busy (and went on holiday) will try and make a start this evening

terrycain, 2016-10-30:
@z3ntu You got a DA chroma yeah? Whats the output of `lsusb` think the PID im looking for is `0043`

z3ntu, 2016-10-30:
I'm not at home ATM but yes, it's the Deathadder chroma (not the 2013 one, I think)

terrycain, 2016-10-30:
Nice :smile: Have made an initial start with the driver. In theory I should have all functionality but want to go over it before I let people test it.

z3ntu, 2016-10-30:
Please let me test it <3 ^^

terrycain, 2016-10-30:
@z3ntu what os you on again? And once Ive got the lsusb output I can finialise things

z3ntu, 2016-10-30:
http://pastebin.com/zmsNWbsu
Arch Linux

terrycain, 2016-10-30:
Ta, what os you on again?

z3ntu, 2016-10-30:
> Arch Linux

terrycain, 2016-10-30:
@z3ntu Try out branch `feature_DeathAdder0x0043` should "just work" once the driver is mounted under `/sys/bus/hid/drivers/razermouse` then we can proceed with testing, if it doesnt bind automatically, go through the wiki and see if you can get it to bind manually.

terrycain, 2016-10-30:
Just pushed udev changes to that branch.

z3ntu, 2016-10-30:
So I just installed the packages `razer-driver-dkms`, `razer-daemon` & `python-razer`, ran `modprobe razermouse` and `razer-service` but what should I do now?

terrycain, 2016-10-30:
Ahh the packages you have found probably dont contain the changes I've made to that branch unless you've made the packages youself

z3ntu, 2016-10-30:
I changed the script to use the new branch. Is something like [in the keyboard wiki page](https://github.com/terrycain/razer-drivers/wiki/Setting-up-the-keyboard-driver) what I should do?

terrycain, 2016-10-30:
Yeah that page, starting at `binding the chroma` section. On the wiki page its looking for interface 2, mouses use interface 0, so your looking for the mouse with the lowest end number, so if you have 3 like
```
0000:1532:0043.000B
0000:1532:0043.000C
0000:1532:0043.000D
```
You want to pick `0000:1532:0043.000B`

z3ntu, 2016-10-30:
It's 1, 2 & 3 for me.
I got it working
```
# cat device_type 
Razer DeathAdder Chroma
```

terrycain, 2016-10-30:
Yeah numbers can reset sometimes. What are the rest of the files in that directory. As ive done something different with this one

z3ntu, 2016-10-30:
```
# ls -al /sys/bus/hid/devices/0003:1532:0043.000*
lrwxrwxrwx 1 root root 0 30. Okt 20:33 /sys/bus/hid/devices/0003:1532:0043.0001 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.2/3-2.2:1.0/0003:1532:0043.0001/
lrwxrwxrwx 1 root root 0 30. Okt 20:33 /sys/bus/hid/devices/0003:1532:0043.0002 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.2/3-2.2:1.1/0003:1532:0043.0002/
lrwxrwxrwx 1 root root 0 30. Okt 20:33 /sys/bus/hid/devices/0003:1532:0043.0003 -> ../../../devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2.2/3-2.2:1.2/0003:1532:0043.0003
```

z3ntu, 2016-10-30:
Btw razercommander doesn't detect the mouse, no idea where the problem is.

terrycain, 2016-10-30:
ls inside the folders :wink: 
And yeah that would be because I havent added daemon support yet.

z3ntu, 2016-10-30:
(I have updated my comment before, maybe it didn't update for you)
Ah okay.

Should `report_descriptor` be something human readable? It's random gibberish right now.

z3ntu, 2016-10-30:
How can I change the effect? The wiki page from the mouse doesn't tell me very much :/

terrycain, 2016-10-30:
A lot of them in there are made by the system which we can ignore. Some of them will take in ascii, and some binary. 

`device_type`, `get_firmware_version`, `get_serial` and `version` should be human readable

Hence they shoudn't really be messed with manually. (bar testing of course).

First of all you want to su to root (as we're be piping stuff) and then cd to the folder containing `scroll_*`

Running the following command should change the brightness of the scroll wheel LED, bear in mind this one might not work at all. (also assuming the scroll wheel LED is currently illuminated)
```
echo -n '255' > ./scroll_led_brightness
echo -n '128' > ./scroll_led_brightness
echo -n '0' > ./scroll_led_brightness
echo -n '128' > ./scroll_led_brightness
echo -n '255' > ./scroll_led_brightness
```

z3ntu, 2016-10-30:
Yes, scroll_led_brightness works! (and the 4 files you mentioned are all working)

terrycain, 2016-10-30:
good good. On to the next item.
Some of the other razer hardware abstract from this but I thought as things arnt being done manually as we have a daemon passing though this complexity is fine. With keyboards when you tell it to do static itll do the following - enable the LED/s, set the static colours, and then set static effect. With the DA chroma, you perform those steps in sequence

So this should enable the scroll wheel, then set colour, then set static (if last effect is static, shouldnt make much difference)
```
echo -ne "\x01" > ./scroll_led_state
echo -ne "\xFF\x00\x00" > ./scroll_led_rgb
echo -ne "\x00" > ./scroll_led_effect
```
Should in theory change scroll wheel to static red colour

z3ntu, 2016-10-30:
Seems to work too (I just forgot that I had the brightness turned to 0 :D)

terrycain, 2016-10-30:
Hehe. Ok there's a ton more but am done for today. I'll comment tomorrow with a list of things left to do (basically just testing the other effects, then repeating all of the above but for the LOGO LED.

z3ntu, 2016-10-30:
Apparently the `led_state` things turn the LED off no matter it is't `\x01` or `\x00`.

terrycain, 2016-10-30:
Yeah currenlty in the code that's what it does. Was copied from one place to another will set it properly once all is verified working

z3ntu, 2016-10-31:
Should I try something like 49e4d28850069cb95d1f57b1acc8226b2d083b5e for the daemon support myself?

terrycain, 2016-10-31:
Leave that to me, i'll get that working once the rest of the stuff is verified.

terrycain, 2016-11-03:
Sorry been super busy, will try and add more stuff tonight

terrycain, 2016-11-03:
Ok the following should enable spectrum effect

```
echo -ne "\x01" > ./scroll_led_state
echo -ne "\x04" > ./scroll_led_effect
```

This sequence should enable breathing/pulsating effect
```
echo -ne "\x01" > ./scroll_led_state
echo -ne "\x00\xFF\x00" > ./scroll_led_rgb
echo -ne "\x02" > ./scroll_led_effect
```

Dont try other values for ./scroll_led_effect, they wont be supported and in theory do nothing but I cant guess at that.

z3ntu, 2016-11-03:
Leaving this here for reference:
```
cd /sys/bus/hid/drivers/hid-generic/
echo -n "0003:1532:0043.0001" > unbind
cd ../razermouse/
echo -n "0003:1532:0043.0001" > bind
cd 0003:1532:0043.0001/
```

Is it normal, that I have to re-set the brightness away from 0 after running the first 2 commands?

And yes, the scrollwheel is going through all colors.

And pulsating works too.

terrycain, 2016-11-03:
Reset brightness after bind/unbind?

z3ntu, 2016-11-03:
No sry, I mean after running
```
echo -ne "\x01" > ./scroll_led_state
echo -ne "\x04" > ./scroll_led_effect
```

terrycain, 2016-11-04:
Hmm most odd, were in uncharted territory here so yeah might have to do that for now.

Ok repeat all that but for the files starting with `logo_`

z3ntu, 2016-11-04:
I'm not at home right now and won't be until Sunday evening...

z3ntu, 2016-11-04:
I'm not at home right now but I can do it in the evening.

terrycain, 2016-11-04:
no rush mate, no rush

[69 more replies]

closed 2016-12-04 by terrycain