openrazer/openrazer #30: issue with installing via PPA

rivetrik, 2016-08-29:
while trying to update the driver from the depricated one to the newest version i get this error after i add the ppa and sudo apt-get update pulls this error:

W: The repository 'http://downloads.sourceforge.net/project/xenlism-wildfire/repo deb/ Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.

also the deamon still installs except the driver, when ever i try to call razer commander from the terminal i get this output:

rik@rik-Blade-Stealth ~> sudo razercommander 
no devices

lah7, 2016-08-29:
`razer-chroma-driver` is the old package name, the new packages are:

* `python3-razer`
* `razer-kernel-modules-dkms`
* `razer-daemon`
* `razer-doc`

GabMus, 2016-08-29:
@rivetrik this sounds more like a problem with the daemon or the udev rules, not the driver itself. Looks like the python lib was imported correctly otherwise razerCommander would have complained differently in the terminal. Try rebooting your computer, if that doesn't work, run `killall razerdaemon && razer-service` and try again

rivetrik, 2016-08-30:
yep, did all that and installed correctly. still no device found, any ideas as to why i get that error?

GabMus, 2016-08-30:
We can figure it out. If you want we can chat here, but it'd be faster if you contact me on telegram @gabmus.

rivetrik, 2016-08-30:
cool, ill go ahead and do that real quick if your up for it now

closed 2016-09-03 by terrycain