openrazer/openrazer #31: Razer Anansi Support

Zerophase, 2016-08-29:
I followed setting up the [keyboard driver](https://github.com/pez2001/razer_chroma_drivers/wiki/Setting-up-the-keyboard-driver) till binding to razerkbd. At that point bind returned no such device.  I was able to successfully unbind from hid-generic with 

> echo -n "0003:1532:010F.0003" > unbind

0003 is the mouse interface for Anansi.
Any suggestions for getting this working with Anansi?

terrycain, 2016-08-29:
Well the Anansi isn't currently supported. There is a supported device list in the README. If you think you could do [this](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol) we could add support.

Zerophase, 2016-08-29:
When I get the free time I'll hunt down the codes.

terrycain, 2016-08-29:
:-)

Spitfire1900, 2016-08-31:
:thumbsup:

Spitfire1900, 2016-09-08:
I can help out if possible. I haven't worked with Linux drivers before but I do own a Razer Anasi that I have attached to a Kubuntu 16.04.1 box.

terrycain, 2016-09-08:
Sure, same as above

Zerophase, 2016-09-09:
I'll most likely have the time in a couple days.

terrycain, 2016-09-09:
Nice

Zerophase, 2016-10-06:
Odd, I can't seem to find a `Leftover Data` field in Wireshark. Definitely getting signals captured whenever I change the lights. I just don't see the figure mentioned in the tutorial.

terrycain, 2016-10-06:
Upload the dump, sometimes if wireshark sees some certain packet it doesnt display the leftover bit

Zerophase, 2016-10-06:
Alright here's the dump. I only see the anansi showing up once with `lsusb`. I don't know if it having two usb connections is the cause of the issue. The back lighting doesn't fully work if I only plug in one of the usb connectors.

[OfToBright_White.pcapng.zip](https://github.com/terrycain/razer-drivers/files/514960/OfToBright_White.pcapng.zip)

terrycain, 2016-10-07:
Cool I'll have a look at it later. And as for the 2 usb that shouldn't
matter and the reason it might not work without 2 could be 1 usb port on
its own doesn't give out enough amps

terrycain, 2016-10-09:
There is some packets there of interest. Its lost in the sea of everything else. Using a filter like this `(usb.data_len == 90) && (usb.endpoint_number.direction == 0)` could help.

Also if you expand the URB setup section of the packet dissection there will be a `Data Fragment` section, you can right click on that and apply it as a column. I could do with some more captures and screenshots of synapse if you can

Zerophase, 2016-11-20:
Finally have time to work on this again. 

Starting to get a feel for which commands notify the keyboard to change color.  Currently, it looks like a command gets sent to notify the keyboard it's time to change color, then a command arrives with the color and brightness, and finally two more commands come in. Not sure why two commands come at the end. I'm guessing one puts it back into typing mode, and the other does something else.

At this point I just need to record the data for a few more colors, and see what "turn keyboard off when monitor is off does."

Believe I should have this in the code base within a couple days.

terrycain, 2016-11-21:
Turning off when monitor is a software thing. Beware Razer synapse likes to be overkill in the commands and could send additional commands to set states that are already set like turning game mode off when game mode is already off.

Zerophase, 2016-11-21:
I know what a few of the states are. There's just a few additional commands I'll toss in if I can figure out what they are. 

Yeah, the daemon just sends the shut lights off command when the monitor turns off. I'll just get the basic functions working, and add more of the features over a few commits.

It definitely looks like the keyboard's hardware could support a pulsing feature if we just send different brightness values on a timer. For some reason Razer never bothered to put that in.

terrycain, 2016-11-21:
Yeah can you attach the packet captures here also so I can verify what you've done is good :wink: As most likely they will use the same protocol the rest of the razer devices use (or at least a subset of it)

Zerophase, 2016-12-08:
Currently, working on getting the Anansi specific commands in. The keyboard binds, and I can turn the lights off, but can't turn them back on for some reason.

This is an example command of what I'm sending:
>echo -n "054F" > matrix_brightness

Here's a spreadsheet of my captures. (Can't post .ods files)
[RazerData.xlsx](https://github.com/terrycain/razer-drivers/files/638112/RazerData.xlsx)

Zerophase, 2016-12-08:
Just got the brightness control working and separated the Anansi specific methods into a separate header.

terrycain, 2016-12-08:
Brightness takes a regular base10 integer between 0 and 255 :wink:

terrycain, 2016-12-08:
Will look at spreadsheet when I get to a PC

terrycain, 2016-12-08:
Dont suppose you can repost that as a text file, as it is now with all the columns but merge subcmd and params :wink:. Then I can easily match that will all the commands I already have as at least most of them will duplicate what I already have.

terrycain, 2016-12-08:
Good work isolating all the commands. Some of the misc ones looks like its turning game led or macro led off (will double chjeck when I get home). Bear in mind razer synapse is evil and spams tons of commands regardless if they need to be sent. Screen waking up and going to screensaver all it will do it enable/disable game/macro leds and just set brightness to 0 or whatever it was.

Zerophase, 2016-12-08:
Here's the txt file.
[RazerData.txt](https://github.com/terrycain/razer-drivers/files/640766/RazerData.txt)

terrycain, 2016-12-09:
Awesome. I'll compare properly when i get home tonight but looks like it uses standard protocol and all the functions will already be implemented

Zerophase, 2016-12-13:
I'm working on getting color switching working for the Anansi, I've run into only one slight problem with that. That's that `get_razer_report` uses different naming conventions than the reverse engineering tutorial. 
How does Num Params, Reserved, and Command line up with  `command_class`, `command_id`, and `data_size`. I'm guessing data_size is Num Params.

terrycain, 2016-12-14:
Ahh yeah basically that tutorial was done long ago. Since then I got a spec explaining the protocol format and adjusted the structure accordingly. If you fill the `razer_report` struct with your binary payload it'll fit exactly and therefore point out what is what ;-)

Sub Cmd has become part of the args array.
```
Num Params | Reserved | Command | Sub Cmd
data_size | command_class | command_id | args[0]
```

Zerophase, 2016-12-14:
Thanks. That's so much easier than guessing and checking. It looks like the Anansi takes additional variables that aren't needed for the other keyboards. I'm going to get this all working, and then see if I can refactor to merge some of the Anansi methods and Chroma methods.

Zerophase, 2016-12-14:
Spectrum cycling, and setting static colors now work. All that's left is the macro keys and finishing up the daemon.

Zerophase, 2016-12-15:
Alright, I'm testing the macros, and I don't see a message sending the key combos. Just 
`Start      ID         Reserved   Num Params  Reserved   Command    Sub Cmd / Parameter`
`00         FF          000000             04               00               08                  00000000`

terrycain, 2016-12-15:
I'll have a look when I get home but I think thats one of the packets I see on all the keyboards that i just ignore :wink:

terrycain, 2016-12-15:
Here you go, this might help.
[info.txt](https://github.com/terrycain/razer-drivers/files/655850/info.txt)

Zerophase, 2016-12-15:
Ok, getting the macro led to turn on makes sense. I think I have the code for that. I'm just trying to figure out how programming the macro buttons for different macros works.

Zerophase, 2016-12-16:
So, I've gotten the macro led to turn on and I can get it to flash, but it takes multiple commands.

The numbers that fire off when the led starts flashing are:
> 00:ff:00:00:00:03:03:02:00:07:01
> 00:ff:00:00:00:04:03:04:00:07:05:05
> 00:ff:00:00:00:03:03:00:00:07:01

This is the code I'm using in `razer_attr_write_mode_macro_effect`. I sends those numbers in order, excpet for the last one. (If I send the last number it adds a delay for when flashing starts)
```c
case USB_DEVICE_ID_RAZER_ANANSI:
		    report = razer_anansi_set_led_effect(NOSTORE, MACRO_LED, enabled);
		    razer_send_payload(usb_dev, &report);
		    
		    report = razer_anansi_set_macro_ready_to_flash();
```

I just wanted to check if it's ok if I send the payload multiple times for a method used in a device file.

terrycain, 2016-12-16:
yep you need to ha. Normally what I do is set it to flashing, then enable the led but the order technically doesnt matter

Zerophase, 2016-12-16:
I think I have pretty much everything ready except for the daemon. I installed the daemon on Arch with `make daemon_install`, and starting it with `/usr/bin/razer-service -v -F`. I believe I added support in all the necessary places in the daemon for Anansi.

stack trace for the service:

```
Traceback (most recent call last):
  File "/usr/bin/razer-service", line 67, in <module>
    run()
  File "/usr/bin/razer-service", line 61, in run
    daemonize(**daemon_args)
  File "/usr/lib/python3.5/site-packages/razer_daemon/daemon.py", line 114, in daemonize
    daemon = RazerDaemon(verbose, log_dir, console_log, run_dir, config_file, test_dir=test_dir)
  File "/usr/lib/python3.5/site-packages/razer_daemon/daemon.py", line 206, in __init__
    self._load_devices(first_run=True)
  File "/usr/lib/python3.5/site-packages/razer_daemon/daemon.py", line 323, in _load_devices
    classes = razer_daemon.hardware.get_device_classes()
  File "/usr/lib/python3.5/site-packages/razer_daemon/hardware/__init__.py", line 24, in get_device_classes
    imported_module = __import__(hw_module, globals=globals(), locals=locals(), fromlist=['*'], level=0)
ImportError: No module named 'razer_daemon.hardware.'
2016-12-16 14:09:47 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-12-16 14:09:47 | razer.screensaver              | ERROR    | Caught exception in run loop
Traceback (most recent call last):
  File "/usr/lib/python3.5/site-packages/razer_daemon/misc/screensaver_thread.py", line 123, in run
    screensaver_active = self._dbus_interface.GetActive()
  File "/usr/lib/python3.5/site-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python3.5/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3.5/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NotSupported: This method is not implemented
```

stack trace for razercommander:
```
ERROR:dbus.proxies:Introspect error on :1.71:/org/razer: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send areply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
DEBUG:dbus.proxies:Executing introspect queue due to error
Traceback (most recent call last):
  File "/usr/share/razercommander/main.py", line 7, in <module>
    import device
  File "/usr/share/razercommander/device.py", line 10, in <module>
    device_manager = rclient.DeviceManager()
  File "/usr/lib/python3.5/site-packages/razer/client/__init__.py", line 27, in __init__
    self._device_serials = self._dbus_devices.getDevices()
  File "/usr/lib/python3.5/site-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python3.5/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3.5/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NoReply: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.

```
#108

Zerophase, 2016-12-18:
So, I just forgot to install the udev rules. I'm supposed to run the daemon as a user, right?

I keep getting 
`PermissionError: [Errno 13] Permission denied: '/sys/bus/hid/devices/0003:1532:010F.0003/device_serial'`

My user account is in the `plugdev` group. shouldn't that let me access the device?

terrycain, 2016-12-18:
You run the daemon as you. have the files in /sys/bus/hid/devices/0003:1532:010F.0003/ been owned by plugdev?

Zerophase, 2016-12-18:
I just added those files to plugdev. I can get razercommander to open, and it lets me change some of the settings. I just need to figure out why only individual key custom colors are coming up. (keys aren't individually assignable) I should be seeing the static and spectrum settings.

z3ntu, 2016-12-19:
That's a problem with razercommander, report [here](https://github.com/GabMus/razercommander).

Zerophase, 2016-12-19:
@z3ntu Filled out a couple reports. The latest version in master broke a couple things, and was listing Custom for every device other than mice.

How does the none effect work for the other keyboards? Is it just turning off brightness for all of the keys, and indicator lights? (game mode, macro mode lights, etc) I believe trying to get that working corrupted my kernel.

terrycain, 2016-12-19:
Yeah that's a bug am fixing.

Nope. None and brightness are separate.

Zerophase, 2016-12-20:
I'm just trying to figure out if this keyboard needs none. When the lights get shut off when the screen turns off 6 codes get sent. I tried sending all of the brightness settings (minus the ones that change the light being disabled) to 'matrix_effect_none`, and some how it made my kernel unbootable. 

[image]

terrycain, 2016-12-20:
pretty sure just the 1st, 3rd and 5th line are needed which disable the LED. the others set the brightness to 0 but just disabling the LED should be enough.

Zerophase, 2016-12-20:
Yeah, that's true. Hmm, I didn't think about it that way. I've just noticed since working through this Razer's keyboard design became more eloquent in terms of the firmware.

Zerophase, 2016-12-20:
Cool, I accidentally found a pulsating led mode. Way too fest for it to not be annoying.

terrycain, 2016-12-20:
Yeah they actually did a good job on the firmware once you know how it works ;-). And yeah I've found that they've left in blinky-blinky mode on a large number of devices though its classed as "not supported"

Zerophase, 2016-12-20:
And, I can't figure out how to turn blinky-blinky mode back on.  Since for some of the effects require multiple commands. (the static effect needs to change the backlight color, and turn the light back on) Would you prefer separate device files for each command, and just have the daemon be responsible for updating each relevant device file?

terrycain, 2016-12-20:
nah one device file per effect
And blinking would be calling set_effect with effect 0x01

terrycain, 2016-12-20:
> nah one device file per effect

As currently we have files for static, spectrum etc... so it would be best to just present the same interface as that (ideally calling the same methods that already exist :wink:)

Zerophase, 2016-12-20:
Yeah, I can call most of the methods that already exist. The only thing is if I set the keyboard to the none effect, when I go back to static I have to make sure to reenable the lights on top of changing colors.

terrycain, 2016-12-20:
Then always call the led state = true method whenever your setting static :wink:

terrycain, 2016-12-20:
Do a synapse and spam all the commands as you like as long as it fits in with the current structure (as if it don't fit in, ill make it fit in :wink:)

Zerophase, 2016-12-21:
Well, I believe it works correctly.

terrycain, 2016-12-21:
Few changes then its good to go. Good job :smile:

terrycain, 2017-01-16:
Merged a while ago

closed 2017-01-16 by terrycain