openrazer/openrazer #33: Crash after installation

claell, 2016-08-29:
I just rebooted after installation of the daemon and kernel modules (via PPA) on Ubuntu 16.04. Now I got a crash report from Razer Service "razer-service crashed with ImportError in [...] No module named 'razer.keyboard'"

terrycain, 2016-08-29:
is python3-razer intalled?

claell, 2016-08-30:
I think not, but I cannot seem to be able to install it via `apt-get install python3-razer`

terrycain, 2016-08-30:
Really? As I installed it on an xenial server (just to test) and it was fine. I'll have a look when I get home, probably LaunchTurd bugging out again. Also what version do you have installed?

terrycain, 2016-08-30:
[pastebin](http://pastebin.com/nRCUdmqS) of installing python3-razer. Should just work on 16.04 unless you have the wrong ppa but I dont see how

claell, 2016-08-31:
I probably have a wrong PPA. I used the one from Polychromatic. Did not know there are two.

GabMus, 2016-08-31:
The driver ppa is separated from polychromatic ppa, yeah

claell, 2016-08-31:
Ok. I will try the other PPA when I am back at my machine (in a week or two).

terrycain, 2016-08-31:
Ahh yeah correct me if im wrong @lah7, originally he had a PPA for it all but now there will be 1 PPA for the core stuff and then a PPA for each client application.

lah7, 2016-08-31:
The PPA for Polychromatic was **just** for Polychromatic. The README did include a separate Debian package for `amd64` - to save the convenience of needing to compile it.

claell, 2016-09-07:
Works with correct PPA.

closed 2016-09-07 by claell