openrazer/openrazer #38: Razer Naga and Tartarus

ArtoriasEdgeworth, 2016-09-04:
The Razer products Naga and Tartarus aren't listed under the supported group.
Hate to be the guy to ask for everything, but if these were made compatible, it would be very greatly appreciated!!

ArtoriasEdgeworth, 2016-09-04:
My mistake, Orbweaver, not Tartarus. And I just noticed that there is a thread dedicated to both.
Very sorry!

terrycain, 2016-09-05:
No problem. It happens.

terrycain, 2016-09-05:
There isn't an issue for Naga. You willing to help add support for it?

ArtoriasEdgeworth, 2016-09-08:
I'm a complete inept when it comes to programming, save for basic RobotC.
I'd love to, but I don't think I'd be of much use.

-Nathan A. Johnson

closed 2016-09-08 by terrycain

claell, 2016-09-09:
You don't need to write a single line of code. You just need to capture some packages with Wireshark. There is also a tutorial. Probably you can do that.

ArtoriasEdgeworth, 2016-09-09:
I can indeed.

-Nathan A. Johnson