openrazer/openrazer #40: Sync devices problem

cravacuore, 2016-09-07:
Hi, I'm trying to sync a new device (Razer Firefly) which is fully supported with my keyboard (Razer BlackWidow Tournament Edition Chroma) also fully supported.

I've 'sync_effects_enabled' with value True at config file (.razer-service/razer.conf). By this I understand that any effect I apply to one of the devices is going to match automatically with the other.

Maybe I'm doing something wrong, is there any step I'm missing?

Thanks in advance for any help you can bring me.

terrycain, 2016-09-07:
Send me log file

cravacuore, 2016-09-07:
> 2016-09-07 16:03:27 | razer                          | INFO     | Initialising Daemon. Pid: 4809
2016-09-07 16:03:27 | razer.screensaver              | INFO     | Initialising DBus Screensaver Thread
2016-09-07 16:03:27 | razer                          | INFO     | Found device.0: 0003:1532:0C00.0012
2016-09-07 16:03:27 | razer.device0                  | INFO     | Initialising device.0 RazerFireFly
2016-09-07 16:03:27 | razer                          | INFO     | Found device.1: 0003:1532:0209.001A
2016-09-07 16:03:27 | razer.device1                  | INFO     | Initialising device.1 RazerBlackWidowChromaTournamentEdition
2016-09-07 16:03:27 | razer                          | INFO     | Adding razer.devices.getDevices method to DBus
2016-09-07 16:03:27 | razer                          | INFO     | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2016-09-07 16:03:27 | razer                          | INFO     | Adding razer.devices.disableTurnOffOnScreensaver method to DBus
2016-09-07 16:03:27 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
2016-09-07 16:03:27 | razer                          | INFO     | Adding razer.daemon.stop method to DBus
2016-09-07 16:03:27 | razer                          | INFO     | Serving DBus
2016-09-07 16:03:29 | razer.screensaver              | INFO     | Initialising DBus screensaver object

terrycain, 2016-09-07:
Where it says
```
Adding razer.devices.syncEffects method to DBus
```
It should say a lot more for both devices.

What os you running? And are you using the latest packages?

cravacuore, 2016-09-07:
I'm running ArchLinux. Was using v1.0.0 release, but now I switch to AUR package (razer-driver-meta).

terrycain, 2016-09-07:
@GabMus would be the best to speak to here

terrycain, 2016-09-07:
Wait so was that log from 1.0.0?

cravacuore, 2016-09-07:
My bad, the lack of lines on the log file was because I set `verbose_logging` to False before.

Here complete log:
```
2016-09-07 18:19:27 | razer                          | INFO     | Initialising Daemon. Pid: 8635
2016-09-07 18:19:27 | razer.screensaver              | INFO     | Initialising DBus Screensaver Thread
2016-09-07 18:19:27 | razer                          | INFO     | Found device.0: 0003:1532:0C00.0012
2016-09-07 18:19:27 | razer.device0                  | INFO     | Initialising device.0 RazerFireFly
2016-09-07 18:19:27 | razer.device0                  | DEBUG    | Adding getSerial method to DBus
2016-09-07 18:19:27 | razer.device0                  | DEBUG    | Adding razer.device.misc.suspendDevice method to DBus
2016-09-07 18:19:27 | razer.device0                  | DEBUG    | Adding razer.device.misc.resumeDevice method to DBus
2016-09-07 18:19:27 | razer.device0                  | DEBUG    | Adding razer.device.misc.getVidPid method to DBus
2016-09-07 18:19:27 | razer.device0                  | DEBUG    | Adding razer.device.misc.getFirmware method to DBus
2016-09-07 18:19:27 | razer.device0                  | DEBUG    | Adding razer.device.misc.getMatrixDimensions method to DBus
2016-09-07 18:19:27 | razer.device0                  | DEBUG    | Adding razer.device.misc.hasMatrix method to DBus
2016-09-07 18:19:27 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceName method to DBus
2016-09-07 18:19:27 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceType method to DBus
2016-09-07 18:19:27 | razer.device0                  | DEBUG    | Adding razer.device.lighting.brightness.getBrightness method to DBus
2016-09-07 18:19:27 | razer.device0                  | DEBUG    | Adding razer.device.lighting.brightness.setBrightness method to DBus
2016-09-07 18:19:27 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setWave method to DBus
2016-09-07 18:19:27 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setStatic method to DBus
2016-09-07 18:19:27 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setSpectrum method to DBus
2016-09-07 18:19:27 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setReactive method to DBus
2016-09-07 18:19:27 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setNone method to DBus
2016-09-07 18:19:27 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setBreathRandom method to DBus
2016-09-07 18:19:27 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setBreathSingle method to DBus
2016-09-07 18:19:27 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setBreathDual method to DBus
2016-09-07 18:19:27 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setCustom method to DBus
2016-09-07 18:19:27 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setKeyRow method to DBus
2016-09-07 18:19:27 | razer                          | INFO     | Found device.1: 0003:1532:0209.001A
2016-09-07 18:19:27 | razer.device1                  | INFO     | Initialising device.1 RazerBlackWidowChromaTournamentEdition
2016-09-07 18:19:27 | razer.device1                  | DEBUG    | Adding getSerial method to DBus
2016-09-07 18:19:27 | razer.device1                  | DEBUG    | Adding razer.device.misc.suspendDevice method to DBus
2016-09-07 18:19:27 | razer.device1                  | DEBUG    | Adding razer.device.misc.resumeDevice method to DBus
2016-09-07 18:19:27 | razer.device1                  | DEBUG    | Adding razer.device.misc.getVidPid method to DBus
2016-09-07 18:19:27 | razer.device1                  | DEBUG    | Adding razer.device.misc.getFirmware method to DBus
2016-09-07 18:19:27 | razer.device1                  | DEBUG    | Adding razer.device.misc.getMatrixDimensions method to DBus
2016-09-07 18:19:27 | razer.device1                  | DEBUG    | Adding razer.device.misc.hasMatrix method to DBus
2016-09-07 18:19:27 | razer.device1                  | DEBUG    | Adding razer.device.misc.getDeviceName method to DBus
2016-09-07 18:19:27 | razer.device1                  | DEBUG    | Adding razer.device.misc.getDeviceType method to DBus
2016-09-07 18:19:27 | razer.device1                  | DEBUG    | Adding razer.device.lighting.brightness.getBrightness method to DBus
2016-09-07 18:19:27 | razer.device1                  | DEBUG    | Adding razer.device.lighting.brightness.setBrightness method to DBus
2016-09-07 18:19:27 | razer.device1                  | DEBUG    | Adding razer.device.lighting.chroma.setWave method to DBus
2016-09-07 18:19:27 | razer.device1                  | DEBUG    | Adding razer.device.lighting.chroma.setStatic method to DBus
2016-09-07 18:19:27 | razer.device1                  | DEBUG    | Adding razer.device.lighting.chroma.setSpectrum method to DBus
2016-09-07 18:19:27 | razer.device1                  | DEBUG    | Adding razer.device.lighting.chroma.setReactive method to DBus
2016-09-07 18:19:27 | razer.device1                  | DEBUG    | Adding razer.device.lighting.chroma.setNone method to DBus
2016-09-07 18:19:27 | razer.device1                  | DEBUG    | Adding razer.device.lighting.chroma.setBreathRandom method to DBus
[... cut ...]
2016-09-07 18:19:27 | razer                          | INFO     | Serving DBus
2016-09-07 18:19:29 | razer.screensaver              | INFO     | Initialising DBus screensaver object
```

Also before from both GUI apps (razerCommander / Polychromatic) sync wasn't working but now it does, at least for Wave and Spectrum modes.

How can I set devices mode respecting my sync preference from terminal?

Setting them from /sys/bus/hid/drivers/«device»/«id»/ with `echo -n "1" > «mode»` does not respect sync. Maybe this is not the correct way.

Thanks.

terrycain, 2016-09-07:
Static, wave, breathing, spectrum should work via sync. Interacting with the driver manually is not a good idea, that defeats the entire purpose of the daemon.

Soon there will be some cli type thing to do it. Am working on it but have higher priority things to deal with.

Have a look at [this](https://github.com/terrycain/razer-drivers/blob/master/scripts/daemon/controller.py) script. run it with `./controller.py --help` or `./controller.py list`, is a work in progress and far from finished.

terrycain, 2016-09-07:
And the `~/.razer-service/razer.conf` doesnt get used yet I dont think.

cravacuore, 2016-09-07:
Great. Thank you very much. I'll close this.

closed 2016-09-07 by cravacuore