openrazer/openrazer #42: daemon not working

xeq, 2016-09-10:


xeq, 2016-09-10:
Well this daemon is not working at all, it suddenly crashed when I changed my workstation name to something else (i'm on latest Ubuntu)
I got an error saying it crashed. 

2016-09-09 20:59:38 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-09-10 11:52:59 | razer                          | INFO     | Initialising Daemon. Pid: 4518
2016-09-10 11:53:00 | razer.screensaver              | INFO     | Initialising DBus Screensaver Thread
2016-09-10 11:53:00 | razer                          | INFO     | Adding razer.devices.getDevices method to DBus
2016-09-10 11:53:00 | razer                          | INFO     | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2016-09-10 11:53:00 | razer                          | INFO     | Adding razer.devices.disableTurnOffOnScreensaver method to DBus
2016-09-10 11:53:00 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
2016-09-10 11:53:00 | razer                          | INFO     | Adding razer.daemon.stop method to DBus
2016-09-10 11:53:00 | razer                          | INFO     | Serving DBus
2016-09-10 11:53:02 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-09-10 11:53:02 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-09-10 11:53:02 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-09-10 11:53:02 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-09-10 11:53:02 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-09-10 11:53:02 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-09-10 11:53:02 | razer.screensaver              | WARNING  | Giving up on screensaver
2016-09-10 12:31:27 | razer                          | INFO     | Initialising Daemon. Pid: 4561
2016-09-10 12:31:27 | razer.screensaver              | INFO     | Initialising DBus Screensaver Thread
2016-09-10 12:31:27 | razer                          | INFO     | Found device.0: 0003:1532:0203.0003
2016-09-10 12:31:27 | razer.device0                  | INFO     | Initialising device.0 RazerBlackWidowChroma
2016-09-10 12:31:27 | razer.device0                  | DEBUG    | Adding getSerial method to DBus
2016-09-10 12:31:27 | razer.device0                  | DEBUG    | Adding razer.device.misc.suspendDevice method to DBus
2016-09-10 12:31:27 | razer.device0                  | DEBUG    | Adding razer.device.misc.resumeDevice method to DBus
2016-09-10 12:31:27 | razer.device0                  | DEBUG    | Adding razer.device.misc.getVidPid method to DBus
2016-09-10 12:31:27 | razer.device0                  | DEBUG    | Adding razer.device.misc.getFirmware method to DBus
2016-09-10 12:31:27 | razer.device0                  | DEBUG    | Adding razer.device.misc.getMatrixDimensions method to DBus
2016-09-10 12:31:27 | razer.device0                  | DEBUG    | Adding razer.device.misc.hasMatrix method to DBus
2016-09-10 12:31:27 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceName method to DBus
2016-09-10 12:31:27 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceType method to DBus
2016-09-10 12:31:27 | razer.device0                  | DEBUG    | Adding razer.device.lighting.brightness.getBrightness method to DBus
2016-09-10 12:31:27 | razer.device0                  | DEBUG    | Adding razer.device.lighting.brightness.setBrightness method to DBus
2016-09-10 12:31:27 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setWave method to DBus
2016-09-10 12:31:27 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setStatic method to DBus
2016-09-10 12:31:27 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setSpectrum method to DBus
2016-09-10 12:31:27 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setReactive method to DBus
2016-09-10 12:31:27 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setNone method to DBus
2016-09-10 12:31:27 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setBreathRandom method to DBus
2016-09-10 12:31:27 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setBreathSingle method to DBus
2016-09-10 12:31:27 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setBreathDual method to DBus
2016-09-10 12:31:27 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setCustom method to DBus
2016-09-10 12:31:27 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setKeyRow method to DBus
2016-09-10 12:31:27 | razer.device0                  | DEBUG    | Adding razer.device.macro.enableMacroKeys method to DBus
2016-09-10 12:31:27 | razer.device0                  | DEBUG    | Adding razer.device.led.gamemode.getGameMode method to DBus
2016-09-10 12:31:27 | razer.device0                  | DEBUG    | Adding razer.device.led.gamemode.setGameMode method to DBus
2016-09-10 12:31:27 | razer.device0                  | DEBUG    | Adding razer.device.led.macromode.getMacroMode method to DBus
2016-09-10 12:31:27 | razer.device0                  | DEBUG    | Adding razer.device.led.macromode.setMacroMode method to DBus
[... cut ...]
2016-09-10 13:05:57 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-09-10 13:05:57 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-09-10 13:05:57 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-09-10 13:05:57 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-09-10 13:05:57 | razer.screensaver              | WARNING  | Giving up on screensaver

Here is my razer.log
The daemon is not working even after I turn it on manually by using razer-service

xeq, 2016-09-10:
and in the crash it says no module named razer.keyboard that's why it crashed

claell, 2016-09-10:
You need to install python3-razer.

terrycain, 2016-09-10:
If you use the latest 1.0.8 i think, it should be fine, fixed that a while ago I believe.

rudeb0t, 2016-09-27:
I have the Razer Blackwidow X Chroma keyboard along with another wireless keyboard/mouse because my rig is spread across two desks. I am experiencing similar issue with Ubuntu 16.04. I installed via PPA as directed:

```
 sudo add-apt-repository ppa:terrz/razerutils
 sudo apt update
 sudo apt install python3-razer razer-kernel-modules-dkms razer-daemon razer-doc
```

I restarted. Then installed razerCommander from the .deb provided here: https://github.com/GabMus/razerCommander but when I run it I get the following output:

```
$ razercommander 
no devices
```

The `razerdaemon` seems to be running:

```
$ ps aux|grep '[r]'azer
rudeb0t    3466  0.2  0.2 346892 30996 ?        Sl   13:28   0:08 razerdaemon
```

`lsmod` output is below:

```
$ lsmod|grep razer
razerkbd               32768  0
hid                   118784  3 hid_generic,razerkbd,usbhid
```

Contents of `/sys/bus/hid/drivers/razerkbd`:

```
# rudeb0t @ eniwetok in /sys/bus/hid/drivers/razerkbd [14:20:25] 
$ ls -l
total 0
--w------- 1 root root 4096 Sep 27 13:44 bind
lrwxrwxrwx 1 root root    0 Sep 27 13:53 module -> ../../../../module/razerkbd
--w------- 1 root root 4096 Sep 27 13:29 new_id
--w------- 1 root root 4096 Sep 27 13:27 uevent
--w------- 1 root root 4096 Sep 27 13:29 unbind
```

I had to manually bind the device by following the instructions here: https://github.com/terrycain/razer-drivers/wiki/Setting-up-the-keyboard-driver to be able to change the fancy LED effects. But razerCommander still does not work because the device list it gets from the daemon is empty.

For due diligence I tried unplugging my other keyboard/mouse and just left with the Razer keyboard and then rebooted. I am still having the same issue, so this rules out the possibility that my other keyboard/mouse is interfering with the driver/daemon.

terrycain, 2016-09-27:
Output of `~/.razer-service/logs`, its in a folder like that I think. Output of `lsusb`. Output of `/tmp/razer_something.log`

rudeb0t, 2016-09-27:
> output of lsusb

```
$ lsusb
Bus 002 Device 004: ID 1532:0216 Razer USA, Ltd 
Bus 002 Device 003: ID 279e:024f  
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

> output of `~/.razer-service/logs/razer.log`

```
$ cat .razer-service/logs/razer.log 
2016-09-27 12:40:41 | razer                          | INFO     | Initialising Daemon (v1.0.11). Pid: 3475
2016-09-27 12:40:41 | razer.screensaver              | INFO     | Initialising DBus Screensaver Thread
2016-09-27 12:40:41 | razer                          | INFO     | Adding razer.devices.getDevices method to DBus
2016-09-27 12:40:41 | razer                          | INFO     | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2016-09-27 12:40:41 | razer                          | INFO     | Adding razer.devices.disableTurnOffOnScreensaver method to DBus
2016-09-27 12:40:41 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
2016-09-27 12:40:41 | razer                          | INFO     | Adding razer.daemon.version method to DBus
2016-09-27 12:40:41 | razer                          | INFO     | Adding razer.daemon.stop method to DBus
2016-09-27 12:40:41 | razer                          | INFO     | Serving DBus
2016-09-27 12:40:43 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-09-27 12:40:43 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-09-27 12:40:43 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-09-27 12:40:43 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-09-27 12:40:44 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-09-27 12:40:44 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-09-27 12:40:44 | razer.screensaver              | WARNING  | Giving up on screensaver
2016-09-27 12:46:26 | razer                          | DEBUG    | DBus called get_serial_list
2016-09-27 12:46:54 | razer                          | DEBUG    | DBus called get_serial_list
2016-09-27 12:51:46 | razer                          | DEBUG    | DBus called get_serial_list
2016-09-27 12:52:17 | razer                          | DEBUG    | DBus called get_serial_list
2016-09-27 13:03:33 | razer                          | DEBUG    | DBus called get_serial_list
2016-09-27 13:05:11 | razer                          | DEBUG    | DBus called get_serial_list
2016-09-27 13:08:56 | razer                          | DEBUG    | DBus called get_serial_list
2016-09-27 13:23:16 | razer                          | DEBUG    | DBus called get_serial_list
2016-09-27 13:28:33 | razer                          | INFO     | Initialising Daemon (v1.0.11). Pid: 3466
2016-09-27 13:28:33 | razer.screensaver              | INFO     | Initialising DBus Screensaver Thread
2016-09-27 13:28:33 | razer                          | INFO     | Adding razer.devices.getDevices method to DBus
2016-09-27 13:28:33 | razer                          | INFO     | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2016-09-27 13:28:33 | razer                          | INFO     | Adding razer.devices.disableTurnOffOnScreensaver method to DBus
2016-09-27 13:28:33 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
2016-09-27 13:28:33 | razer                          | INFO     | Adding razer.daemon.version method to DBus
2016-09-27 13:28:33 | razer                          | INFO     | Adding razer.daemon.stop method to DBus
2016-09-27 13:28:33 | razer                          | INFO     | Serving DBus
2016-09-27 13:28:35 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-09-27 13:28:35 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-09-27 13:28:35 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-09-27 13:28:35 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-09-27 13:28:35 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-09-27 13:35:31 | razer                          | DEBUG    | DBus called get_serial_list
2016-09-27 14:23:36 | razer                          | DEBUG    | DBus called get_serial_list
2016-09-27 15:10:20 | razer.screensaver              | INFO     | Suspend screensaver
2016-09-27 15:13:44 | razer.screensaver              | INFO     | Resume screensaver
2016-09-27 16:38:20 | razer.screensaver              | INFO     | Suspend screensaver
2016-09-27 16:38:28 | razer.screensaver              | INFO     | Resume screensaver
2016-09-27 17:47:38 | razer.screensaver              | ERROR    | Caught exception in run loop
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer_daemon/misc/screensaver_thread.py", line 122, in run
    screensaver_active = self._dbus_interface.GetActive()
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
[... cut ...]
2016-09-27 19:50:28 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-09-27 19:50:28 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-09-27 19:50:28 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-09-27 19:50:28 | razer.screensaver              | WARNING  | Giving up on screensaver
```

I don't see any `/tmp/razer_something.log` but there is an empty directory `/tmp/tmp_pwach33__razer_daemon`

terrycain, 2016-09-27:
Ok good, you not seeing the razer log in /tmp means udev isnt doing its thing. Which after looking, have found I've missed out the line for the Bw X Chroma

Ok in `/lib/udev/rules.d/99-razer.rules` there will be the following line
```
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0214", ENV{is_razer_kbd}="yes"
```
You want to add this line after it
```
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0216", ENV{is_razer_kbd}="yes"
```

Then save it, reboot and see if `/tmp/razer_udev.log` appears, and more output should appear in the daemon service log and in theory `razercommander` should work.

If that all works, I'll add it to the codebase and it'll be out in the next release.

SpainTrain, 2016-09-28:
Hi, similar issue here with a BlackWidow Tournament Edition Stealth (RZ03-00811700-R3M1).

I followed the directions and added the relevant ID to `/lib/udev/rules.d/99-razer.rules`:

```bash
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="011c", ENV{is_razer_kbd}="yes"
```

However `razer_mount razerkbd` exits with non-zero exit code and I cannot figure out how to get more logging to determine why.  I have pasted relevant logs and output below:

###### syslog

```
Sep 28 14:14:15 spainbuntu2 gnome-session[2645]: 2016-09-28 14:14:15 | razer                          | INFO     | Initialising Daemon (v1.0.12). Pid: 2817
Sep 28 14:14:15 spainbuntu2 gnome-session[2645]: 2016-09-28 14:14:15 | razer.screensaver              | INFO     | Initialising DBus Screensaver Thread
Sep 28 14:14:15 spainbuntu2 gnome-session[2645]: 2016-09-28 14:14:15 | razer                          | INFO     | Adding razer.devices.getDevices method to DBus
Sep 28 14:14:15 spainbuntu2 gnome-session[2645]: 2016-09-28 14:14:15 | razer                          | INFO     | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
Sep 28 14:14:15 spainbuntu2 gnome-session[2645]: 2016-09-28 14:14:15 | razer                          | INFO     | Adding razer.devices.disableTurnOffOnScreensaver method to DBus
Sep 28 14:14:15 spainbuntu2 gnome-session[2645]: 2016-09-28 14:14:15 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
Sep 28 14:14:15 spainbuntu2 gnome-session[2645]: 2016-09-28 14:14:15 | razer                          | INFO     | Adding razer.daemon.version method to DBus
Sep 28 14:14:15 spainbuntu2 gnome-session[2645]: 2016-09-28 14:14:15 | razer                          | INFO     | Adding razer.daemon.stop method to DBus
Sep 28 14:14:15 spainbuntu2 gnome-session[2645]: 2016-09-28 14:14:15 | razer                          | INFO     | Serving DBus
Sep 28 14:14:17 spainbuntu2 gnome-session[2645]: 2016-09-28 14:14:17 | razer.screensaver              | INFO     | Initialising DBus screensaver object
Sep 28 14:16:24 spainbuntu2 gnome-session[2645]: 2016-09-28 14:16:24 | razer                          | DEBUG    | DBus called get_serial_list
Sep 28 14:17:49 spainbuntu2 systemd-udevd[11068]: Process 'razer_mount razerkbd 0003:1532:011C.000A' failed with exit code 1.
Sep 28 14:17:49 spainbuntu2 systemd-udevd[11045]: Process 'razer_mount razerkbd 0003:1532:011C.0008' failed with exit code 1.
Sep 28 14:17:49 spainbuntu2 systemd-udevd[11066]: Process 'razer_mount razerkbd 0003:1532:011C.0009' failed with exit code 1.
Sep 28 14:54:38 spainbuntu2 systemd-udevd[12741]: Process 'razer_mount razerkbd 0003:1532:011C.000B' failed with exit code 1.
Sep 28 14:54:38 spainbuntu2 systemd-udevd[12742]: Process 'razer_mount razerkbd 0003:1532:011C.000C' failed with exit code 1.
Sep 28 14:54:38 spainbuntu2 systemd-udevd[12720]: Process 'razer_mount razerkbd 0003:1532:011C.000D' failed with exit code 1.

```

###### /tmp/razer_udev.log 

```
Driver razerkbd
Driver razerkbd
Device_ID 0003:1532:011C.000B
Device_ID 0003:1532:011C.000C
Unbinding 0003:1532:011C.000B from hid-generic
Unbinding 0003:1532:011C.000C from hid-generic
Driver razerkbd
Device_ID 0003:1532:011C.000D
Unbinding 0003:1532:011C.000D from hid-generic
Binding 0003:1532:011C.000B to razerkbd
Binding 0003:1532:011C.000C to razerkbd
Binding 0003:1532:011C.000D to razerkbd
Driver razerkbd
Device_ID 0003:1532:011C.000B
Driver razerkbd
Device_ID 0003:1532:011C.000C
Driver razerkbd
Device_ID 0003:1532:011C.000D
```

###### ~/.razer-service/logs/razer.log 

```
2016-09-28 14:14:15 | razer                          | INFO     | Initialising Daemon (v1.0.12). Pid: 2817
2016-09-28 14:14:15 | razer.screensaver              | INFO     | Initialising DBus Screensaver Thread
2016-09-28 14:14:15 | razer                          | INFO     | Adding razer.devices.getDevices method to DBus
2016-09-28 14:14:15 | razer                          | INFO     | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2016-09-28 14:14:15 | razer                          | INFO     | Adding razer.devices.disableTurnOffOnScreensaver method to DBus
2016-09-28 14:14:15 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
2016-09-28 14:14:15 | razer                          | INFO     | Adding razer.daemon.version method to DBus
2016-09-28 14:14:15 | razer                          | INFO     | Adding razer.daemon.stop method to DBus
2016-09-28 14:14:15 | razer                          | INFO     | Serving DBus
2016-09-28 14:14:17 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-09-28 14:16:24 | razer                          | DEBUG    | DBus called get_serial_list
```

Any suggestions on how to figure out why this isn't working are much appreciated :smile_cat:

terrycain, 2016-09-28:
BlackWidow Tournament Edition Stealth isnt supported by daemon or driver, open a new issue

SpainTrain, 2016-09-28:
[image]

Hah, thanks.  Done - https://github.com/terrycain/razer-drivers/issues/55

rudeb0t, 2016-09-29:
Just getting back to this now. I added the line to `/lib/udev/rules.d/99-razer.rules` and restarted. I still don't get anything in the `/tmp/` directory.

```
$ ls /tmp                            
config-err-8mn3sc                                                       systemd-private-4ea1565a94694bedb4e5afa0f2fb167a-rtkit-daemon.service-LbeoG9       tmp_v3d4rf3m_razer_daemon
systemd-private-4ea1565a94694bedb4e5afa0f2fb167a-colord.service-M1FgtP  systemd-private-4ea1565a94694bedb4e5afa0f2fb167a-systemd-timesyncd.service-dImDPY  unity_support_test.0

$ ls -a /tmp/tmp_v3d4rf3m_razer_daemon
.  ..

$ tree ~/.razer-service 
/home/rudeb0t/.razer-service
├── daemon.pid
├── data
├── logs
│   └── razer.log
└── razer.conf

2 directories, 3 files

```

I got the following in my `/var/log/syslog`

```
Sep 29 09:37:02 eniwetok kernel: [   27.624202] razerkbd 0003:1532:0216.0005: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer BlackWidow X Chroma] on usb-0000:00:1d.0-1.2/input1
Sep 29 09:37:02 eniwetok kernel: [   27.624259] razerkbd 0003:1532:0216.0006: input,hidraw3: USB HID v1.11 Mouse [Razer Razer BlackWidow X Chroma] on usb-0000:00:1d.0-1.2/input2
Sep 29 09:37:02 eniwetok kernel: [   27.680225] razerkbd 0003:1532:0216.0004: input,hidraw4: USB HID v1.11 Keyboard [Razer Razer BlackWidow X Chroma] on usb-0000:00:1d.0-1.2/input0
```

So I tried unplugging this wireless mouse/keyboard that I have plugged in. Then I unplugged my Razer keyboard and then plugged it back in and hey presto! There is now a `/tmp/udev_razer.log`. razerCommander still says "no devices" though. I even `pkill razerdaemon` and restarted razer-service manually with all the command line options found in `/etc/xdg/autostart/razer-service.desktop`.

Further information, the mouse/keyboard that seems to be interfering with udev detection of my razer when I have both of them plugged in has the following ID as per lsusb:

```
Bus 002 Device 006: ID 279e:024f
```

rudeb0t, 2016-09-29:
Attached is the `/tmp/razer_udev.log`

[razer_udev.txt](https://github.com/terrycain/razer-drivers/files/499631/razer_udev.txt)

rudeb0t, 2016-09-29:
Update:

So I unplugged my other USB keyboard/mouse leaving just the Razer keyboard plugged in and rebooted. I still don't get the `/tmp/udev_razer.log` file. However when I unplug my Razer keyboard and plug it back in, the file appears. So it looks like I have to detach then re-attach my keyboard in order for `udev` to pick up and run razer_mount. Restarting the razer-daemon after this still causes razercommander to say "no devices".

terrycain, 2016-09-29:
First of all. Are you in the plugdev group?

rudeb0t, 2016-09-29:
> First of all. Are you in the plugdev group?

Yes. I just used all the defaults in this Ubuntu install and `groups` confirms I am in `plugdev` group. Not doing anything unusual except for having two keyboards attached to it.

terrycain, 2016-09-29:
Ok `tree /sys/bus/hid/drivers/ra*` Two keyboards shoudlnt affect anything. Also complete contents of `/lib/udev/rules.d/99-razer.rules`.

rudeb0t, 2016-09-30:
> tree /sys/bus/hid/drivers/ra*

```
$ tree /sys/bus/hid/drivers/ra* 
/sys/bus/hid/drivers/razerkbd
├── 0003:1532:0216.0004 -> ../../../../devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.0/0003:1532:0216.0004
├── 0003:1532:0216.0005 -> ../../../../devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.1/0003:1532:0216.0005
├── 0003:1532:0216.0006 -> ../../../../devices/pci0000:00/0000:00:1d.0/usb2/2-1/2-1.2/2-1.2:1.2/0003:1532:0216.0006
├── bind
├── module -> ../../../../module/razerkbd
├── new_id
├── uevent
└── unbind
```

> complete contents of /lib/udev/rules.d/99-razer.rules

```
# This seems to work
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="011b", ENV{is_razer_kbd}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="010d", ENV{is_razer_kbd}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="011a", ENV{is_razer_kbd}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0203", ENV{is_razer_kbd}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0205", ENV{is_razer_kbd}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0208", ENV{is_razer_kbd}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0209", ENV{is_razer_kbd}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0214", ENV{is_razer_kbd}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0216", ENV{is_razer_kbd}="yes"

KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0042", ENV{is_razer_mouse}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0044", ENV{is_razer_mouse}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0045", ENV{is_razer_mouse}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0046", ENV{is_razer_mouse}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0048", ENV{is_razer_mouse}="yes"
KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="002f", ENV{is_razer_mouse}="yes"

KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="0c00", ENV{is_razer_firefly}="yes"

SUBSYSTEM=="hid", DRIVER=="hid-generic", ENV{is_razer_kbd}=="yes", RUN+="razer_mount razerkbd $kernel"

SUBSYSTEM=="hid", DRIVER=="hid-generic", ENV{is_razer_mouse}=="yes", ATTRS{bInterfaceNumber}=="00", RUN+="razer_mount razermouse $kernel"

SUBSYSTEM=="hid", DRIVER=="hid-generic", ENV{is_razer_firefly}=="yes", ATTRS{bInterfaceNumber}=="00", RUN+="razer_mount razerfirefly $kernel"

KERNEL=="event[0-9]*", GROUP:="plugdev"
```

rudeb0t, 2016-09-30:
Note the above is from a fresh reboot. So it looks like udev rules are triggering, but there is no `/tmp/udev_razer.log` probably because udev rules trigger on boot before `/tmp` becomes writeable.

terrycain, 2016-09-30:
My bad there be 2 typos in the daemon. I'll fix it and get a new package out shortly

terrycain, 2016-09-30:
Launchpad is making some new packages, eventually when you get them, update and reboot

rudeb0t, 2016-10-06:
I just got around to updating my system today. Looks like the updated packages fixed the issue as I am now able to run razercommander, which crashes on exit but that's because of an uncaught exception in razercommander.

terrycain, 2016-10-06:
Ok so thats razercommander's issue, so I deem this as resolved.

closed 2016-10-06 by terrycain