openrazer/openrazer #43: Razer ManO'War

jnylen, 2016-09-10:
Hi,
Thanks for a glorious driver. I'll try to help you in fixing a driver for the new ManO'War.

lsusb: Bus 003 Device 002: ID 1532:0a02 Razer USA, Ltd
https://gist.github.com/jnylen/c0f45319ef937c39333b2ab084b8b66a

Images: http://imgur.com/a/OEw8X

The most important one is to turn the mic monitor down to 0 by default on boot if the user has specified it otherwise the mic gives off an echo if you are talking into the mic. It seems hardware specified. And seems to lose the setting on reboot of computer or turning the headset off.

It also seems the volume down, up and mic mute on the scroll wheel of the volume doesnt seem to do the right thing, one pastes, one crashes the whole system etc.

Where do you want the pcapng uploaded?

terrycain, 2016-09-11:
> Thanks for a glorious driver

Much appreciated thank you. Some credit goes to Tim for the original though.

---
> I'll try to help you in fixing a driver for the new ManO'War

Excellent :smile:. I already have some captures from a kraken to cross reference with and my contact at Razer is working on a specification for me too.

---
> lsusb: Bus 003 Device...

Cool, _looks_ like it uses the last interface to control the headset, the pcap will confirm

---
> Images...

Great :smile: 

---
> The most important one is to turn the mic monitor down to 0

I can imagine it must be extremely annoying

---
> one crashes the whole system

LOL really?

---
> Where do you want the pcapngs uploaded?

Just drag them onto this issue, that should do. Otherwise what ever means is easiest, google drive etc...

---
> Would be lovable if I can somehow add a battery checker for you, but I don't know how I can get that easily.

Ha yeah I got this for my Mamba Chroma, it eluded me for a while. With synapse open, make sure your mow is doing as little as possible to result in less traffic. Then every n seconds synapse will check the battery. Some captures of a few battery checks, along with what synapse reports the battery to be should be sufficient. Only reason I noticed this as my Mamba was set to no effect and i accidentally left wireshark running and I was like wtf is that, then I had to learn how to get responses from the device.

Would of been a perfect issue if the pcap was included ha :wink:. Nearly 100% of the info I need to start looking into it.

Also if you haven't already, a packet capture started before windows is on, continuing all the way though until synapse is started and open on the screen. That should contain the command to get the serial number, ideally if you could send the serial number (by email if you care that much) that would be helpful too. Only because the serial number is used quite heavily in the supporting software. I can output a dummy serial if needed as 99% of the time you wont have 2 mow's plugged but is not idea.

jnylen, 2016-09-11:
> LOL really?
Yeah, I have to reboot/reset DE when I accidently click or change the volume via the headset.

> Just drag them onto this issue, that should do. Otherwise what ever means is easiest, google drive etc...
Apparently github doesnt allow uploads with that file type, so uploaded them here: https://github.com/jnylen/razer-manowar-wireshark
I'll add windows boot and battery a bit later today.

terrycain, 2016-09-11:
Np. Will have a look later today or tomorrow sometime

jnylen, 2016-09-11:
Do you have an email I can send the S/N to?

terrycain, 2016-09-11:
terry@terrys-home.co.uk

terrycain, 2016-09-11:
Dont suppose you know what colour those captures used?

jnylen, 2016-09-11:
I don't sorry.

terrycain, 2016-09-11:
Ahh that would of helped as they've changed the protocol yet again.

terrycain, 2016-09-20:
Ok have pestered my contacts at Razer and I'm waiting on a reply. Mainly as if they give me the spec, its far less work for me :) will comment again when I hear back from them.

jnylen, 2016-09-20:
Alright thanks.
Just write if you need any more stuff

z3ntu, 2016-12-29:
bump, Kraken v1 & v2 driver is finished.

terrycain, 2016-12-29:
Got the spec for kraken v1 and 2 (hence they are now supported), still no spec for manowar, but will ask again when I email next

jnylen, 2017-01-26:
Hi @terrycain,
Hope you  get it asap then. Ping me if you need anything.

terrycain, 2017-01-26:
Yeah still nothing :unamused: Will have another look at the captures again to see if i notice anything this weekend.

iScrE4m, 2017-02-16:
Hey, is there anything I can do to speed this up? The volume sliders on the headphones are indeed very painful.

terrycain, 2017-02-16:
Could you get me a packet capture of you setting static mode to red blue red green blue red yellow green, specifically in that order, ideally the exact colours not lighter or darker shades

iScrE4m, 2017-02-16:
How exactly would I do that?

z3ntu, 2017-02-16:
https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol

iScrE4m, 2017-02-16:
Thanks, I'll take it home over the weekend and look into it :)

iScrE4m, 2017-02-19:
Results: Decided to return the headset, it was doing stupid things on Windows PC too, the mic feedback is really stupid and the sound isn't all that impressive for the price

terrycain, 2017-03-09:
Is anyone still after this?

jnylen, 2017-03-09:
I am.

Still using it.

terrycain, 2017-03-10:
Ok let's start again :smile:.

To start with can I get a packet capture of:
None -> Static Red -> Static Green -> Static Blue -> None -> Static Green -> None -> Static Blue -> None

jnylen, 2017-03-24:
Hi @terrycain,
Sorry for the slow answer.
Is the repo with the old files not enough?

If not i'll see if I can install it again.

terrycain, 2017-03-24:
Hopefully later but at the moment I can't differentiate any of the packets, hopefully with the order above it'll help me find out what is what

ChrisFernandez, 2017-07-12:
any news on this?

z3ntu, 2017-07-12:
Not yet.

terrycain, 2017-07-12:
> To start with can I get a packet capture of:
None -> Static Red -> Static Green -> Static Blue -> None -> Static Green -> None -> Static Blue -> None

z3ntu, 2017-07-18:
@ChrisFernandez Can you get us these packet captures?

Whitetigerswt, 2017-08-01:
I will get you this ASAP, thanks for your work.

minerinnorway, 2017-08-13:
Hey, i've tried to make a pcap but i'm pretty new to reverse engineering usb devices but here is one if it helps. 

[man o war pcap.zip](https://github.com/terrycain/razer-drivers/files/1220631/man.o.war.pcap.zip)

The I've changed the lights like this: None -> Static Red -> Static Green -> Static Blue -> None -> Static Green -> None -> Static Blue -> None

Sorry for the bad english!
Hope it helps!

Basux, 2018-02-24:
Hi im new into this and i dont know how to install this in openrazer can somone help me please (sorry for the bad english)

z3ntu, 2018-02-25:
Follow the instructions on our website: https://openrazer.github.io

But FYI the version you get there doesn't include support for the ManO'War.

mcarifio, 2018-07-11:
Is ManO'War supported at this point for say Ubuntu bionic. Doesn't look like it yet.

thomasaiman, 2018-09-07:
For anyone else checking here, the ManO'War seems to be fine on my Antergos (Arch-based) system, except the RGB lights don't work. Sound, microphone, on/off buttons, and volume wheels do work.

jnylen, 2019-01-20:
Same as @thomasaiman. It works fine now.

z3ntu, 2019-01-22:
It still doesn't have openrazer support (i.e. RGB lights), so I'm reopening this.

doorGunnerTeddy, 2019-10-20:
Hey together!
I just want to know, if there is any progress? It would be cool, if the Headset would be fully supported by the driver...

gokumatrix, 2020-02-02:
Reading the thread it seems like there's a way to get it to work without all of the features. Being there's no full support yet, is that still possible? I just need it to have sound right now.

jnylen, 2020-02-03:
@gokumatrix I'm still using it. And it works on archlinux with the usb stick. You should try to update your linux kernel if you haven't already. Otherwise you have to figure out why it doesn't work.

z3ntu, 2023-09-01:
@jnylen Please keep the issue open since RGB lights are not supported yet. If you do not want to receive further notifications (e.g. because you don't have the device anymore), you can click the "Unsubscribe" button on the right side.