openrazer/openrazer #44: Daemon support for Abyssus 2014

claell, 2016-09-11:
Currently there seems only driver support for it.

terrycain, 2016-09-11:
Working on fixing some FN issues in keyboard, but someones already alerted me to this, also need to add my mamba wired and the mamba te so will do them in one go. Should be done once i've got this latest bit of code tested and pushed.

terrycain, 2016-09-20:
Should be in daemon now, just adding some more stuff then will push out some more packages

closed 2016-09-21 by terrycain

claell, 2016-09-22:
Thanks, will test when I am at my device with Abyssus again.

terrycain, 2016-09-22:
No problem, of course up date to the latest packages, not sure if the python library supports everything fully but that can be a seperate issue.

claell, 2016-09-23:
Device is displayed now in Polychromatic. So I assume the support is successfully implemented.