openrazer/openrazer #48: ImporError: No module named 'razer_daemon'

DasHammett, 2016-09-16:
Hi

I compiled the driver just fine on my Gentoo machine on kernel 4.7.3 (make & make install) but I cannot run razer-service with the following error message:

`Traceback (most recent call last):
  File "/usr/sbin/razer-service", line 6, in <module>
    from razer_daemon.daemon import daemonize 
ImportError: No module named 'razer_daemon''`

I have checked the FAQ and I indeed have 2 folders in /usr/lib/python3/dist-packages/
- razer
- razer_daemon

I can confirm that the installation process creates those 2 directories with files in them.

Any help is appreciated

Thank you

terrycain, 2016-09-16:
Some distributions use site-packages instead of dist-packages. Have a look for other python3 modules and see where they are

volvofixthis, 2016-09-17:
hello! use this please PYTHONDIR=/usr/lib64/python3.4/site-packages make -s all install

terrycain, 2016-09-17:
> You can obtain path with run python or python3 and use this:

As the daemon and library are python3 only I'd just say python3. The following command will also print out the above.
```
python3 -c 'import sys; print("\n".join(sys.path))'
```

Once its worked, ill add an option in the Makefile for make gentoo_install which ill set this pythondir variable.

DasHammett, 2016-09-17:
Indeed I copied the modules to `/usr/lib64/python3.4/site-packages` and now razer-service starts correctly. 

I have also been able to run razerCommander without webkit-gtk... It just does not run as user, but root. I have already contected razerCommander author.

Please close the issue at your earliest convenience

closed 2016-09-17 by terrycain