openrazer/openrazer #49: Razer Blackwidow Ultimate 2014 Support

Adobe-Android, 2016-09-18:
I recently found a good deal on this keyboard and decided to buy it because I heard how awesome the open source support was on Linux. Unfortunately, the 2014 model doesn't support certain functionality. What can I do to help change that?

terrycain, 2016-09-18:
Run `lsusb -v -d VID:PID` and list what features it has. Ideally screenshots from screens on Razer synapse, look [here](https://github.com/terrycain/razer-drivers/issues/51) for example (there's a link in that issue for an example of lsusb output)

MinervaW1993, 2016-09-18:
Hi,

I happen to have a BW 2014 in hand. Here is the lsusb

Bus 004 Device 005: ID 1532:011a Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x011a 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer BlackWidow Ultimate
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8

MinervaW1993, 2016-09-18:
Hi,

I just have a BW 2014. Here is the lsusb

[lsusb](https://gist.github.com/MinervaW1993/8873c649d27e47f307859f1af90e5c05)

terrycain, 2016-09-18:
Ooh you should be in luck (depending on how you look at it). Its a clone of the 2013 so driver support is there. Just need to add it do daemon in think. Am a bit busy but should be done by end of next week if not sooner

Adobe-Android, 2016-09-19:
Thanks for your help so far. I never expected everyone to chip in and get such a quick response. I look forward to testing it out. If you still need an idea of features, Razer's website is a good source and if you want those screenshots, I can do that.

terrycain, 2016-09-19:
Nah no need my friend looks like it's identcal to my 2013 but probably with different colours. Unless they've done something dodgy it should be alright

terrycain, 2016-09-19:
If your up for it try following [this](https://github.com/terrycain/razer-drivers/wiki/Setting-up-the-keyboard-driver) as the driver _should_ already work in theory. If you've installed the packages from the PPA you can skip the first step of setting up the kernel module.

Adobe-Android, 2016-09-20:
I already followed this nice link.
http://www.omgubuntu.co.uk/2016/08/razer-chroma-keyboard-linux-configure-app
Do I need to do anything else? Because, the extra features don't currently work.

terrycain, 2016-09-20:
What features?

Adobe-Android, 2016-09-20:
Macro keys and application profiles don't work for me, but game mode and brightness levels work fine.

terrycain, 2016-09-20:
Well the 2012-2014 don't have individual addressable keys so they're either all on or all off.

Have you hit the `enable macro keys` option?

Adobe-Android, 2016-09-20:
Yes, I did in software. When I hit the hardware button it flashes and then goes out.

terrycain, 2016-09-20:
Ok so when you hit fn+f9 does it flash for a bit then go out or does it stay steady till another key is pressed?

Adobe-Android, 2016-09-20:
It flashes for a bit.

Adobe-Android, 2016-09-20:
I can confirm this: 
cat device_type
Razer BlackWidow Ultimate 2013

terrycain, 2016-09-20:
Ok then `enable macro keys` didn't do what its supposed to. Clicking that should stop it flashing. Take that up with @lah7

Try doing echo -n '1' > macro_keys

Then try Fn+F9

Adobe-Android, 2016-09-20:
Still flashing..

terrycain, 2016-09-20:
did you run that as root?

terrycain, 2016-09-20:
Wait what PPA are you using?

Adobe-Android, 2016-09-20:
ppa:lah7/polychromatic

Adobe-Android, 2016-09-20:
https://github.com/lah7/polychromatic/releases/download/v0.2.0/razer-chroma-driver_20160612_ubuntu_amd64.deb

terrycain, 2016-09-20:
That's where your going wrong. @lah7 your gonna have to deal with this as people are gonna keep getting old daemon/driver.

@Adobe-Android go [here](https://terrycain.github.io/razer-drivers/) scroll down this has PPA for latest driver. Am releasing version 1.0.9 soon which will help and then just waiting on @lah7 to merge new daemon compatibility.

Adobe-Android, 2016-09-20:
I kinda figured and hoped it would be an older version. Thank you. I'll get back to you in a moment.

terrycain, 2016-09-20:
Yeah dont know if new polychromatic workes with the new driver but I know its in the works.

terrycain, 2016-09-20:
Just released version 1.0.9, that might help a bit. Should be out on the PPA in a while, launchpad takes a while

Adobe-Android, 2016-09-20:
Here's a pastebin with issues I'm encountering.
http://pastebin.com/PvyFkDmr

Adobe-Android, 2016-09-20:
Nevermind that pastebin, absolutely no errors when installing now.

terrycain, 2016-09-20:
cool cool. Can you macro now?

Adobe-Android, 2016-09-20:
Could you possibly give me a more specific directory? 

> Ok then enable macro keys didn't do what its supposed to. Clicking that should stop it flashing. Take that up with @lah7

Try doing echo -n '1' > macro_keys
in the driver folder

Then try Fn+F9

Adobe-Android, 2016-09-21:
The only thing I care to use macros for is to launch applications.

lah7, 2016-09-21:
@terrycain Polychromatic is all set for the new driver, the README and v0.3.0 release notes are clear. it's likely users coming from the old driver/daemon, or possibly that OMG Ubuntu article, I'll contact them.

Adobe-Android, 2016-09-21:
Awesome! Can't wait to check it out. Hope that solves things. If it was
just me using the keyboard I'd just let it go, but I know this will help
many others too. I got a lot of good feedback and cooperation from all of
you. I can't thank you enough for your help.

Adobe-Android, 2016-09-21:
Thanks to those who made this possible!

terrycain, 2016-09-21:
The place where you did `cat device_type`

Adobe-Android, 2016-09-21:
After playing with the options, I tried this and got results. "echo -n '1' > mode_macro"
It won't stop blinking now though. Still not sure if the macros actually work yet.

Adobe-Android, 2016-09-21:
Should be noted that I installed the windows driver and the device driver update added some important options you were referring to earlier so that might be a good recommendation if it is possible for the user.

terrycain, 2016-09-21:
Wait is it blinking indefinitely?

Adobe-Android, 2016-09-21:
Yes, until I do a reboot.

terrycain, 2016-09-21:
Ahh thats because macro_keys != mode_macro

terrycain, 2016-09-21:
reboot. then `echo -n '1' > macro_keys` then does Fn+F9 do anything?

Adobe-Android, 2016-09-21:
I know that they are two different commands. I played with both. I'll be back in a moment.

Adobe-Android, 2016-09-21:
Also, you should know that since I updated the hardware device it flashes fast and then gets stuck in the slow flashing phase.

terrycain, 2016-09-21:
most odd, not sure whats up then

Adobe-Android, 2016-09-21:
Ya, seems to be a deeper issue?

terrycain, 2016-09-21:
yeah, it doesnt act like my 2013 or my chroma

Adobe-Android, 2016-09-21:
Just let me know if there is something I can do to help debug the situation.

terrycain, 2016-09-21:
lemme have a think on it

ricardonutella, 2016-11-22:
Total noob on Linux Mint 18 here. Did this ever end up working? I have the Stealth edition of the same model. There's a stackexchange post (https://askubuntu.com/questions/660262/macro-keys-for-razer-blackwidow-ultimate) with the same lsusb ID as mine but I don't know that if by using that it would conflict with this program or if it's even applicable or will work in my case...and I just realized one of the answers links back to this projects github.

terrycain, 2016-11-22:
Whaths vid pid again?

ricardonutella, 2016-11-22:
I actually couldn't access it, I used the -v -d command stated above and got a syntax error...then my hard drive running in a usb enclosure crashed. Don't worry about it, lol. I'm starring and following this anyway. I want to use the keyboard to learn programming with in linux, that was my original plan anyway, now I got to go to the local hardware repair place.

terrycain, 2016-12-31:
This issue still needed?

closed 2017-01-16 by terrycain