openrazer/openrazer #50: The daemon won't start

lukovdm, 2016-09-18:
When I try to start the daemon I get this message:
`2016-09-18 20:40:13 | razer                          | INFO     | Initialising Daemon. Pid: 13192
2016-09-18 20:40:13 | razer.screensaver              | INFO     | Initialising DBus Screensaver Thread
2016-09-18 20:40:13 | razer                          | INFO     | Found device.0: 0003:1532:0203.0003
2016-09-18 20:40:13 | razer.device0                  | INFO     | Initialising device.0 RazerBlackWidowChroma
2016-09-18 20:40:13 | razer.device0                  | CRITICAL | Could not get serial
Traceback (most recent call last):
  File "/usr/sbin/razer-service", line 67, in <module>
    run()
  File "/usr/sbin/razer-service", line 61, in run
    daemonize(**daemon_args)
  File "/usr/lib/python3/dist-packages/razer_daemon/daemon.py", line 110, in daemonize
    daemon = RazerDaemon(verbose, log_dir, console_log, run_dir, config_file, test_dir=test_dir)
  File "/usr/lib/python3/dist-packages/razer_daemon/daemon.py", line 202, in __init__
    self._load_devices()
  File "/usr/lib/python3/dist-packages/razer_daemon/daemon.py", line 320, in _load_devices
    razer_device = device_class(device_path, device_number, self._config, testing=self._test_dir is not None)
  File "/usr/lib/python3/dist-packages/razer_daemon/hardware/keyboards.py", line 99, in __init__
    super(RazerBlackWidowChroma, self).__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/razer_daemon/hardware/keyboards.py", line 19, in __init__
    super(MacroKeyboard, self).__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/razer_daemon/hardware/device_base.py", line 67, in __init__
    DBusService.__init__(self, self.BUS_PATH, object_path)
  File "/usr/lib/python3/dist-packages/razer_daemon/dbus_services/service.py", line 85, in __init__
    super(DBusService, self).__init__(bus_object, object_path)
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 455, in __init__
    validate_object_path(object_path)
ValueError: Invalid object path '/org/razer/device/': ends with '/' and is not just '/'
2016-09-18 20:40:15 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-09-18 20:40:15 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-09-18 20:40:15 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-09-18 20:40:15 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-09-18 20:40:15 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-09-18 20:40:15 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-09-18 20:40:15 | razer.screensaver              | WARNING  | Giving up on screensaver`

I just installed all the newest stuff from the ubuntu ppa.

terrycain, 2016-09-18:
Try it a few times, does it do it consistently?

lukovdm, 2016-09-18:
No, I just tried it again, this time it worked. Thanks

terrycain, 2016-09-18:
Yeah it's a known issue, am working on it. First time I've seen it with a keyboard. There is a sort of fix in the latest code which will make that less likely until I fix it properly. 

So basically, yeah I know, am working on it. Will close issue though as should be fixed soonish :-)

closed 2016-09-18 by terrycain