openrazer/openrazer #51: Razer Imperator 2012

MinervaW1993, 2016-09-18:
First thank you for your work on the driver ^ ^

I am having a Razer Imperator 2012 and the driver does not support it currently. I would like to provide some info (Synapse screenshot, lsusb, Wireshark control response data) for it.

Synapse screenshot
[image]
[image]
[image]
[image]
lsusb and CONTROL response data from Wireshark
[\](https://gist.github.com/MinervaW1993/25cbf7b51d2e1333756e656e07e3c057)

terrycain, 2016-09-18:
Wicked :-) Think my mamba has all of those things so should be easy to add. Good work on that mate :wink:. Just for reference you recorded the response not the request, in wireshart you wanted to packed before the one you selected.
And the packet dump would also have been useful as I have a library that extracts all that info in a nice to read format :smile:

Am busy for a bit. Then have to fix some Mamba TE stuff so can add in that not long after. Rough ETA is by end of week if not then not long after that. Feel free to join our telegram chat, that normally has a few of us spamming it with rubbish. Can add me `terrycain` or speak to @gabmus

terrycain, 2016-09-18:
> First thank you for your work on the driver ^ ^

And thanks, has been great so far :smile:. Should really start making a list of who's been helping out with device support.

MinervaW1993, 2016-09-18:
I uploaded some pcapng dump files (wireshark) for my testings (filtered to show only the info of the mouse). All of them are done by Synapse (if I press the button the mouse will move and wireshark will be filled with a lot of junk). I think you can understand what I did in Synapse by  filename.

[Impre2.tar.gz](https://github.com/terrycain/razer-drivers/files/478987/Impre2.tar.gz)

terrycain, 2016-09-18:
Heh you can add a filter to eliminate the other interfaces. Thanks that's fine. And another thing, normally I plug in a second device so that the one I'm capturing doesn't move ;-)

MinervaW1993, 2016-09-18:
@terrycain The dump files are already filtered. Also I use another mouse to operate Synapse in Virtualbox so the Imperator was not moving when collecting these data. I mean if I press the buttons on the Imperator, the Imperator will move.

terrycain, 2016-09-19:
Ahh I see. Did you not try lifting the Imperator off the desk :wink:

terrycain, 2016-09-20:
Ok I have a driver for you to test, familiarise yourself with [this](https://github.com/terrycain/razer-drivers/wiki/Setting-up-the-keyboard-driver) page.

Checkout the `mouse_driver_additions` [branch](https://github.com/terrycain/razer-drivers/tree/mouse_driver_additions) - that _should_ be a direct link.

1. Make the driver with `make driver_verbose`
1. Load the driver with `sudo insmod ./driver/razermouse.ko`
1. Goto the wiki page, start at the Binding Driver section [here](https://github.com/terrycain/razer-drivers/wiki/Setting-up-the-keyboard-driver#binding-the-chroma-to-the-keyboard-driver).
1. Work your way down to the Using the driver section, and then paste the output of the `ls -l` when your inside the device driver thing. (Aka I want to see files like `device_type`, `get_serial` etc...). BTW when it talks about USB interfaces, as your using a mouse you want the first (aka 0th) interface not the 3rd.

Make sense?

MinervaW1993, 2016-09-20:
Hi,

Here is the output of ls -l 

`stephanie@minerva /sys/bus/hid/drivers/razermouse/0003:1532:002F.0002 $ ls -l
total 0
-r--r--r-- 1 root root 4096 Sep 20 13:13 country
-r--r--r-- 1 root root 4096 Sep 20 13:13 device_type
lrwxrwxrwx 1 root root    0 Sep 20 13:13 driver -> ../../../../../../../../bus/hid/drivers/razermouse
-r--r--r-- 1 root root 4096 Sep 20 13:13 get_firmware_version
-r--r--r-- 1 root root 4096 Sep 20 13:13 get_serial
drwxr-xr-x 3 root root    0 Sep 20 13:13 hidraw
drwxr-xr-x 3 root root    0 Sep 20 13:13 input
-r--r--r-- 1 root root 4096 Sep 20 13:13 modalias
-rw-rw---- 1 root root 4096 Sep 20 13:13 mode_logo
-rw-rw---- 1 root root 4096 Sep 20 13:13 mode_scroll
-rw-rw---- 1 root root 4096 Sep 20 13:13 poll_rate
drwxr-xr-x 2 root root    0 Sep 20 13:13 power
-r--r--r-- 1 root root 4096 Sep 20 13:13 report_descriptor
--w--w---- 1 root root 4096 Sep 20 13:13 set_mouse_dpi
lrwxrwxrwx 1 root root    0 Sep 20 13:06 subsystem -> ../../../../../../../../bus/hid
-rw-r--r-- 1 root root 4096 Sep 20 13:13 uevent
`

MinervaW1993, 2016-09-20:
Here is the output

stephanie@minerva /sys/bus/hid/drivers/razermouse/0003:1532:002F.0002 $ ls -l 
total 0 
-r--r--r-- 1 root root 4096 Sep 20 13:13 country
-r--r--r-- 1 root root 4096 Sep 20 13:13 device_type
lrwxrwxrwx 1 root root    0 Sep 20 13:13 driver -> ../../../../../../../../bus/hid/drivers/razermouse
-r--r--r-- 1 root root 4096 Sep 20 13:13 get_firmware_version
-r--r--r-- 1 root root 4096 Sep 20 13:13 get_serial
drwxr-xr-x 3 root root    0 Sep 20 13:13 hidraw
drwxr-xr-x 3 root root    0 Sep 20 13:13 input
-r--r--r-- 1 root root 4096 Sep 20 13:13 modalias
-rw-rw---- 1 root root 4096 Sep 20 13:13 mode_logo
-rw-rw---- 1 root root 4096 Sep 20 13:13 mode_scroll
-rw-rw---- 1 root root 4096 Sep 20 13:13 poll_rate
drwxr-xr-x 2 root root    0 Sep 20 13:13 power
-r--r--r-- 1 root root 4096 Sep 20 13:13 report_descriptor
--w--w---- 1 root root 4096 Sep 20 13:13 set_mouse_dpi
lrwxrwxrwx 1 root root    0 Sep 20 13:06 subsystem -> ../../../../../../../../bus/hid
-rw-r--r-- 1 root root 4096 Sep 20 13:13 uevent

MinervaW1993, 2016-09-20:
And I tried echo "0" to mode_logo or alias. They are working for the lights. I am not sure what to play with set_mouse_api?

terrycain, 2016-09-20:
DPI would take 2 shorts.
Look at `./scripts/razer_mouse/set_mouse_dpi.py` you probally want to call it something like this

```
./scripts/razer_mouse/set_mouse_dpi.py --device "0003:1532:002F.0002" --dpi_x 100
OR
./scripts/razer_mouse/set_mouse_dpi.py --device "0003:1532:002F.0002" --dpi_x 100 --dpi_y 200
```
No clue why someone would want to set the dpi independantly but it should work

terrycain, 2016-09-21:
Added to current codebase

closed 2016-09-21 by terrycain

MinervaW1993, 2016-09-22:
Hi, udev did not bind the Imperator to razermouse. I think 

KERNEL=="????:????:????.????", DRIVERS=="usb", SUBSYSTEMS=="usb", ATTRS{idVendor}=="1532", ATTRS{idProduct}=="002F", ENV{is_razer_mouse}="yes"

in this line "002F" should be "002f" for 99-razer.rules

MinervaW1993, 2016-09-22:
All idProduct should be lower case I think.

terrycain, 2016-09-22:
Ahh good spot. If you've installed it via ppa the file should be located somewhere like `/lib/udev/rules.d/99-razer.rules`. Would you mind making that change and confirming it works :wink:. Making the change and rebooting is probably the nicest way to test it

othersmith, 2016-09-22:
I updated the 99-razer.rules file so that 002F was 002f, rebooted and now the device is picked up and it shows in polychromatic.  Before polychromatic said no device found.

terrycain, 2016-09-22:
I'll merge that and release new package today or tomorrow then :)

terrycain, 2016-09-22:
The community is growing :smile:

MinervaW1993, 2016-09-22:
@othersmith I didn't see my mouse in polychromatic even if it changed to 002f...

In /sys/bus/hid:
stephanie@minerva /sys/bus/hid/drivers/razermouse $ ls -la
total 0
drwxr-xr-x 2 root root    0 Sep 22 16:42 .
drwxr-xr-x 5 root root    0 Sep 22 16:42 ..
lrwxrwxrwx 1 root root    0 Sep 22 16:39 0003:1532:002F.0002 -> ../../../../devices/pci0000:00/0000:00:1d.0/usb4/4-1/4-1.6/4-1.6:1.0/0003:1532:002F.0002
--w------- 1 root root 4096 Sep 22 16:39 bind
lrwxrwxrwx 1 root root    0 Sep 22 16:42 module -> ../../../../module/razermouse
--w------- 1 root root 4096 Sep 22 16:42 new_id
--w------- 1 root root 4096 Sep 22 16:42 uevent
--w------- 1 root root 4096 Sep 22 16:42 unbind

In python razer daemon:
Python 3.5.2 (default, Jun 28 2016, 08:46:01) 
[GCC 6.1.1 20160602] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import razer.client
>>> a = razer.client.DeviceManager()
>>> a.devices
[<RazerKeyboard PM1544038402140>]
>>> 

No mouse found :(

terrycain, 2016-09-22:
Did you try a reboot?

MinervaW1993, 2016-09-23:
@terrycain @othersmith 

I reinstalled and it is working now.