openrazer/openrazer #52: drivers don't seem to detect keyboard on the new razer blade stealth (7500u version)

arjxbh, 2016-09-18:
It seems that the razer blade stealth is supported, but the new razer blade stealth with the same keyboard doesn't work.

terrycain, 2016-09-19:
Ooh the new one might be slightly different then. What's the output of `lsusb`

arjxbh, 2016-09-19:
arjun@Stealthbuntu:~$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 04f3:223c Elan Microelectronics Corp. 
Bus 001 Device 004: ID 1532:0220 Razer USA, Ltd 
Bus 001 Device 003: ID 0bda:579f Realtek Semiconductor Corp. 
Bus 001 Device 002: ID 0cf3:e300 Atheros Communications, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

terrycain, 2016-09-20:
Ok, there are multiple different versions of the blade, whilst I could just enable the current blade driver for this one it would be better to do things properly. Do you think you'll be able to follow [this](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol) guide?

JELGT2011, 2016-09-21:
Hey so I have the 2016 razer blade and am interested in getting this working.  Here's my `lsusb` output:
```bash
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 04f3:20b3 Elan Microelectronics Corp. 
Bus 001 Device 005: ID 1532:020f Razer USA, Ltd 
Bus 001 Device 004: ID 0bda:579f Realtek Semiconductor Corp. 
Bus 001 Device 003: ID 0cf3:e300 Atheros Communications, Inc. 
Bus 001 Device 002: ID 1038:1369 SteelSeries ApS 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Should I also try going through the USB reverse engineering guide?

terrycain, 2016-09-21:
@JELGT2011 Yeah sure, I have a feeling there's at least 3 different blades now.

terrycain, 2016-09-21:
Look at [this](https://github.com/terrycain/razer-drivers/issues/54) issue as an example of what I want :wink: That is, pictures, packet captures and `lsusb -v -d`

arjxbh, 2016-09-21:
I'll look through that guide and see what I come up with.  As far as I know there is just the two: the 2015 blade stealth and the 2016 blade stealth, but maybe there is a difference between the 16GB 2016 rbs, and 8GB 2016 rbs? I have a 16GB/QHD/256SSD.

terrycain, 2016-09-21:
All I know is the "keyboard" usb device seen by linux has 3 different usb product ID's

JELGT2011, 2016-09-21:
Just to be clear, I do not have the same computer as @arjxbh.  I'm using the 2016 razer blade (not stealth).

arjxbh, 2016-09-22:
Oh whoops. Misread that. Thanks for clarification.

terrycain, 2016-09-22:
Ahh then @JELGT2011 if/when you get some data for me would you put it in a separate issue as then can split up the blade and the stealth

tnedev, 2016-10-18:
I have the same issue on Mint Cinnamon with 4.7 Kernel and Razer Blade Stealth with 7500U.
Both Polychromatic and razerCommander show No Devices Found.

terrycain, 2016-10-18:
Lsusb

tnedev, 2016-10-18:
Fixed it by copying the udev rules from the latest master. Works like a charm!

terrycain, 2016-10-18:
Sorry haven't released a package in a while

arjxbh, 2016-10-21:
@tnedev - can you explain what you did to make it work?  I copied 99-razer.rules from install_files/udev/ to my system at /lib/udev/rules.d/ and it totally stopped my keyboard from working at all.  I had to connect an external usb keyboard and use that to reinstall the razer drivers to get my keyboard back.

tnedev, 2016-10-21:
@arjxbh 
Install the drivers from master 

1. git clone https://github.com/terrycain/razer-drivers.git
2. cd razer-drivers/
3. make
4. sudo make install
5. sudo reboot

terrycain, 2016-10-21:
Am away from pc till mid next week but when I've released a package you'll need to uninstall all yoube done here. Just saying :wink:

arjxbh, 2016-10-21:
awesome thanks!

vicfryzel, 2016-10-24:
Just want to point out that I tried @tnedev's suggestion on a New Razer Blade Stealth 4k, and it bricked the keyboard on reboot (no input at all accepted). I had to plug in an external keyboard and clean out master (sudo make uninstall) to get things working again on the next reboot.

terrycain, 2016-10-24:
@vicfryzel to not scare other users, it did not "brick the keyboard" it just changed the driver from the standard one to the razerkbd one which probably doesnt support the 4k. Run `lsusb` and post the output

terrycain, 2016-10-24:
@tnedev just to confirm the latest master works fro you yeah as im going to release a package

tnedev, 2016-10-24:
Yes, master worked for me.

vicfryzel, 2016-10-24:
@terrycain seems the same on the 4k.

```
○ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 04f3:2356 Elan Microelectronics Corp. 
Bus 001 Device 004: ID 1532:0220 Razer USA, Ltd 
Bus 001 Device 003: ID 0bda:579f Realtek Semiconductor Corp. 
Bus 001 Device 002: ID 0cf3:e300 Atheros Communications, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

terrycain, 2016-10-24:
@vicfryzel Hmm it should of worked, once launchpad decides to release my package try again

vicfryzel, 2016-10-25:
@terrycain can confirm that your latest package works. Had to install the deb manually for 16.04, but otherwise it's fine.

terrycain, 2016-10-25:
Ahh that's because launch pad won't simultaneously build for multiple distros, I set the build off last night and now I need to go in there and tell it to copy the package to 16.04 and 16.10

arjxbh, 2016-10-25:
upgraded all my razer related packages on ubuntu 16.04.  Polychromatic now finds my keyboard and works.  New issue, however, when I press the caps lock key it crashes my OS.

terrycain, 2016-10-25:
@arjxbh uninstall the razer packages, reboot does it still happen?

arjxbh, 2016-10-25:
I think this issue is unrelated to these drivers.  I found it here as well:

http://askubuntu.com/questions/833411/uncheck-caps-lock-crashes-ubuntu-installation

terrycain, 2016-10-25:
Awesome :-) as there's only 1 tiny part that has the capabilities to do something like that and that entire part is off by default. 

Unless I'm mistaken I think this issue is done :-)

closed 2016-10-25 by terrycain

supersambo, 2017-02-04:
I'm having this same issue on a Razer Blade Stealth QHD (not the 4k one). Here is what lsusb gives me 
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 04f3:223c Elan Microelectronics Corp. 
Bus 001 Device 005: ID 1532:0220 Razer USA, Ltd 
Bus 001 Device 004: ID 0bda:579f Realtek Semiconductor Corp. 
Bus 001 Device 003: ID 0cf3:e300 Atheros Communications, Inc. 
Bus 001 Device 002: ID 046d:c52f Logitech, Inc. Unifying Receiver
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

terrycain, 2017-02-04:
```
sudo lsmod | grep razer
```

supersambo, 2017-02-04:
doesn't return anything

terrycain, 2017-02-04:
what os are you on? what does ```sudo modprobe razerkbd``` output?

supersambo, 2017-02-04:
sorry forgot to mention. I'm on ubuntu 16.10. I get an error
`modprobe: ERROR: could not insert 'razerkbd': Required key not available`

terrycain, 2017-02-04:
disable secure boot, remove drivers, reinstall

terrycain, 2017-02-04:
> Before raising an issue saying something doesn't work, read [this](https://github.com/terrycain/razer-drivers/wiki/Troubleshooting) Wiki page, try not to create new issues if one exists, reopen it.

-----

>

supersambo, 2017-02-04:
sorry! I was convinced I had that disabled :(. Worked even without reinstalling. Thanks!