openrazer/openrazer #53: Razer Deathstalker Chroma support

ashantyk, 2016-09-20:
any change to get support for Razer Deathstalker Chroma?

terrycain, 2016-09-20:
Sure. Read [this](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol)

AndKe, 2016-12-12:
I was looking for the same..

z3ntu, 2016-12-12:
@AndKe Run lsusb and ls -lah /dev/input/by-id and paste output (in proper code blocks)

(thanks to @terrycain for the text block ^^)

AndKe, 2016-12-12:
I only thought about buying that keyboard, now I'll do it just to help you out - hoping it's more or less compatible as-is :)

z3ntu, 2016-12-12:
Just FYI it will probably take some time to get it working (as @terrycain has many things planned) except the driver is very very similar to another one. But ask terry about details ;)

terrycain, 2016-12-12:
In _theory_ it should be very similar to all the other keyboards. I'd still like you to do [this](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol) if you can (us here, and the guys on the chat can help out) as then we can be 100% but itl'll probably just be a case of looking at it and saying "yep this is the same as the chroma"

AndKe, 2016-12-13:
It's in the snail-mail.

usineur, 2016-12-13:
Hi, I also have a Razer Deathstalker Chroma and I am interested in support for this device. Here are the outputs

<details><summary>lsusb -d 1532:0204 -v</summary>
<pre><code>Bus 003 Device 002: ID 1532:0204 Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0204 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1</code></pre>
</details>

<br/>

<details><summary>ls -l /dev/input/by-id</summary>
<pre><code>total 0
lrwxrwxrwx. 1 root root 9 11 déc.  18:19 usb-Razer_Razer_DeathStalker_Chroma-event-kbd -> ../event2
lrwxrwxrwx. 1 root root 9 11 déc.  18:19 usb-Razer_Razer_DeathStalker_Chroma-if01-event-kbd -> ../event3
lrwxrwxrwx. 1 root root 9 11 déc.  18:19 usb-Razer_Razer_DeathStalker_Chroma-if02-event-mouse -> ../event4
lrwxrwxrwx. 1 root root 9 11 déc.  18:19 usb-Razer_Razer_DeathStalker_Chroma-if02-mouse -> ../mouse0</code></pre></details>

z3ntu, 2016-12-13:
@usineur How did you do the dropdown? :o

usineur, 2016-12-13:
```md
<details>
 <summary>Write a summary here</summary>
 <pre><code>Copy your content here</code></pre>
</details>
```

terrycain, 2016-12-14:
Nice

terrycain, 2016-12-15:
Also need packet captures and images of razer synapse as requested in that link to the wiki page above

AndKe, 2016-12-20:
just got mine, but given it's VID:  (0204), it fits right between those two:

Razer BlackWidow Chroma	1532:0203
Razer Blade Stealth	1532:0205

It would make sense to just try to handle it like one of those, and see if it works... there is very little chance of them actually doing something unique for this one (?)

terrycain, 2016-12-20:
Yes technically we could but I'd rather have at least screenshots from razer synapse if that's alright. Packet captures would be nicer

AndKe, 2016-12-20:
I just packed it down for return, it made a high frequency noise/beeping,
the frequency was so high, that the apparent volume changed when I rotated
head +/- 20 degrees - very annoying.  The noise went away if lights were
off.

terrycain, 2016-12-20:
Lol

terrycain, 2016-12-20:
You getting a new one or just ditching it?

CalcProgrammer1, 2016-12-21:
I just threw the DeathStalker device ID into razerkbd and compiled it.  Force removed hid_generic and re-inmodded the .ko file and it detected it.  The DeathStalker only has 6 LED lighting zones, but in the Chroma SDK they are mapped to individual keys on the 22x6 Blackwidow Chroma grid (http://developer.razerzone.com/chroma/razer-chroma-led-profiles/).  I'm going to test to see if it actually takes this grid as a USB input.  I assume it does.  The brightness control and breathing mode seem to work, and device_serial reads a valid serial number.  I'll keep testing things on it.

CalcProgrammer1, 2016-12-21:
[image]
[image]

Wireshark captures for displaying green and for displaying a rainbow (magenta, magenta, blue, cyan, green yellow) using the Chroma SDK and Keyboard Visualizer on Windows.  Don't mind the screwed up scaling and why there are two magenta zones and no red...I need to fix that.  I captured this on Windows using USBPcap and Wireshark (no VM necessary).

terrycain, 2016-12-21:
> (http://developer.razerzone.com/chroma/razer-chroma-led-profiles/)

Damn that link would of saved me much effort ;)

usineur, 2016-12-23:
@CalcProgrammer1 Just tried on my deathstalker, all matrix_effect_* are working well. Thank you.

codyctl, 2016-12-28:
Hello everyone, sorry a bit off topic but I didnt want to open an issue for a question. I have a Razer BlackWidow X Chroma and I have the driver installed and everything works great. I can adjust things like brightness, set static colors and enable effects by writing to the matrix_effect_* files. I'm trying to figure out how I can address individual zones for my keyboard and what format I need to write to matrix_effect_custom and matrix_custom_frame.

According to Razer my keyboard has 103 zones arranged in a 6x22 matrix. So for example how would I set the A key to RGB(255, 0, 0) red and leave everything else dark.

Thanks for all the work you guys have done with these drivers so far!

CalcProgrammer1, 2016-12-28:
You write to matrix_custom_frame.  First three bytes are row ID, start column ID, and end column ID, then you provide 3 bytes RGB data for each LED from start column to end column.  A Blackwidow Chroma has 6 rows (0-5) and 22 columns (0-21).

codyctl, 2016-12-28:
Can you address every key in one write to the file, or can you only address one row at a time?

terrycain, 2016-12-29:
@codyctl OR just use the daemon + python library as the driver can change and i wont inform anyone and its not supposed to be used directly. [Here](https://github.com/terrycain/razer-drivers/blob/master/scripts/daemon/chroma_keyboard/custom_effect.py) is an example using client. It has no documentation yet but its a start.

codyctl, 2016-12-29:
@terrycain The python lib and sample you cited work perfectly. Thank you much!

terrycain, 2016-12-29:
@codyctl I started writing _proper_ documentation/examples for the pylib today. Should make things simpler in the future.

terrycain, 2017-01-16:
DS Chroma is in main codebase, packages will be out soonish

closed 2017-01-16 by terrycain