openrazer/openrazer #54: Razer Orochi (wired mode)

MinervaW1993, 2016-09-21:
This is another mouse I have. 
 
lsusb and response data:https://gist.github.com/MinervaW1993/8cdefdb37187f39edadc966d1fc75c6d

Synapse screenshot
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

While idling wireshark got an auto response every 30 seconds. Maybe it is checking battery status?

Wireshark packets:
[orochi.tar.gz](https://github.com/terrycain/razer-drivers/files/484240/orochi.tar.gz)

MinervaW1993, 2016-09-21:
This is only for wired mode. For bluetooth mode... well big challenge.

terrycain, 2016-09-21:
@MinervaW1993 Your in luck as if its anything like my wireless mamba, most of the hard work should be done :wink:. Your conclusions are pretty accurate, if you look in one of the driver files there is a struct that denotes what bytes are what.

Is the mouse bluetooth or does it have a dock like the mamba?

I do love pcapng's I can cat them together and they just act as one file :smile:. Having separate ones is good, and then when I want to through it through my program I can just join up a few.

terrycain, 2016-09-21:
And LOL at synapse, the idle packet you keep seeing isnt supported by the mouse, but synapse keeps sending it ha.

terrycain, 2016-09-21:
Ok look at branch `RazerOrochi_0x0048`. You know the procedure:wink: [this](https://github.com/terrycain/razer-drivers/wiki/Using-the-mouse-driver) should help with various device files, if things arnt in there then tell me as they'll need adding.

MinervaW1993, 2016-09-21:
ls -l output:
-r--r--r-- 1 root root 4096 Sep 21 09:39 country
-r--r----- 1 root root 4096 Sep 21 09:39 device_type
lrwxrwxrwx 1 root root    0 Sep 21 09:39 driver -> ../../../../../../../../bus/hid/drivers/razermouse
-r--r----- 1 root root 4096 Sep 21 09:39 get_firmware_version
-r--r----- 1 root root 4096 Sep 21 09:39 get_serial
drwxr-xr-x 3 root root    0 Sep 21 09:39 hidraw
drwxr-xr-x 3 root root    0 Sep 21 09:39 input
-r--r--r-- 1 root root 4096 Sep 21 09:39 modalias
--w--w---- 1 root root 4096 Sep 21 09:39 mode_breath
--w--w---- 1 root root 4096 Sep 21 09:39 mode_none
--w--w---- 1 root root 4096 Sep 21 09:39 mode_reactive
-rw-rw---- 1 root root 4096 Sep 21 09:39 mode_scroll
--w--w---- 1 root root 4096 Sep 21 09:39 mode_spectrum
--w--w---- 1 root root 4096 Sep 21 09:39 mode_static
-rw-rw---- 1 root root 4096 Sep 21 09:39 poll_rate
drwxr-xr-x 2 root root    0 Sep 21 09:39 power
-r--r--r-- 1 root root 4096 Sep 21 09:39 report_descriptor
-rw-rw---- 1 root root 4096 Sep 21 09:39 set_brightness
--w--w---- 1 root root 4096 Sep 21 09:39 set_idle_time
--w--w---- 1 root root 4096 Sep 21 09:39 set_low_battery_threshold
--w--w---- 1 root root 4096 Sep 21 09:39 set_mouse_dpi
lrwxrwxrwx 1 root root    0 Sep 21 09:33 subsystem -> ../../../../../../../../bus/hid
-rw-r--r-- 1 root root 4096 Sep 21 09:39 uevent
-r--r----- 1 root root 4096 Sep 21 09:39 version

All settings work.

The mouse is Bluetooth 4.0, but unfortuanely I don't have a Linux device which has any bluetooth interface :(

terrycain, 2016-09-21:
Nice :smile: If you ever get a linux with teh bluetooth then your welcome to add wireless support ;). Will merge this to main codebase

closed 2016-09-21 by terrycain

z3ntu, 2017-01-06:
@MinervaW1993 What's the model number of the mouse? There are 3 different Orochi editions afaik (2011, 2013 & 2015).