openrazer/openrazer #55: BlackWidow Tournament Edition Stealth Driver Support

SpainTrain, 2016-09-28:
Request for driver support.

SpainTrain, 2016-09-29:
I started digging into this and got things compiling and installing locally.

However, when I try to manually bind the device to the driver, it doesn't even run `razer_kbd_probe`.  Instead I get:

```
echo: write error: No such device driver
```

The device is definitely in `/sys/bus/hid/devices/` and is not bound to any other drivers in `/sys/bus/hid/drivers/`.

Is this a common error with an obvious solution?  I would love to contribute support for this keyboard (or at least get a PR 90% there).  Even just a good RTFM source for this kind of driver dev would be helpful (googling wasn't fruitful - too general or unrelated issues). Cheers.

terrycain, 2016-09-29:
#51 and #54 are good examples of what I want. But as this keyboard is a subset of the chroma, all the support logic is probably already there, so a list of features and the output of `lsusb -vvv -d` should be a good start. If you can easily get pictures from razer synapse that would also be most useful.

terrycain, 2016-12-31:
Hey, any update on this

z3ntu, 2017-02-27:
@SpainTrain Are you still interested in support?

SpainTrain, 2017-02-27:
For unrelated reasons I'm no longer using that keyboard, so I don't have an immediate need for this. I also didn't get quite far enough to open a reasonable PR.   Thanks!

LukeL99, 2017-05-19:
I'd like to bump this. Please see attached and let me know what else I can do to help @terrycain @z3ntu. I don't have Windows installed, but if push comes to shove I could probably figure out a way to get the synapse screens. No chroma features on this board.

```
lsusb -vvv -d 1532:

Bus 003 Device 005: ID 1532:011c Razer USA, Ltd 
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x011c 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               8
```

terrycain, 2017-05-20:
What features does this kb have exactly

LukeL99, 2017-05-22:
Function key lighting - also interested in being able to switch function lock on and off to use them as media keys by default. Does this driver support that?

terrycain, 2017-05-22:
Currently no. Synapse screenshots and packet captures would be needed.

z3ntu, 2020-03-19:
Now tracked at #1050

closed 2020-03-19 by z3ntu