openrazer/openrazer #56: Razer BlackWidow X Chroma Tournament Edition support

calarco, 2016-09-30:
I cant't bind this keyboard. It's the Blackwidow X Chroma without the numpad so I guess it's not that different. What can I do to help?

terrycain, 2016-09-30:
run `lsusb`

calarco, 2016-09-30:
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 002: ID 046d:c24f Logitech, Inc. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 28de:1142  
Bus 001 Device 002: ID 1241:1503 Belkin Keyboard
Bus 001 Device 005: ID 09da:9066 A4Tech Co., Ltd. F3 V-Track Gaming Mouse
Bus 001 Device 004: ID 1532:021a Razer USA, Ltd 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

terrycain, 2016-09-30:
What OS you running?

calarco, 2016-09-30:
Archlinux. I also have a Windows partition whit Synapse if you need screenshots.

terrycain, 2016-09-30:
Nah am just too pro, should be fine :wink:

terrycain, 2016-09-30:
Think I already know this but never hurts. Output of  `ls -l /dev/input/by-id`

calarco, 2016-09-30:
```
total 0
lrwxrwxrwx 1 root root 10 sep 30 18:37 usb-A4TECH_USB_Device-event-joystick -> ../event11
lrwxrwxrwx 1 root root 10 sep 30 18:37 usb-A4TECH_USB_Device-if01-event-mouse -> ../event12
lrwxrwxrwx 1 root root  9 sep 30 18:37 usb-A4TECH_USB_Device-if01-mouse -> ../mouse2
lrwxrwxrwx 1 root root  6 sep 30 18:37 usb-A4TECH_USB_Device-joystick -> ../js1
lrwxrwxrwx 1 root root  9 sep 30 18:37 usb-Logitech_G29_Driving_Force_Racing_Wheel-event-joystick -> ../event1
lrwxrwxrwx 1 root root  6 sep 30 18:37 usb-Logitech_G29_Driving_Force_Racing_Wheel-joystick -> ../js0
lrwxrwxrwx 1 root root  9 sep 30 18:37 usb-Razer_Razer_BlackWidow_X_Tournament_Edition_Chroma-event-kbd -> ../event3
lrwxrwxrwx 1 root root  9 sep 30 18:37 usb-Razer_Razer_BlackWidow_X_Tournament_Edition_Chroma-if01-event-kbd -> ../event4
lrwxrwxrwx 1 root root  9 sep 30 18:37 usb-Razer_Razer_BlackWidow_X_Tournament_Edition_Chroma-if02-event-mouse -> ../event5
lrwxrwxrwx 1 root root  9 sep 30 18:37 usb-Razer_Razer_BlackWidow_X_Tournament_Edition_Chroma-if02-mouse -> ../mouse1
lrwxrwxrwx 1 root root  9 sep 30 18:37 usb-USB_KB_USB_KB-event-kbd -> ../event0
lrwxrwxrwx 1 root root  9 sep 30 18:37 usb-Valve_Software_Steam_Controller-event-mouse -> ../event2
lrwxrwxrwx 1 root root  9 sep 30 18:37 usb-Valve_Software_Steam_Controller-mouse -> ../mouse0

```

terrycain, 2016-09-30:
Ok should be supported in 1.0.13, when you install make sure you add yourself to plugdev group. And reboot, (trust its the easiest way and solves lots of problems). @gabmus deals with the Arch packaging, I've prodded him so should be out in the next day or so.

Some of us are hanging out on Telegram, so feel free to join, that's normally where you'll get the most up-to-date info and better "support"

calarco, 2016-09-30:
Thank you. Will try and report. This is good customer support 😁

terrycain, 2016-10-01:
Yeah packages should be out now.

> This is good customer support

Yeah I should start charging for it ha :wink:

calarco, 2016-10-01:
Tested and works great. Thanks a lot.

closed 2016-10-01 by calarco