openrazer/openrazer #57: Razer Kraken Chroma support

z3ntu, 2016-10-01:
https://github.com/pez2001/razer_chroma_drivers/issues/148

Just that this repo also has an issue about that :)

terrycain, 2016-10-19:
Have a spec. Am away from PC for just over a week so wont start driver development until then. Will have a look at spec and figure out what its doing as its rather different than the normal stuff.

Also have a kraken arriving in a few weeks :smile:

z3ntu, 2016-10-19:
Awesome!

terrycain, 2016-10-19:
At work so leaving this here for my own reference - 359

terrycain, 2016-10-25:
@z3ntu you mind getting me a packet capture of everything like literally push all the buttons (on headset) click all the options on synapse etc...

Have read through the spec and it all looks ok. Will take a while to get a driver out as am busy as well as this driver will need to be rather different

z3ntu, 2016-10-25:
So everything but without explaining all the things I did? I'll try to do it when I am home.

terrycain, 2016-10-25:
Awesome and wicked if the headphones do the mute then all's good as theres less for me to do. Just make sure you get all the effects a couple of times :smile:

z3ntu, 2016-10-25:
@terrycain Should I also play audio in the capture?

z3ntu, 2016-10-25:
[kraken_razer_drivers_everything.pcapng.gz](https://github.com/terrycain/razer-drivers/files/551706/kraken_razer_drivers_everything.pcapng.gz)
Synapse crashed in the middle so I hope everything is okay :) and apparently the OS can toggle the microphone on and off and something appeared in wireshark when I toggled it with the button.

terrycain, 2016-10-25:
:smile: cheers wont be able to put proper driver effort in till sunday but will go over capture tomorrow. Can linux toggle mute natively? As if its sending the wrong keycode then we're have to get funky with the driver. Putting lsusb output [here](http://pastebin.com/wfUk5FiS) for reference

z3ntu, 2016-10-25:
Yeah, mute also works from Linux (that the headset light goes on); just never used that :D 

And back to muting: muting from pulseaudio turns on the light but when you press the button on the headset, pulseaudio's mute doesn't go.

terrycain, 2016-10-25:
Can you run xev (dont press anything else or you'll get spam) and then spam the mute button, does it produce any outptu?

z3ntu, 2016-10-25:
Dope, xev doesn't output anything on the mute presses.

terrycain, 2016-10-25:
Hmm, must not be understanding the event then. Ok we'll have to employ all the hax to try and generate events. Good think is we know this so thats a start :)

terrycain, 2016-11-15:
I have a taken v2 :-). Am busy with tons of other stuff but it is on the list :-)

z3ntu, 2016-11-15:
Actually I just remembered that I think that some time ago the microphone button worked as a left click (that was back in my Ubuntu days so last year somewhere)... (now on Arch)

terrycain, 2016-11-15:
Lol I'll have a look once I've finished some mice drivers as kraken will be a lot of work

terrycain, 2016-12-31:
Done

closed 2016-12-31 by terrycain