openrazer/openrazer #58: Cleaning up code base, by replacing if / else if statements with switch statement

Zerophase, 2016-10-02:
Currently, I'm still waiting on my second keyboard to come before adding support for the Anansi. One thing I noticed while looking over [razerkbd_driver.c ](https://github.com/terrycain/razer-drivers/blob/master/driver/razerkbd_driver.c) It looks like we can simplify a decent chunk of the code by replacing `if else if `statements with `switch` statements using fallthrough.

for instance lines 1449 through 1473 have:
```
if(usb_dev->descriptor.idProduct == USB_DEVICE_ID_RAZER_BLACKWIDOW_ORIGINAL ||
       usb_dev->descriptor.idProduct == USB_DEVICE_ID_RAZER_BLACKWIDOW_ULTIMATE_2012 ||
       usb_dev->descriptor.idProduct == USB_DEVICE_ID_RAZER_BLACKWIDOW_ULTIMATE_2013)
    {
        // Set BlackWidow Ultimate to static colour
        razer_set_static_mode_blackwidow_ultimate(usb_dev);

    } else if(usb_dev->descriptor.idProduct == USB_DEVICE_ID_RAZER_BLACKWIDOW_CHROMA ||
              usb_dev->descriptor.idProduct == USB_DEVICE_ID_RAZER_BLACKWIDOW_CHROMA_TE ||
              usb_dev->descriptor.idProduct == USB_DEVICE_ID_RAZER_BLACKWIDOW_X_CHROMA ||
              usb_dev->descriptor.idProduct == USB_DEVICE_ID_RAZER_BLACKWIDOW_X_CHROMA_TE ||
              usb_dev->descriptor.idProduct == USB_DEVICE_ID_RAZER_BLACKWIDOW_ULTIMATE_2016 ||
              usb_dev->descriptor.idProduct == USB_DEVICE_ID_RAZER_BLADE_STEALTH ||
              usb_dev->descriptor.idProduct == USB_DEVICE_ID_RAZER_TARTARUS_CHROMA)
    {
        // Set BlackWidow Chroma to static colour
        if(count == 3)
        {
            razer_set_static_mode(usb_dev, (struct razer_rgb*)&buf[0]);
        }

    } else
    {
        printk(KERN_WARNING "razerkbd: Cannot set static mode for this device");
    }
```
It could become:

```
switch(usb_dev->descriptor.idProduct)
{
     case USB_DEVICE_ID_RAZER_BLACKWIDOW_ORIGINAL:
     case USB_DEVICE_ID_RAZER_BLACKWIDOW_ULTIMATE_2012:
     case USB_DEVICE_ID_RAZER_BLACKWIDOW_ULTIMATE_2013:
          razer_set_static_mode_blackwidow_ultimate(usb_dev);
          break;
   // rest of the cases for other devices go below.
}
```

There are other places where it looks like switch statements with fallthrough would simplify the code a bit. This should aid in maintenance over time, and might have slight performance improvements as well. (don't believe the performance gains matter though) I just want to double check that this is something desired before I start making a ton of tiny refactors.

terrycain, 2016-10-02:
Technically the compiler should optimise them to the same thing but you are correct in thinking the switch would look nicer. Feel free to pull request, just make sure the logics the same yeah :wink:. I've been writing too much python lately, I keep forgetting switch case exists.

Zerophase, 2016-10-02:
Yeah, the compiler should optimize it, but sometimes it doesn't.

terrycain, 2016-10-03:
Yeah I've noticed it's been picky a few times. So yeah feel free, so long as the diffs clearly show what's happening (and not tons of whitespace changes :wink:) all should be fine

terrycain, 2016-11-15:
Am closing this as an pretty sure I merged lots of switch/cases. If not they'll be here soon :wink:

closed 2016-11-15 by terrycain