openrazer/openrazer #59: Razer Naga Chroma Support

zdjm, 2016-10-03:
Love what you are doing My keyboard works 100% flawless. Can you please add in the Naga Chroma support, ill add in what ever is needed or give you any info you may need.

terrycain, 2016-10-03:
Output of `lsusb` and `ls -l /dev/input/by-id`.
Also if you have access to a Windows pc/vm with synapse on it, a screenshot of every synapse option would be good too

ali, 2016-10-08:
Hi @terrycain. I also have a Razer Naga Chroma and am interested in helping add support for this device. Here's the output from `lsusb` and `ls -l /dev/input/by-id` that you wanted. I'll provide screenshots of options in Synapse soon.

<details>
<summary>`lsusb -d 1532:0053 -v`</summary>
[lsusb.txt](https://github.com/terrycain/razer-drivers/files/517326/lsusb.txt)
```
Bus 001 Device 003: ID 1532:0053 Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0053 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Naga Chroma
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
```
</details>
<details>
<summary>`ls -l /dev/input/by-id`</summary>
```
total 0
lrwxrwxrwx 1 root root 9 Oct  7 09:22 usb-Razer_Razer_Naga_Chroma-event-mouse -> ../event0
lrwxrwxrwx 1 root root 9 Oct  7 09:22 usb-Razer_Razer_Naga_Chroma-if01-event-kbd -> ../event1
lrwxrwxrwx 1 root root 9 Oct  7 09:22 usb-Razer_Razer_Naga_Chroma-if02-event-kbd -> ../event2
lrwxrwxrwx 1 root root 9 Oct  7 09:22 usb-Razer_Razer_Naga_Chroma-mouse -> ../mouse0
```
</details>

zdjm, 2016-10-10:
what he said @ali  thanks 
[image]
[image]
[image]

terrycain, 2016-10-10:
Ok I believe accleration is a os based thing so were ignore that. 

I need the following packet captures:
1x capture of DPI going from min to max a few times

For each effect on scroll wheel (static, breathing etc...)
1x effect -> none -> effect -> none -> effect
e.g. 1x static(green) -> none -> static(green) -> none ->static(green)
e.g. 1x breathing -> none -> breathing -> none ->breathing

And the repeat the packet captures for each effect on the thumbgrid and logo

Also could do with brightness going from dim->bright->dim->bright for scroll wheel, thumbgrid and logo

Make sense?

zdjm, 2016-10-10:
It makes sense, but what program do you want me to use for the packet capture? I assume you want this on windows side.

z3ntu, 2016-10-10:
Wireshark on Linux with Windows running in a VM on top of that machine running Wireshark. (afaik)

terrycain, 2016-10-10:
@z3ntu ta, yeah what he said. Its all on the wiki :wink:

artemjackson, 2016-12-04:
@terrycain Thank you for the great work on drivers!
Could you also add support for Naga Hex v2? I just ordered the mouse and will be able to provide you with all necessary information if needed.

terrycain, 2016-12-04:
If you provide the info sure :wink:, open different issue though. If you join the telegram chat when you have the mouse the lads there will be sure to help as well.

CalcProgrammer1, 2016-12-22:
Naga Chroma seems to work differently from DeathAdder Chroma despite appearing to have a similar design.  The set_led_effect command is broken on this mouse.  Looking at the packet capture from Synapse, the command is 0x0D and it looks like the RGB data is packed in to the same packet if you command a mode with RGB data (such as breathing).  The basic stuff works (device type/firmware/serial) and setting the logo and scroll wheel brightness works, but the effects and rgb do not.  I haven't tested dpi.  The key grid is its own LED tab in Synapse with the same options as the logo and scroll wheel.  I'll investigate more later.

terrycain, 2016-12-22:
Yep it uses `extended effects` as the protocol spec calls it. Leave the naga, am working on that as i also have one.

KazWolfe, 2016-12-26:
Also note that there's a Naga Epic Chroma, which should (?) be a similar device.

<details><summary>Output of <code>lsusb</code></summary>
<pre>
Bus 001 Device 023: ID 1532:003e Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x003e 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Naga Epic Chroma
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           84
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        2
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      0 No Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength     159
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x83  EP 3 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0010  1x 16 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)
</pre>
</details>

<details><summary>Output of <code>ls /dev/input/by-id</code> command</summary>
<pre>
lrwxrwxrwx 1 root root 10 Dec 26 00:31 usb-Razer_Razer_Naga_Epic_Chroma-event-mouse -> ../event12
lrwxrwxrwx 1 root root 10 Dec 26 00:31 usb-Razer_Razer_Naga_Epic_Chroma-if01-event-kbd -> ../event13
lrwxrwxrwx 1 root root 10 Dec 26 00:31 usb-Razer_Razer_Naga_Epic_Chroma-if02-event-kbd -> ../event14
lrwxrwxrwx 1 root root  9 Dec 26 00:31 usb-Razer_Razer_Naga_Epic_Chroma-mouse -> ../mouse3
</pre>
</details>

terrycain, 2016-12-26:
oooh, havent seen a wireless one with a dock since I did my Mamba. Am I right in thinking it can only do static/pulsating/spectrum cycling?

CalcProgrammer1, 2016-12-26:
Naga Epic Chroma can do custom effects in wired mode for sure, haven't tried wireless but I think they added it recently.  For a while the SDK didn't work with any wireless devices, but either way Synapse did work with them and likely sends the same messages.

terrycain, 2016-12-26:
Hmm, in that case, I'd like packet captures of the naga epic as the spec I have says it cant (though theres been times the spec was wrong)

KazWolfe, 2016-12-26:
I can probably get a packet capture (raw, if that's okay) some time within the next few days. Do you want anything special or is just what the wiki has enough?

The dock itself has no lights or anything, everything is on the mouse. The mouse itself is able to do spectrum cycling, etc. in both wired and wireless mode. Under Syanpse, it can be updated in either operating mode.

terrycain, 2016-12-26:
Err if you paste some screenshots of all the options in synase I can get all specific in what i want :wink: and no rush, I have a few devices here am working on at the moment.

KazWolfe, 2016-12-31:
Screenshots from Naga Epic Chroma, as requested:

[image]

[image]

[image]

Lighting effects are same for both options.

[image]

terrycain, 2016-12-31:
OK Packet capture time, "shouldn't" be difficult, i'd recommend to use a different mouse whilst doing it (otherwise you get packet spam).

* 1 capture of setting scroll wheel to spectrum -> none -> spectrum -> none
* ^^ same as above for thumbgrid
* 1 capture of clicking each brightnses mode in wired
* ^^ same as above for wireless
* 1 capture of you clicking a few dpi numbers (like 800, 1500, 2000)
* 1 capture of you clicking "sleep after idle" numbers, ideally 1->3->5->...->15
* 1 capture of blink scrollwheel light when power level is low, 5%->10->15->20->25
* 1 capture of synapse open, left for like 15 mins (on wireless) (make sure you have the filters set right, and that you save only the packets that got past the filter)
* ^^ same as above but on wired
* 1 capture starting before the vm is on, stopping when synapse is open on the desktop (aka full startup)
* 1 capture of selecting 128, 500, 1000 poll rate

Am pretty confident I already have code in the driver to do all of these but I have seen how often razer changes stuff when you'd wouldn't expect it so I like to have a bit of evidence. On the bright side, shouldn't take long to add support there after.

simpleauthority, 2017-01-27:
OK. Here are my packet caps and screenshots for this device. For the lighting changes, I applied to 'scroll wheel' then cloned to the rest of the device. If you want me to do scroll wheel only, thumb pad only, and or something else lemme know and I'll do it then push to my repo...

Pics: http://imgur.com/a/8dJn4
Caps repo: https://github.com/simpleauthority/razer-naga-chroma-packet-captures

Hope I did it right :)

simpleauthority, 2017-01-27:
Edit: Started on the requests from Terry's above post including each component's lighting options, will continue in a bit :)

terrycain, 2017-01-30:
@simpleauthority what distro you running

maciek134, 2017-02-01:
Whoops, started adding support for Naga Epic Chroma yesterday (didn't see that issue), @terrycain are you working on it or should I continue?

terrycain, 2017-02-01:
@maciek134 Yeah as far as im aware its identical to the Naga Hex V2, have a branch here `feature_nagachroma_0x0053`

simpleauthority, 2017-02-01:
@terrycain Manjaro/Arch

maciek134, 2017-02-01:
@terrycain It has more buttons and based on what I've seen in packets it sends different key ids:
```
#      0  1  2  3
#   -------------
# 0 |  3  6  9 12
# 1 |  2  5  8 11
# 2 |  1  4  7 10

NAGA_EPIC_CHROMA_KEY_MAPPING = {
    '1': (2, 0),
    '2': (1, 0),
    '3': (0, 0),
    '4': (2, 1),
    '5': (1, 1),
    '6': (0, 1),
    '7': (2, 2),
    '8': (1, 2),
    '9': (0, 2),
    '10': (2, 3),
    '11': (1, 3),
    '12': (0, 3)
}
...
NAGA_EPIC_CHROMA_EVENT_MAPPING = {
    0x1e: '1',
    0x1f: '2',
    0x20: '3',
    0x21: '4',
    0x22: '5',
    0x23: '6',
    0x24: '7',
    0x25: '8',
    0x26: '9',
    0x27: '10',
    0x2d: '11',
    0x2e: '12'
}
```

terrycain, 2017-02-02:
Ahh yeah. I'll add that to the list, doesn't affect the rest though :wink: @z3nth arch install instructions please :smile:

z3ntu, 2017-02-02:
You probably will get some dependency issues because some packages needed are only available in the AUR -> install them first (AUR helper or whatever you like)
**Run the following commands:**
```
git clone https://aur.archlinux.org/razer-drivers-git.git
cd razer-drivers-git
curl https://gist.githubusercontent.com/z3ntu/28cb958ce9a778de0975e54c54b77318/raw/0afe285366c35799924717b3286f460425e69bd6/PKGBUILD.nagachroma -o PKGBUILD
makepkg -sr
sudo pacman -U python-razer-git-*.tar* razer-daemon-git-*.tar* razer-driver-dkms-git-*.tar*
```
If you need additional help, you can reach me on Telegram @z3ntu.

[Also, if you don't have dkms installed already, don't forget to install the kernel headers for your kernel!](https://wiki.archlinux.org/index.php/Dynamic_Kernel_Module_Support#Installation)

maciek134, 2017-02-02:
@terrycain the branch you mentioned doesn't contain anything for Naga Epic Chroma (DEV_ID 0x003F for wireless and 0x003E for wired), added classes based on HexV2 and it doesn't work, I'm going to look through packets later, I guess I'll PR when it works?

terrycain, 2017-02-02:
@maciek134 thats an entirely different device

maciek134, 2017-02-02:
@terrycain everything I wrote in this comments was in relation to https://github.com/terrycain/razer-drivers/issues/59#issuecomment-269851454 I thought it was better not to write about it here, but the discussion was already started.

terrycain, 2017-02-02:
@maciek134 yeah i think razer have 2 versions of the naga epic chroma both called exactly that, hence I differentiate devices by their usb id not the name :wink:

usineur, 2017-02-02:
Is that possible to currently change the "1 ->12" buttons bindings ?

terrycain, 2017-02-02:
> Is that possible to currently change the "1 ->12" buttons bindings ?

Ish, not very well

simpleauthority, 2017-02-02:
After following @z3ntu's instructions, the device is detected in polychromatic however setting options for light display does not seem to work -- watching log shows requests are coming through. Unfortunately I am not versed enough with C++ or how drivers work in order to help with this :p Sorry

maciek134, 2017-02-03:
@simpleauthority I have the same symptoms for Naga Epic Chroma (0x003e/0x003f), probably different commands are needed. I can patch yours in once I'm done with mine. Actually, I wrote my comments based on this one: https://github.com/terrycain/razer-drivers/issues/59#issuecomment-269184445, I misunderstood what mouse you were talking about.

terrycain, 2017-02-03:
@maciek134 go to a different issue, you device is inheriantly different.

@simpleauthority 
```
ls -l /sys/bus/hid/drivers
ls -l /sys/bus/hid/drivers/razermouse
ls -l /sys/bus/hid/drivers/razermouse/0*/
```

simpleauthority, 2017-02-03:
@terrycain Sure, here ya go. :)

<details><summary>ls -l /sys/bus/hid/drivers</summary><p>

```
total 0
drwxr-xr-x 2 root root 0 Feb  3 00:27 hid-generic
drwxr-xr-x 2 root root 0 Feb  3 08:27 razerfirefly
drwxr-xr-x 2 root root 0 Feb  3 08:27 razerkbd
drwxr-xr-x 2 root root 0 Feb  3 08:27 razermouse
```
</p></details>
<details><summary>ls -l /sys/bus/hid/drivers/razermouse</summary><p>

```
total 0
lrwxrwxrwx 1 root root    0 Feb  3 08:27 0003:1532:0053.0004 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.0/0003:1532:0053.0004
lrwxrwxrwx 1 root root    0 Feb  3 08:27 0003:1532:0053.0005 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.1/0003:1532:0053.0005
lrwxrwxrwx 1 root root    0 Feb  3 08:27 0003:1532:0053.0006 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-7/1-7:1.2/0003:1532:0053.0006
--w------- 1 root root 4096 Feb  3 08:27 bind
lrwxrwxrwx 1 root root    0 Feb  3 11:24 module -> ../../../../module/razermouse
--w------- 1 root root 4096 Feb  3 11:24 new_id
--w------- 1 root root 4096 Feb  3 08:27 uevent
--w------- 1 root root 4096 Feb  3 11:24 unbind

```
</p></details>
<details><summary>ls -l /sys/bus/hid/drivers/razermouse/0*/</summary><p>

```
'/sys/bus/hid/drivers/razermouse/0003:1532:0053.0004/':
total 0
-r--r--r-- 1 root plugdev 4096 Feb  3 08:27 country
-rw-rw---- 1 root plugdev 4096 Feb  3 08:27 device_mode
-r--r----- 1 root plugdev 4096 Feb  3 08:27 device_serial
-r--r----- 1 root plugdev 4096 Feb  3 08:27 device_type
-rw-rw---- 1 root plugdev 4096 Feb  3 08:27 dpi
lrwxrwxrwx 1 root plugdev    0 Feb  3 08:27 driver -> ../../../../../../../bus/hid/drivers/razermouse
-r--r----- 1 root plugdev 4096 Feb  3 08:27 firmware_version
drwxr-xr-x 3 root plugdev    0 Feb  3 08:27 hidraw
drwxr-xr-x 3 root plugdev    0 Feb  3 08:27 input
-rw-rw---- 1 root plugdev 4096 Feb  3 08:27 logo_led_brightness
--w--w---- 1 root plugdev 4096 Feb  3 08:27 logo_matrix_effect_breath
--w--w---- 1 root plugdev 4096 Feb  3 08:27 logo_matrix_effect_none
--w--w---- 1 root plugdev 4096 Feb  3 08:27 logo_matrix_effect_reactive
--w--w---- 1 root plugdev 4096 Feb  3 08:27 logo_matrix_effect_spectrum
--w--w---- 1 root plugdev 4096 Feb  3 08:27 logo_matrix_effect_static
-rw-rw---- 1 root plugdev 4096 Feb  3 08:27 matrix_brightness
--w--w---- 1 root plugdev 4096 Feb  3 08:27 matrix_custom_frame
--w--w---- 1 root plugdev 4096 Feb  3 11:13 matrix_effect_breath
--w--w---- 1 root plugdev 4096 Feb  3 08:27 matrix_effect_custom
--w--w---- 1 root plugdev 4096 Feb  3 08:27 matrix_effect_none
--w--w---- 1 root plugdev 4096 Feb  3 08:27 matrix_effect_reactive
--w--w---- 1 root plugdev 4096 Feb  3 08:27 matrix_effect_spectrum
--w--w---- 1 root plugdev 4096 Feb  3 08:27 matrix_effect_static
-r--r--r-- 1 root plugdev 4096 Feb  3 08:27 modalias
-rw-rw---- 1 root plugdev 4096 Feb  3 08:27 poll_rate
drwxr-xr-x 2 root plugdev    0 Feb  3 08:27 power
-r--r--r-- 1 root plugdev 4096 Feb  3 08:27 report_descriptor
-rw-rw---- 1 root plugdev 4096 Feb  3 08:27 scroll_led_brightness
--w--w---- 1 root plugdev 4096 Feb  3 08:27 scroll_matrix_effect_breath
--w--w---- 1 root plugdev 4096 Feb  3 08:27 scroll_matrix_effect_none
--w--w---- 1 root plugdev 4096 Feb  3 08:27 scroll_matrix_effect_reactive
--w--w---- 1 root plugdev 4096 Feb  3 08:27 scroll_matrix_effect_spectrum
--w--w---- 1 root plugdev 4096 Feb  3 08:27 scroll_matrix_effect_static
lrwxrwxrwx 1 root plugdev    0 Feb  3 08:27 subsystem -> ../../../../../../../bus/hid
--w--w---- 1 root plugdev 4096 Feb  3 08:27 test
-rw-r--r-- 1 root plugdev 4096 Feb  3 08:27 uevent
-r--r----- 1 root plugdev 4096 Feb  3 08:27 version

'/sys/bus/hid/drivers/razermouse/0003:1532:0053.0005/':
total 0
-r--r--r-- 1 root plugdev 4096 Feb  3 08:27 country
lrwxrwxrwx 1 root plugdev    0 Feb  3 08:27 driver -> ../../../../../../../bus/hid/drivers/razermouse
drwxr-xr-x 3 root plugdev    0 Feb  3 08:27 hidraw
drwxr-xr-x 3 root plugdev    0 Feb  3 08:27 input
-r--r--r-- 1 root plugdev 4096 Feb  3 08:27 modalias
drwxr-xr-x 2 root plugdev    0 Feb  3 08:27 power
-r--r--r-- 1 root plugdev 4096 Feb  3 08:27 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Feb  3 08:27 subsystem -> ../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Feb  3 08:27 uevent

'/sys/bus/hid/drivers/razermouse/0003:1532:0053.0006/':
total 0
-r--r--r-- 1 root plugdev 4096 Feb  3 08:27 country
lrwxrwxrwx 1 root plugdev    0 Feb  3 08:27 driver -> ../../../../../../../bus/hid/drivers/razermouse
drwxr-xr-x 3 root plugdev    0 Feb  3 08:27 hidraw
drwxr-xr-x 3 root plugdev    0 Feb  3 08:27 input
-r--r--r-- 1 root plugdev 4096 Feb  3 08:27 modalias
drwxr-xr-x 2 root plugdev    0 Feb  3 08:27 power
-r--r--r-- 1 root plugdev 4096 Feb  3 08:27 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Feb  3 08:27 subsystem -> ../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Feb  3 08:27 uevent
```

</p></details>

terrycain, 2017-02-03:
This will to to start, paste commands individually as otherwise you might miss some stuff.

sudo to root then `cd /sys/bus/hid/drivers/razermouse/0003:1532:0053.0004/`
```
# General Info
cat device_mode
cat device_serial
cat device_type
cat firmware_version

# Get/Set DPI
cat dpi
echo -ne '\x00\x64' > dpi
cat dpi
echo -ne '\x05\xdc' > dpi
cat dpi

## This should only affect the thumbgrid
# Get/Set brightness
cat matrix_brightness
echo -n '128' > matrix_brightness
cat matrix_brightness
echo -n '255' > matrix_brightness

# Spectrum
echo -n '1' > matrix_effect_spectrum

# Static
echo -ne '\xFF\x00\x00' > matrix_effect_static
echo -ne '\xFF\xFF\x00' > matrix_effect_static
echo -ne '\xFF\x00\xFF' > matrix_effect_static
```

simpleauthority, 2017-02-04:
@terrycain 

Alright, here we go:
```
cat device_mode: https://gist.github.com/7684a99a6a2c1599bd1d2317a72c8946
cat device_serial: https://gist.github.com/159c9c8bba899532d3ea0ca6cdc7564d
cat device_type: https://gist.github.com/d1d034c85878264c1b84eaec0cdfefb2
cat firmware_version: **https://gist.github.com/9d409ae14ec69af2378a11e4492af5d7
```

Getting/setting dpi works great.

Trying to enable matrix_effect_spectrum with '1' or any of the matrix_effect_static commands did anything noticeable to the mouse. Further, root didn't have permission to read any of the matrix_effect_* files. Odd.

terrycain, 2017-02-05:
Yeah those files arnt readable, if you look at the file permissions it says so :wink:

Have pushed new changes to the branch so you'll need to reinstall. As root try the following again, if the thumbgrid works then i'll make that change to the rest of things
```
echo -ne '\xFF\x00\x00' > matrix_effect_static
echo -ne '\xFF\xFF\x00' > matrix_effect_static
echo -ne '\xFF\x00\xFF' > matrix_effect_static
```

simpleauthority, 2017-02-06:
@terrycain Cool. Reinstalled. /sys/bus/hid/drivers/razermouse does not have a directory that ends in 0004 anymore, only 0009, 000A, and 000B.

simpleauthority, 2017-02-06:
The one ending in 0009 looked exactly the way 0004 looked, so I tried the commands you gave me anyway - no effect, though.

terrycain, 2017-02-06:
did you reboot? as reinstalling alone wont unload driver from memory and load in the new one. And if you rebooted the numbers should be the same ;-)

simpleauthority, 2017-02-08:
@terrycain Sorry, I think I replied in telegram. :p But yes, I did. And using the command on the folder that had the same directory structure as 0004 did not do anything.

BTW I think the numbers changed because the USB port is different. Maybe.

terrycain, 2017-02-08:
> BTW I think the numbers changed because the USB port is different. Maybe

Possibly.

And yeah I'll have a look nearer the weekend

theshadow124, 2017-03-15:
Hello, just wanted to add the screen shot of the side view: 
[image]

terrycain, 2017-03-17:
Where did we get to with this?

seifer44, 2017-05-15:
I have just tested this out pretty thoroughly with my Razer Naga Chroma (not the epic). Most stuff appears to be working [on this page](https://github.com/terrycain/razer-drivers/wiki/Using-the-mouse-driver) (minus the charging and dock stuff, which I didn't test).

Stuff that isn't working:

* Only the keyboard lighting by the thumb seems to respond to the driver changes. The mouse wheel and Razer logo are not changed at all.
* It looks like DPI isn't getting reset right.

```
seifer44-lx 0003:1532:0053.0001 # /home/seifer44/gitrepos/razer-drivers/scripts/razer_mouse/driver/set_mouse_dpi.py -d "0003:1532:0053.0001" --dpi_x 1800 --dpi_y 1800
Done
seifer44-lx 0003:1532:0053.0001 # cat dpi
1600:1600
```

As expected, the mouse wheel left-click and right-click isn't working, but interesting enough, my DPI buttons are working with what I had set them in Windows (which was forward a page and back a page). I'm not sure if this is known behavior or not.

I am running on Linux Mint 18.1. Please let me know if I can provide any more information, as I'd love to help this project out.

terrycain, 2017-05-20:
ok lets start from the beginning as I've forgotten all about this.

Does the driver bind automatically on reboot or do you bind manually?
also output of `ls -lah /sys/bus/.../0003:1532:0053.0001/`

seifer44, 2017-05-20:
I have to bind manually. I thought it was binding on reboot, but I appear to have been mistaken.

```
seifer44-lx devices # ls -lah /sys/bus/hid/devices/0003\:1532\:0053.0001/
total 0
drwxr-xr-x 5 root root    0 May 20 15:43 .
drwxr-xr-x 5 root root    0 May 20 07:37 ..
-r--r--r-- 1 root root 4.0K May 20 15:43 country
lrwxrwxrwx 1 root root    0 May 20 07:37 driver -> ../../../../../../../bus/hid/drivers/hid-generic
drwxr-xr-x 3 root root    0 May 20 07:37 hidraw
drwxr-xr-x 3 root root    0 May 20 07:37 input
-r--r--r-- 1 root root 4.0K May 20 15:43 modalias
drwxr-xr-x 2 root root    0 May 20 15:43 power
-r--r--r-- 1 root root 4.0K May 20 15:43 report_descriptor
lrwxrwxrwx 1 root root    0 May 20 07:37 subsystem -> ../../../../../../../bus/hid
-rw-r--r-- 1 root root 4.0K May 20 07:37 uevent
```

terrycain, 2017-05-20:
Ahh can you rebind it and run the `ls`

seifer44, 2017-05-20:
Whoops. My bad.

```
seifer44-lx razermouse # ls -lah /sys/bus/hid/devices/0003\:1532\:0053.0001/
total 0
drwxr-xr-x 5 root root    0 May 20 15:47 .
drwxr-xr-x 5 root root    0 May 20 15:43 ..
-r--r--r-- 1 root root 4.0K May 20 15:47 country
-rw-rw---- 1 root root 4.0K May 20 15:47 device_mode
-r--r----- 1 root root 4.0K May 20 15:47 device_serial
-r--r----- 1 root root 4.0K May 20 15:47 device_type
-rw-rw---- 1 root root 4.0K May 20 15:47 dpi
lrwxrwxrwx 1 root root    0 May 20 15:47 driver -> ../../../../../../../bus/hid/drivers/razermouse
-r--r----- 1 root root 4.0K May 20 15:47 firmware_version
drwxr-xr-x 3 root root    0 May 20 15:47 hidraw
drwxr-xr-x 3 root root    0 May 20 15:47 input
-rw-rw---- 1 root root 4.0K May 20 15:47 logo_led_brightness
--w--w---- 1 root root 4.0K May 20 15:47 logo_matrix_effect_breath
--w--w---- 1 root root 4.0K May 20 15:47 logo_matrix_effect_none
--w--w---- 1 root root 4.0K May 20 15:47 logo_matrix_effect_reactive
--w--w---- 1 root root 4.0K May 20 15:47 logo_matrix_effect_spectrum
--w--w---- 1 root root 4.0K May 20 15:47 logo_matrix_effect_static
-rw-rw---- 1 root root 4.0K May 20 15:47 matrix_brightness
--w--w---- 1 root root 4.0K May 20 15:47 matrix_custom_frame
--w--w---- 1 root root 4.0K May 20 15:47 matrix_effect_breath
--w--w---- 1 root root 4.0K May 20 15:47 matrix_effect_custom
--w--w---- 1 root root 4.0K May 20 15:47 matrix_effect_none
--w--w---- 1 root root 4.0K May 20 15:47 matrix_effect_reactive
--w--w---- 1 root root 4.0K May 20 15:47 matrix_effect_spectrum
--w--w---- 1 root root 4.0K May 20 15:47 matrix_effect_static
-r--r--r-- 1 root root 4.0K May 20 15:43 modalias
-rw-rw---- 1 root root 4.0K May 20 15:47 poll_rate
drwxr-xr-x 2 root root    0 May 20 15:43 power
-r--r--r-- 1 root root 4.0K May 20 15:47 report_descriptor
-rw-rw---- 1 root root 4.0K May 20 15:47 scroll_led_brightness
--w--w---- 1 root root 4.0K May 20 15:47 scroll_matrix_effect_breath
--w--w---- 1 root root 4.0K May 20 15:47 scroll_matrix_effect_none
--w--w---- 1 root root 4.0K May 20 15:47 scroll_matrix_effect_reactive
--w--w---- 1 root root 4.0K May 20 15:47 scroll_matrix_effect_spectrum
--w--w---- 1 root root 4.0K May 20 15:47 scroll_matrix_effect_static
lrwxrwxrwx 1 root root    0 May 20 15:43 subsystem -> ../../../../../../../bus/hid
--w--w---- 1 root root 4.0K May 20 15:47 test
-rw-r--r-- 1 root root 4.0K May 20 15:43 uevent
-r--r----- 1 root root 4.0K May 20 15:47 version
```

terrycain, 2017-05-21:
Ok cd into that folder and run 
```
cat device_mode
cat device_serial
cat device_type
cat dpi
cat firmware_version
cat poll_rate
```

seifer44, 2017-05-21:
```
seifer44-lx 0003:1532:0053.0001 # cat device_mode
0:0
seifer44-lx 0003:1532:0053.0001 # cat device_serial
PM1605H0080XXXX
seifer44-lx 0003:1532:0053.0001 # cat device_type
Razer Naga Chroma
seifer44-lx 0003:1532:0053.0001 # cat dpi
2100:2100
seifer44-lx 0003:1532:0053.0001 # cat firmware_version
v1.0
seifer44-lx 0003:1532:0053.0001 # cat poll_rate
500
seifer44-lx 0003:1532:0053.0001 # 
```

I censored the last 4 digits of my serial. Literal digits, if that's of any need.

alexriedl, 2017-06-15:
Any progress on this? Or is there any additional information needed?

terrycain, 2017-06-15:
Ok now to test lighting
```
echo -n '255' > ./scroll_led_brightness
```

seifer44, 2017-06-16:
My mouse is set to pulse between two lighting schemes, and it continued to do so during this test. However, I _did_ notice a difference between the intensity of the lighting effects when each option was applied.

```
seifer44-lx 0003:1532:0053.0001 # ls -lah
total 0
drwxr-xr-x 5 root root    0 Jun 15 13:01 .
drwxr-xr-x 5 root root    0 Jun 15 13:01 ..
-r--r--r-- 1 root root 4.0K Jun 15 19:08 country
-rw-rw---- 1 root root 4.0K Jun 15 19:08 device_mode
-r--r----- 1 root root 4.0K Jun 15 19:08 device_serial
-r--r----- 1 root root 4.0K Jun 15 19:08 device_type
-rw-rw---- 1 root root 4.0K Jun 15 19:08 dpi
lrwxrwxrwx 1 root root    0 Jun 15 19:07 driver -> ../../../../../../../bus/hid/drivers/razermouse
-r--r----- 1 root root 4.0K Jun 15 19:08 firmware_version
drwxr-xr-x 3 root root    0 Jun 15 19:07 hidraw
drwxr-xr-x 3 root root    0 Jun 15 19:07 input
-rw-rw---- 1 root root 4.0K Jun 15 19:08 logo_led_brightness
--w--w---- 1 root root 4.0K Jun 15 19:08 logo_matrix_effect_breath
--w--w---- 1 root root 4.0K Jun 15 19:08 logo_matrix_effect_none
--w--w---- 1 root root 4.0K Jun 15 19:08 logo_matrix_effect_reactive
--w--w---- 1 root root 4.0K Jun 15 19:08 logo_matrix_effect_spectrum
--w--w---- 1 root root 4.0K Jun 15 19:08 logo_matrix_effect_static
-rw-rw---- 1 root root 4.0K Jun 15 19:08 matrix_brightness
--w--w---- 1 root root 4.0K Jun 15 19:08 matrix_custom_frame
--w--w---- 1 root root 4.0K Jun 15 19:08 matrix_effect_breath
--w--w---- 1 root root 4.0K Jun 15 19:08 matrix_effect_custom
--w--w---- 1 root root 4.0K Jun 15 19:08 matrix_effect_none
--w--w---- 1 root root 4.0K Jun 15 19:08 matrix_effect_reactive
--w--w---- 1 root root 4.0K Jun 15 19:08 matrix_effect_spectrum
--w--w---- 1 root root 4.0K Jun 15 19:08 matrix_effect_static
-r--r--r-- 1 root root 4.0K Jun 15 19:08 modalias
-rw-rw---- 1 root root 4.0K Jun 15 19:08 poll_rate
drwxr-xr-x 2 root root    0 Jun 15 19:08 power
-r--r--r-- 1 root root 4.0K Jun 15 19:08 report_descriptor
-rw-rw---- 1 root root 4.0K Jun 15 19:08 scroll_led_brightness
--w--w---- 1 root root 4.0K Jun 15 19:08 scroll_matrix_effect_breath
--w--w---- 1 root root 4.0K Jun 15 19:08 scroll_matrix_effect_none
--w--w---- 1 root root 4.0K Jun 15 19:08 scroll_matrix_effect_reactive
--w--w---- 1 root root 4.0K Jun 15 19:08 scroll_matrix_effect_spectrum
--w--w---- 1 root root 4.0K Jun 15 19:08 scroll_matrix_effect_static
lrwxrwxrwx 1 root root    0 Jun 15 13:02 subsystem -> ../../../../../../../bus/hid
--w--w---- 1 root root 4.0K Jun 15 19:08 test
-rw-r--r-- 1 root root 4.0K Jun 15 13:01 uevent
-r--r----- 1 root root 4.0K Jun 15 19:08 version
seifer44-lx 0003:1532:0053.0001 # echo -n '255' > ./scroll_led_brightness
seifer44-lx 0003:1532:0053.0001 # echo -n '255' > ./logo_led_brightness
seifer44-lx 0003:1532:0053.0001 # echo -n '255' > ./matrix_brightness
seifer44-lx 0003:1532:0053.0001 # cat ./scroll_led_brightness
255
seifer44-lx 0003:1532:0053.0001 # cat ./logo_led_brightness
255
seifer44-lx 0003:1532:0053.0001 # cat ./matrix_brightness
255
seifer44-lx 0003:1532:0053.0001 # echo -n '128' > ./scroll_led_brightness
seifer44-lx 0003:1532:0053.0001 # echo -n '128' > ./logo_led_brightness
seifer44-lx 0003:1532:0053.0001 # echo -n '128' > ./matrix_brightness
seifer44-lx 0003:1532:0053.0001 # cat ./scroll_led_brightness
128
seifer44-lx 0003:1532:0053.0001 # cat ./logo_led_brightness
128
seifer44-lx 0003:1532:0053.0001 # cat ./matrix_brightness
128
seifer44-lx 0003:1532:0053.0001 # echo -n '255' > ./scroll_led_brightness
seifer44-lx 0003:1532:0053.0001 # echo -n '255' > ./logo_led_brightness
seifer44-lx 0003:1532:0053.0001 # echo -n '255' > ./matrix_brightness
```

terrycain, 2017-06-16:
That looks good, brightness is working.

Will drum up some commands to test the other effects over the weekend

terrycain, 2017-06-25:
```
echo -n '1' > ./matrix_effect_spectrum
# Should make thumbgrid slowly cycle through all colours

echo -ne '\x01\x00\xFF\x00' > ./matrix_effect_reactive
# Should make thumbgrid flash green whne you click buttons on mouse

echo -ne '\x00\xFF\x00\xFF\x00\x00' > ./matrix_effect_breath
# Should make thumbgrid fade between green and red
echo -ne '\x00\x00\xFF' > ./matrix_effect_breath
# Should make thumbgrid fade between blue and black
echo -ne '\x00' > ./matrix_effect_breath
# Should make thumbgrid fade between "random" colours

echo -ne '\xFF\x00\x00' > ./matrix_effect_static
# Should make thumbgrid be a static red
echo -ne '\x00\xFF\x00' > ./matrix_effect_static
# Should make thumbgrid be a static green

echo -n '1' > ./matrix_effect_none
# Should make thumbgrid be unlit / no effect
```

Run each one of these seperatly, "should" change effects

[39 more replies]

closed 2017-09-01 by terrycain