openrazer/openrazer #63: New Razer Blade Stealth (Late 2016) Support

krillr, 2016-10-08:
Would be great to get support for the updated stealth that just came out. As far as I can tell, the only real change is that the USB device ID changed from 0205 to 0220.

krillr, 2016-10-08:
Didn't mean to close this -.-

terrycain, 2016-10-08:
Gimme a week and ill see if I can whip up something.

If you want if you try something in the meantime change [this](https://github.com/terrycain/razer-drivers/blob/master/driver/razerkbd_driver.h#L37) line to use 0x0220 then run `make driver_verbose` then inside of the drivers folder there is a file `razerkbd.ko` so you want to insert it with something like this `sudo insmod ./driver/razerkbd.ko`.

Then read [this](https://github.com/terrycain/razer-drivers/wiki/Setting-up-the-keyboard-driver) and see if you can get the stealth to bind to it.

krillr, 2016-10-08:
I went ahead and threw some code together to add support, it seems to work well! I've created a pull request.

terrycain, 2016-10-08:
Have you tested each and every option?

krillr, 2016-10-08:
So far everything except macros and gaming mode (which I don't think work on any stealth), and custom modes (I haven't tested this), work fine. I'll fiddle about with custom modes tomorrow -- I have no idea how those are supposed to work, but they seem neat.

terrycain, 2016-10-08:
Yeah you might not have a game mode led. If you [this](https://github.com/terrycain/razer-drivers/blob/master/scripts/driver/keyboard_keys.py) script it'll need a bit of editing but should help out with custom mode

krillr, 2016-10-08:
Yup, that worked great. Custom modes verified working!

terrycain, 2016-10-08:
Ok cool, I'll merge it later and turn off the game mode options as not to confuse everyone. One thing to verify is with width of the stealth, when you press enter on that script are there areas of the keyboard that dont light up? There are some gaps in the middle, I'm interested in the edges

krillr, 2016-10-08:
The edge is column 15 (meaning column 16-22 don't light anything).

terrycain, 2016-10-08:
Ahh no numpad? what about column 0?

krillr, 2016-10-08:
Column 0 also fails to light any keys, each row starts at index 1.

terrycain, 2016-10-08:
Would make sense, column 0 would be for the macro keys M1-5 on other keyboards. Ok currently daemon doesnt support an offset in the LED matrix, I'll pull in those changes, go over them again just to make sure they're sane, add support for an offset (so column 1 starts) and then we should be good to go :)

I assume you've tested the daemon?

krillr, 2016-10-08:
Sounds good. Yup, tested the daemon and it works great.

terrycain, 2016-10-09:
Suspend is always dodgy on linux :wink: Merged PR, will release a new update soonish

z3ntu, 2016-10-09:
So close?

terrycain, 2016-10-09:
Yeah might as well
> Btw @terrycain it's awesome to see the speed of development of this application! :)

Yeah :smile:

closed 2016-10-09 by terrycain