openrazer/openrazer #66: Driver Recipe Fails for Fedora 24 Cinnamon

Althrin, 2016-10-13:
After installing dependencies the make for Fedora 24 does not work. As well the dependencies are not labelled correctly in the download page, they are named differently for some packages in Fedora. I have spent an hour trying to find differences in names. Please update your documentation with package names for both Debian **AND** Fedora, not just Debian, it does not help.

[root@localhost LogicTree]# cd razer-drivers
[root@localhost razer-drivers]# sudo make -s all install
\n::\033[32m Compiling Razer kernel modules\033[0m
========================================
Makefile:19: recipe for target 'driver' failed
make: *** [driver] Error 2

z3ntu, 2016-10-13:
I don't think this log says anything interesting. Are there more (interesting) error messages?

Althrin, 2016-10-14:
That would be the point, it throws no errors other than above, and the site is badly documented. No idea where to go from here, but I think the site should be corrected so that other Fedora users do not waste their time.

z3ntu, 2016-10-14:
Why aren't you doing `sudo make fedora_install` (as said in the wiki) ? And about the dependencies: You can take a look at the Arch Linux dependencies which should/could match Fedora's better.
https://aur.archlinux.org/packages/python-razer/
https://aur.archlinux.org/packages/razer-daemon/
https://aur.archlinux.org/packages/razer-driver-dkms/

But I'll try to install Fedora in a VM and try it myself (never used Fedora before^^)

Althrin, 2016-10-14:
Right, so, before today, I was following the website, which looks like, has since been changed between today and yesterday, the `sudo make fedora_install` part was not mentioned on the website's instructions, I also see the dependencies section was removed from the site. I'll try this and update you.

Althrin, 2016-10-14:
Still no luck.

```
[root@localhost razer-drivers]# make fedora_install
\n::\033[31m NOT SUPPORTED YET\033[0m
=====================================================
[root@localhost razer-drivers]# 

```

terrycain, 2016-10-14:
Ok first of all less of the bold yeah it just makes me less likely to help you. Now there's no official rpm's as of yet but it is on the to do list. 

Running `make driver_verbose` will yield more info but most likely you don't have the kernel headers installed. Reboot before you install kernel headers as yum/dnf likes to install headers for the latest kernel which isn't always the running kernel.

When I get home I'll quickly whip up a script that'll build rpm's with `fpm` as I haven't got the time mess with spec files

Althrin, 2016-10-14:
Apologies, when I saw that Fedora was labeled as a supported OS for this, I expected it to not be so anti-fedora in the information, that's all.

terrycain, 2016-10-14:
It used to be before the fork and then I removed it from the makefile as wanted to clean it up and do it right as changed the layout of things but other stuff came up.

terrycain, 2016-10-25:
Sorry mate forgot about this will work on it now

terrycain, 2016-10-25:
Have pushed a branch `feature_fedorafpm` if you check that out and run `make fedora_package` that should result in some RPMs (You'll need `fpm` and `rpmbuild` installed),

The RPMs are totally untested and might not work at all :wink: but should have correct dependencies (best to install it with `yum`/`dnf`, also probably best to install them all at once with `dnf install *.rpm`)

Wont be able to test things myself for a while as am busy but that should be a good start

terrycain, 2016-11-15:
Any luck

z3ntu, 2016-12-07:
Please close in favor of #86

closed 2016-12-24 by terrycain