openrazer/openrazer #67: NO DEVICE in ubuntu 16.10

rivetrik, 2016-10-20:
as the title says, after the last update my razer blade stealth (original) won't connect to the daemon both polychromatic and razer commander state no device connected

rivetrik, 2016-10-20:
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 04f3:223c Elan Microelectronics Corp. 
Bus 001 Device 004: ID 1532:0205 Razer USA, Ltd 
Bus 001 Device 003: ID 0bda:579f Realtek Semiconductor Corp. 
Bus 001 Device 002: ID 0cf3:e300 Atheros Communications, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

terrycain, 2016-10-20:
Hmm, anything in .razer-service/log (the path is something like that). What version of package you using?

rivetrik, 2016-10-21:
yes, there is a rather lengthy log file, you want me to post the whole thing here or on telegram?

terrycain, 2016-10-21:
Err, put the log on pastebin but feel free to continue the discussion on telegram if you want

rivetrik, 2016-10-21:
i can do that tonight when i get home then

rivetrik, 2016-10-22:
alright here is the last bit of the log file i copied after a fresh boot, im assuming this is what you need. if you need the whole thing i can do that as well

<iframe src="//pastebin.com/embed_iframe/mKCGJxWG" style="border:none;width:100%"></iframe>

terrycain, 2016-10-22:
Ok hmm, so the daemon has the razer stealth class which is good. This makes me think the driver isn't loaded
If theres a file `/tmp/razer_udev.log` then the output of that.
Run `lsmod`
And `ls -lah /lib/udev/rules.d/` and `ls -lah /etc/udev/rules.d/` (think the last path is right, its something like that)

And have you installed the kernel driver package (aka what packages did you install)? what ppa are you using?

rivetrik, 2016-10-22:
installed using the ppa and packages directly from https://terrycain.github.io/razer-drivers/#project

the only thing in the temp folder is a folder named tmp_3l37tvgt_razer_daemon

output from lsmod  
Module                  Size  Used by
ccm                    20480  2
rfcomm                 77824  0
arc4                   16384  2
algif_hash             16384  1
algif_skcipher         20480  1
af_alg                 16384  6 algif_hash,algif_skcipher
cmac                   16384  3
uhid                   20480  1
bnep                   20480  2
snd_hda_codec_hdmi     45056  1
snd_hda_codec_realtek    86016  1
snd_hda_codec_generic    73728  1 snd_hda_codec_realtek
hid_rmi                24576  0
8250_dw                16384  0
i2c_designware_platform    16384  0
i2c_designware_core    20480  1 i2c_designware_platform
intel_rapl             20480  0
x86_pkg_temp_thermal    16384  0
intel_powerclamp       16384  0
coretemp               16384  0
snd_soc_skl            65536  0
snd_soc_skl_ipc        45056  1 snd_soc_skl
snd_soc_sst_ipc        16384  1 snd_soc_skl_ipc
kvm_intel             188416  0
snd_soc_sst_dsp        32768  1 snd_soc_skl_ipc
snd_hda_ext_core       28672  1 snd_soc_skl
snd_soc_sst_match      16384  1 snd_soc_skl
snd_soc_core          233472  1 snd_soc_skl
snd_compress           20480  1 snd_soc_core
kvm                   598016  1 kvm_intel
ac97_bus               16384  1 snd_soc_core
snd_pcm_dmaengine      16384  1 snd_soc_core
irqbypass              16384  1 kvm
snd_hda_intel          36864  3
crct10dif_pclmul       16384  0
crc32_pclmul           16384  0
snd_hda_codec         135168  4 snd_hda_intel,snd_hda_codec_hdmi,snd_hda_codec_generic,snd_hda_codec_realtek
ghash_clmulni_intel    16384  0
snd_hda_core           86016  7 snd_hda_intel,snd_hda_codec,snd_hda_ext_core,snd_soc_skl,snd_hda_codec_hdmi,snd_hda_codec_generic,snd_hda_codec_realtek
snd_hwdep              16384  1 snd_hda_codec
aesni_intel           167936  10
nls_iso8859_1          16384  1
aes_x86_64             20480  1 aesni_intel
lrw                    16384  1 aesni_intel
glue_helper            16384  1 aesni_intel
ablk_helper            16384  1 aesni_intel
cryptd                 24576  4 ablk_helper,ghash_clmulni_intel,aesni_intel
snd_pcm               110592  8 snd_hda_intel,snd_hda_codec,snd_pcm_dmaengine,snd_hda_ext_core,snd_hda_core,snd_soc_skl,snd_hda_codec_hdmi,snd_soc_core
joydev                 20480  0
intel_cstate           16384  0
ath10k_pci             45056  0
snd_seq_midi           16384  0
intel_rapl_perf        16384  0
snd_seq_midi_event     16384  1 snd_seq_midi
snd_rawmidi            32768  1 snd_seq_midi
ath10k_core           335872  1 ath10k_pci
ath                    32768  1 ath10k_core
snd_seq                69632  2 snd_seq_midi_event,snd_seq_midi
mac80211              757760  1 ath10k_core
hid_multitouch         20480  0
input_leds             16384  0
razerkbd               32768  0
serio_raw              16384  0
uvcvideo               90112  0
[... 145 lines cut ...]
-rw-r--r-- 1 root root  396 Sep 26 07:24 85-keyboard-configuration.rules
-rw-r--r-- 1 root root 1.7K Sep 27 01:16 85-nm-unmanaged.rules
-rw-r--r-- 1 root root  221 Nov 18  2014 85-regulatory.rules
-rw-r--r-- 1 root root  489 Aug  7 21:08 90-alsa-restore.rules
-rw-r--r-- 1 root root  265 Sep 26 07:24 90-console-setup.rules
-rw-r--r-- 1 root root 1.6K Aug 19 12:32 90-fwupd-devices.rules
-rw-r--r-- 1 root root 1.9K Jun  8 01:02 90-libgpod.rules
-rw-r--r-- 1 root root 1.5K Sep 28 07:44 90-libinput-model-quirks.rules
-rw-r--r-- 1 root root 6.5K Sep 19 13:57 90-pulseaudio.rules
-rw-r--r-- 1 root root  847 Apr 26 09:47 95-cd-devices.rules
-rw-r--r-- 1 root root 2.5K Jun 10 18:18 95-upower-csr.rules
-rw-r--r-- 1 root root 8.0K Jun 10 18:18 95-upower-hid.rules
-rw-r--r-- 1 root root  354 Jun 10 18:18 95-upower-wup.rules
-rw-r--r-- 1 root root 1.5K Sep 17 06:58 97-hid2hci.rules
-rw-r--r-- 1 root root 2.8K Sep 30 17:41 99-razer.rules
-rw-r--r-- 1 root root 3.9K Oct  2 03:33 99-systemd.rules
 

output from ls -lah /etc/udev/rules.d/
total 12K
drwxr-xr-x 2 root root 4.0K Oct 22 07:28 ./
drwxr-xr-x 4 root root 4.0K Oct 12 15:42 ../
-rw-r--r-- 1 root root  631 Jul 12 15:20 mixxx.usb.rules

and yes the kernel module package is installed

terrycain, 2016-10-25:
Ok looks like the udev rule isnt being ran, does `/lib/udev/razer_mount` exist? And can you bind the driver manually / does `sudo modprobe razerkbd` work

rivetrik, 2016-10-26:
yes razer_mount does exsist, and the modprob command does not work, neither does razer-service

terrycain, 2016-10-26:
Ok Razer service is part of the razer-daemon package. And if modprobe didn't work then dkms hasn't run. Mind uninstalling all Razer packages (with --purge) and then reinstalling them (they should be v1.0.17)

rivetrik, 2016-10-26:
alright, i did as you suggested and purged everything then reinstalled. it was the correct version but still no dice. if it helps i noticed it stopped working after the new stealth and blade were released.

terrycain, 2016-10-26:
Yeah I looked around that time and theres nothing that should be stopping anything. Does `modprobe razerkbd` work now?

rivetrik, 2016-10-26:
it does not, still the same no device error

terrycain, 2016-10-26:
Ok dkms must not be working
Run `dkms status`

rivetrik, 2016-10-27:
> dkms status
razer_chroma_driver, 1.0.0, 4.8.0-26-generic, x86_64: installed

terrycain, 2016-10-27:
Run `uname -r` then `sudo modprobe razerkbd; lsmod | grep razer`

Your not running a custom kernel by any chance?

rivetrik, 2016-10-27:
nope, running stock kernel , actually this might help, i ran razer commander with sudo and its not connecting to the daemon , here is the output 

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 175, in activate_name_owner
    return self.get_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 361, in get_name_owner
    's', (bus_name,), **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.razer': no such name

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer/client/__init__.py", line 17, in __init__
    self._dbus = session_bus.get_object("org.razer", "/org/razer")
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 248, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 180, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 278, in start_service_by_name
    'su', (bus_name, flags)))
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.razer was not provided by any .service files

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/share/razercommander/main.py", line 7, in <module>
    import device
  File "/usr/share/razercommander/device.py", line 10, in <module>
    device_manager = rclient.DeviceManager()
  File "/usr/lib/python3/dist-packages/razer/client/__init__.py", line 19, in __init__
    raise DaemonNotFound("Could not connect to daemon")
razer.client.DaemonNotFound: Could not connect to daemon

ill try the command you suggested and report back in a quick sec

terrycain, 2016-10-27:
Ooh the driver is loaded that's good. Run I think it's. `tree /sys/bus/hid/drive

rivetrik, 2016-10-27:
/sys/bus/hid/drivers
├── hid-generic
│   ├── 0003:1532:0205.0001 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:1532:0205.0001
│   ├── 0003:1532:0205.0002 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:1532:0205.0002
│   ├── 0003:1532:0205.0003 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0205.0003
│   ├── 0005:045E:0805.0006 -> ../../../../devices/virtual/misc/uhid/0005:045E:0805.0006
│   ├── bind
│   ├── module -> ../../../../module/hid_generic
│   ├── new_id
│   ├── uevent
│   └── unbind
├── hid-multitouch
│   ├── 0003:04F3:223C.0004 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/0003:04F3:223C.0004
│   ├── bind
│   ├── module -> ../../../../module/hid_multitouch
│   ├── new_id
│   ├── uevent
│   └── unbind
├── hid-rmi
│   ├── 0018:06CB:5F41.0005 -> ../../../../devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-6/i2c-15320205:00/0018:06CB:5F41.0005
│   ├── bind
│   ├── module -> ../../../../module/hid_rmi
│   ├── new_id
│   ├── uevent
│   └── unbind
└── razerkbd
    ├── bind
    ├── module -> ../../../../module/razerkbd
    ├── new_id
    ├── uevent
    └── unbind

14 directories, 16 files

terrycain, 2016-10-30:
Btw if you surround code blocks with triple backticks (`` ` `` ) then it'll print them with monospaced font

Have you tried unbinding them from hid-generic
```
0003:1532:0205.0001
0003:1532:0205.0002
0003:1532:0205.0003
```
and binding them to the driver now (after reinstalling packages)?

rivetrik, 2016-11-02:
i have not, it should be the same as before the ppa { instructions via the wiki right?}

terrycain, 2016-11-02:
What comands did you run?

rivetrik, 2016-11-02:
the ones located here https://github.com/terrycain/razer-drivers/wiki/Setting-up-the-keyboard-driver , those are the right ones correct?

terrycain, 2016-11-02:
Yeah, what error did you get?

rivetrik, 2016-11-03:
no devices found, `Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 175, in activate_name_owner
    return self.get_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 361, in get_name_owner
    's', (bus_name,), **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.razer': no such name

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/razer/client/__init__.py", line 17, in __init__
    self._dbus = session_bus.get_object("org.razer", "/org/razer")
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 248, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 180, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib/python3/dist-packages/dbus/bus.py", line 278, in start_service_by_name
    'su', (bus_name, flags)))
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.razer was not provided by any .service files

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/share/razercommander/main.py", line 7, in <module>
    import device
  File "/usr/share/razercommander/device.py", line 10, in <module>
    device_manager = rclient.DeviceManager()
  File "/usr/lib/python3/dist-packages/razer/client/__init__.py", line 19, in __init__
    raise DaemonNotFound("Could not connect to daemon")
razer.client.DaemonNotFound: Could not connect to daemon

rivetrik, 2016-11-03:
same error, no devices found 

 raise DaemonNotFound("Could not connect to daemon")
razer.client.DaemonNotFound: Could not connect to daemon

terrycain, 2016-11-03:
No I meant when you tried to bind the device to the driver

rivetrik, 2016-11-04:
as far as i know when i bound the driver, it went smoothly, i didn't get an error

terrycain, 2016-11-04:
ok bind the driver again and copy me all the terminal text

rivetrik, 2016-11-11:
`root@rik-Blade-Stealth:/home/rik# ls -l /sys/bus/hid/devices/
total 0
lrwxrwxrwx 1 root root 0 Nov 11 01:25 0003:04F3:223C.0006 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-9/1-9:1.0/0003:04F3:223C.0006
lrwxrwxrwx 1 root root 0 Nov 11 01:25 0003:1532:0205.0003 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:1532:0205.0003
lrwxrwxrwx 1 root root 0 Nov 11 01:25 0003:1532:0205.0004 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:1532:0205.0004
lrwxrwxrwx 1 root root 0 Nov 11 01:25 0003:1532:0205.0005 -> ../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0205.0005
lrwxrwxrwx 1 root root 0 Nov 11 01:25 0005:045E:0805.0008 -> ../../../devices/virtual/misc/uhid/0005:045E:0805.0008
lrwxrwxrwx 1 root root 0 Nov 11 01:25 0018:06CB:5F41.0007 -> ../../../devices/pci0000:00/0000:00:15.1/i2c_designware.1/i2c-6/i2c-15320205:00/0018:06CB:5F41.0007

root@rik-Blade-Stealth:/home/rik# lsusb -v -d 1532:0205

Bus 001 Device 005: ID 1532:0205 Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0205 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Blade Stealth
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          107
    bNumInterfaces          4
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              500mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      61
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
[... 100 lines cut ...]
root@rik-Blade-Stealth:/sys/bus/hid/drivers/hid-generic# ls -l
total 0
lrwxrwxrwx 1 root root    0 Nov 11 01:33 0003:1532:0205.0003 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:1532:0205.0003
lrwxrwxrwx 1 root root    0 Nov 11 01:33 0003:1532:0205.0004 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:1532:0205.0004
lrwxrwxrwx 1 root root    0 Nov 11 01:33 0003:1532:0205.0005 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0205.0005
lrwxrwxrwx 1 root root    0 Nov 11 01:33 0005:045E:0805.0008 -> ../../../../devices/virtual/misc/uhid/0005:045E:0805.0008
--w------- 1 root root 4096 Nov 11 01:28 bind
lrwxrwxrwx 1 root root    0 Nov 11 01:33 module -> ../../../../module/hid_generic
--w------- 1 root root 4096 Nov 11 01:33 new_id
--w------- 1 root root 4096 Nov 11 01:22 uevent
--w------- 1 root root 4096 Nov 11 01:28 unbind
root@rik-Blade-Stealth:/sys/bus/hid/drivers/hid-generic# echo -n "0003:1532:0205.0004" > unbind
root@rik-Blade-Stealth:/sys/bus/hid/drivers/hid-generic# cd /sys/bus/hid/drivers/razerkbd
root@rik-Blade-Stealth:/sys/bus/hid/drivers/razerkbd# echo -n "0003:1532:0205.0004" > bind
root@rik-Blade-Stealth:/sys/bus/hid/drivers/razerkbd# ls -l
total 0
lrwxrwxrwx 1 root root    0 Nov 11 02:48 0003:1532:0205.0004 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:1532:0205.0004
--w------- 1 root root 4096 Nov 11 02:48 bind
lrwxrwxrwx 1 root root    0 Nov 11 01:33 module -> ../../../../module/razerkbd
--w------- 1 root root 4096 Nov 11 01:33 new_id
--w------- 1 root root 4096 Nov 11 01:22 uevent
--w------- 1 root root 4096 Nov 11 01:27 unbind
`

i have rebooted and even tried with the 00 device as well still no luck, razer commander and polychromatic both show no device or daemon connected

terrycain, 2016-11-11:
Well for starters in this instance you should of bound all 3 of those `1532:0205` endpoints :wink:. But the good think here is they do bind to the driver successfully so the issue I believe is with udev. Once it gets autobound to the driver, the daemon will start and then the apps will connect.

Am rather busy atm with some big changes, once they're done I'll look into debugging your udev

claell, 2016-11-12:
I think I am experiencing the same issue with my Abyssus 2014 since the upgrade to Ubuntu 16.10.

terrycain, 2016-11-12:
Oooh it could just be the way 16.10 does udev. Am a bit busy atm with some driver refactoring but will spin up a 16.10 vm to see what it does.

Sorgrum, 2016-11-12:
Also getting the same issue on 16.10

terrycain, 2016-11-13:
Ok pretty much confirmed its a 16.10 issue as am on 16.04 and rebooted earlier and all 5 of my devices loaded. Am nearly finished "the great refactor" so should be able to look at this in a few days/ possibly tomorrow

rivetrik, 2016-11-13:
awesome!

terrycain, 2016-11-13:
Hmm is a very odd issue. Udev doesnt want to run the set of rules at boot but if I trigger udev manually it'll happily bind the devices.

terrycain, 2016-11-14:
Ok Udev is the issue. No clue why it wont work all the tests and debugging says it should just work. Have decided that we shall add more complexity to the mix and I'll just reinvent the wheel and have a service essentially perform what udev does.

I've got 2 devices pretty much nearly ready. Once they're released and the package is out I'll make this my main priority.

Just to try something. Try running 
```
pkill razer-service
# Wait 5 seconds
sudo udevadm trigger
# Wait 5 seconds
razer-service --verbose --run-dir ~/.razer-service/data --log-dir ~/.razer-service/logs --pid-file ~/.razer-service/daemon.pid
```
This `might` work.

ganiserb, 2016-11-17:
I have had the same issue since I installed 16.**04** on my PC. That last three commands you posted worked.

terrycain, 2016-11-17:
Wtf most odd as I spun up a vm on stock 14.04 and it worked fine but 14.10 didn't.

Soonish I'll make a 2nd service that will only deal with bind/unbind

ganiserb, 2016-11-17:
@terrycain Maybe you misread my comment? I mentioned *16*.04. Since the beginning I've been having the following problem: https://github.com/lah7/polychromatic/issues/34

Tray applet randomly failed to start after booting up the PC and trowed a dbus error. I think it's related to this issue. Maybe I'm mistaken.

terrycain, 2016-11-17:
I meant 16.04 and 16.10 sorry. I blame phone keyboard.

sranger, 2016-12-20:
I'm having the same issue in Ubuntu GNOME 16.10. I tried the udevadm fix but it didn't find my devices

terrycain, 2016-12-20:
Am working on a fix. Think I've solved it but havent released it yet.

sranger, 2016-12-20:
Sounds good; thanks for the update

claell, 2016-12-21:
I can confirm that this has been fixed for me with the latest update!

sranger, 2016-12-21:
Daemon is finding my devices as well! Thanks!

Though, they're both receiving updates for their color state when I change the other device's settings. Not sure if it's a daemon issue or not; I posted an issue over at lah7's page (https://github.com/lah7/polychromatic/issues/44)

sranger, 2016-12-21:
Nevermind; didn't realize he had a setting to enable that functionality

terrycain, 2016-12-21:
Nice

closed 2016-12-21 by terrycain

VartanKalad, 2016-12-29:
I have a razer Anansi and have been having the no device found issue for polychromatic controller.  I have installed the dependencies and have been trying to follow the different suggestions here with no luck. I tried purging everything and following your installation guide and everything works fine until I try to bind the device in razerkbd.  When I try to bind it I get the error: bash:echo: write error: No such device.  I am running Linux Mint 18.1.

terrycain, 2016-12-29:
Anasi hasnt been released yet in package form.

VartanKalad, 2016-12-29:
Ok, that makes sense.

rsandoz, 2016-12-30:
razerblade 0x0210 pro 2016 owner here (firefly 0x0c00 and deathadderelite 0x0051)
ubuntu 16.04
firefly showed in polychromatic off the getgo.
mouse and kbd were bound to hid.  rebind last keyboard to your driver and was able to send wave but restarting daemon got red "Daemon not responding" (maybe I need to bind the rest as well.

`rsandoz@rsandoz-Blade-Pro:/lib/udev$ tree /sys/bus/hid/drivers
/sys/bus/hid/drivers
├── hid-generic
│   ├── 0003:1532:005C.000F -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:1532:005C.000F
│   ├── 0003:1532:005C.0010 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:1532:005C.0010
│   ├── 0003:1532:005C.0011 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2/0003:1532:005C.0011
│   ├── 0003:1532:0210.0005 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:1532:0210.0005
│   ├── 0003:1532:0210.0006 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:1532:0210.0006
│   ├── 0003:1532:0210.0007 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0210.0007
│   ├── 0018:0488:D020.0008 -> ../../../../devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-CRQ1001:00/0018:0488:D020.0008
│   ├── bind
│   ├── module -> ../../../../module/hid_generic
│   ├── new_id
│   ├── uevent
│   └── unbind
├── hid-multitouch
│   ├── 0003:04F3:2355.0004 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/0003:04F3:2355.0004
│   ├── bind
│   ├── module -> ../../../../module/hid_multitouch
│   ├── new_id
│   ├── uevent
│   └── unbind
├── razerfirefly
│   ├── 0003:1532:0C00.000C -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.0/0003:1532:0C00.000C
│   ├── 0003:1532:0C00.000D -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.1/0003:1532:0C00.000D
│   ├── 0003:1532:0C00.000E -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-3/1-3:1.2/0003:1532:0C00.000E
│   ├── bind
│   ├── module -> ../../../../module/razerfirefly
│   ├── new_id
│   ├── uevent
│   └── unbind
├── razerkbd
│   ├── bind
│   ├── module -> ../../../../module/razerkbd
│   ├── new_id
│   ├── uevent
│   └── unbind
└── razermouse
    ├── bind
    ├── module -> ../../../../module/razermouse
    ├── new_id
    ├── uevent
    └── unbind

21 directories, 20 files
rsandoz@rsandoz-Blade-Pro:/lib/udev$ cd /sys/bus/hid/drivers/hid-generic
rsandoz@rsandoz-Blade-Pro:/sys/bus/hid/drivers/hid-generic$ ls -l
total 0
lrwxrwxrwx 1 root root    0 Dec 29 21:24 0003:1532:005C.000F -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:1532:005C.000F
lrwxrwxrwx 1 root root    0 Dec 29 21:24 0003:1532:005C.0010 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.1/0003:1532:005C.0010
lrwxrwxrwx 1 root root    0 Dec 29 21:24 0003:1532:005C.0011 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.2/0003:1532:005C.0011
lrwxrwxrwx 1 root root    0 Dec 29 21:24 0003:1532:0210.0005 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:1532:0210.0005
lrwxrwxrwx 1 root root    0 Dec 29 21:24 0003:1532:0210.0006 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:1532:0210.0006
lrwxrwxrwx 1 root root    0 Dec 29 21:24 0003:1532:0210.0007 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:0210.0007
lrwxrwxrwx 1 root root    0 Dec 29 21:24 0018:0488:D020.0008 -> ../../../../devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-0/i2c-CRQ1001:00/0018:0488:D020.0008
--w------- 1 root root 4096 Dec 29 21:24 bind
lrwxrwxrwx 1 root root    0 Dec 29 21:24 module -> ../../../../module/hid_generic
--w------- 1 root root 4096 Dec 29 21:24 new_id
--w------- 1 root root 4096 Dec 29 20:49 uevent
--w------- 1 root root 4096 Dec 29 21:01 unbind
rsandoz@rsandoz-Blade-Pro:/sys/bus/hid/drivers/hid-generic$ echo -n "0003:1532:0210.0007" > unbind
bash: unbind: Permission denied
rsandoz@rsandoz-Blade-Pro:/sys/bus/hid/drivers/hid-generic$ sudo su
root@rsandoz-Blade-Pro:/sys/bus/hid/drivers/hid-generic# echo -n "0003:1532:0210.0007" > unbind
root@rsandoz-Blade-Pro:/sys/bus/hid/drivers/hid-generic# cd /sys/bus/hid/drivers/razerkbd
[... 65 lines cut ...]
-r--r----- 1 root root 4096 Dec 29 21:31 firmware_version
drwxr-xr-x 3 root root    0 Dec 29 21:31 hidraw
drwxr-xr-x 3 root root    0 Dec 29 21:31 input
--w--w---- 1 root root 4096 Dec 29 21:31 logo_led_state
-rw-rw---- 1 root root 4096 Dec 29 21:31 matrix_brightness
--w--w---- 1 root root 4096 Dec 29 21:31 matrix_custom_frame
--w--w---- 1 root root 4096 Dec 29 21:31 matrix_effect_breath
--w--w---- 1 root root 4096 Dec 29 21:31 matrix_effect_custom
--w--w---- 1 root root 4096 Dec 29 21:31 matrix_effect_none
--w--w---- 1 root root 4096 Dec 29 21:31 matrix_effect_reactive
--w--w---- 1 root root 4096 Dec 29 21:31 matrix_effect_spectrum
--w--w---- 1 root root 4096 Dec 29 21:31 matrix_effect_starlight
--w--w---- 1 root root 4096 Dec 29 21:31 matrix_effect_static
--w--w---- 1 root root 4096 Dec 29 21:31 matrix_effect_wave
-r--r--r-- 1 root root 4096 Dec 29 21:31 modalias
drwxr-xr-x 2 root root    0 Dec 29 21:31 power
-r--r--r-- 1 root root 4096 Dec 29 21:31 report_descriptor
lrwxrwxrwx 1 root root    0 Dec 29 21:31 subsystem -> ../../../../../../../bus/hid
-rw-rw---- 1 root root 4096 Dec 29 21:31 test
-rw-r--r-- 1 root root 4096 Dec 29 21:31 uevent
-r--r----- 1 root root 4096 Dec 29 21:31 version
root@rsandoz-Blade-Pro:/sys/bus/hid/drivers/razerkbd/0003:1532:0210.0007# cat device_mode 
0:0
root@rsandoz-Blade-Pro:/sys/bus/hid/drivers/razerkbd/0003:1532:0210.0007# echo -n "1" > matrix_effect_wave 
root@rsandoz-Blade-Pro:/sys/bus/hid/drivers/razerkbd/0003:1532:0210.0007# `

terrycain, 2016-12-30:
Triple backticks are what your after :wink:

What distro you running?
Binding works, you probably need to change the files to be owned by group `plugdev` (and make sure your user is in plugdev group)

rsandoz, 2016-12-30:
ubuntu 16.04 LTS.  I pulled your driver using ppa:terrz/razerutils.  But it seems that is old or at least doesn't contain my mouse (005c) in the source code.

z3ntu, 2016-12-30:
There was no stable version since support was added. Earliest point for next release will be Sunday (according to @terrycain )

rsandoz, 2016-12-31:
I git the latest and manually ran the necessary Makefile updates.
I tried the following test:

Mouse is not plugged in 
```
root@rsandoz-Blade-Pro:/sys/bus/hid# lsmod | grep raz
razerkbd               40960  0
razerfirefly           32768  0
hid                   118784  6 i2c_hid,hid_multitouch,hid_generic,razerkbd,usbhid,razerfirefly
root@rsandoz-Blade-Pro:/sys/bus/hid# find . | grep 1532:005C
```
I plug mouse in (doesn't move pointer)
```
root@rsandoz-Blade-Pro:/sys/bus/hid# lsmod | grep raz
razermouse             40960  0
razerkbd               40960  0
razerfirefly           32768  0
hid                   118784  7 i2c_hid,hid_multitouch,razermouse,hid_generic,razerkbd,usbhid,razerfirefly
root@rsandoz-Blade-Pro:/sys/bus/hid# find . | grep 1532:005C
./devices/0003:1532:005C.0018
./devices/0003:1532:005C.0019
./devices/0003:1532:005C.001A
```

(I notice there are no binds in ./drivers/razermouse, so I try manually)
```
root@rsandoz-Blade-Pro:/sys/bus/hid# cd drivers/razermouse/
root@rsandoz-Blade-Pro:/sys/bus/hid/drivers/razermouse# echo -n "0003:1532:005C.0018" > bind
bash: echo: write error: No such device
root@rsandoz-Blade-Pro:/sys/bus/hid/drivers/razermouse# echo -n "0003:1532:005C.0019" > bind
bash: echo: write error: No such device
root@rsandoz-Blade-Pro:/sys/bus/hid/drivers/razermouse# echo -n "0003:1532:005C.001A" > bind
bash: echo: write error: No such device
```

(I can bind to hid-generic though)
```
root@rsandoz-Blade-Pro:/sys/bus/hid/drivers/razermouse# cd ../hid-generic/
root@rsandoz-Blade-Pro:/sys/bus/hid/drivers/hid-generic# echo -n "0003:1532:005C.0018" > bind
root@rsandoz-Blade-Pro:/sys/bus/hid/drivers/hid-generic# echo -n "0003:1532:005C.0019" > bind
root@rsandoz-Blade-Pro:/sys/bus/hid/drivers/hid-generic# echo -n "0003:1532:005C.001A" > bind
```
(Mouse work fine now, at least until I reboot or replug)

terrycain, 2016-12-31:
Gimme a min, there a new version out soonish

terrycain, 2016-12-31:
Package are building in launchpad as we speak, v1.1.5, purge all razer stuff, install that, reboot and see what happens

[30 more replies]