openrazer/openrazer #68: No such device

materangai, 2016-10-25:
When I run /sys/bus/hid/drivers/razerkbd# echo -n "0003:1532:020F.0001" > bind
The following error is returned: bash: echo: write error: No such device

terrycain, 2016-10-25:
Firstly the title "No such device" isnt helpful. Secondly what distribution you running, what device are you trying to bind, why are you trying to bind manually.

Run `lsusb` but the device your trying to use isnt supported.

materangai, 2016-10-25:
Sorry!!!

I am running Kali, laptop is Blade 14" (thought it was in the supported list)

Trying to bind manually as it was using the hid-generic - if its unsupported I guess that's why.

Output of lsusb:
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 04f3:20b3 Elan Microelectronics Corp. 
Bus 001 Device 005: ID 1532:020f Razer USA, Ltd 
Bus 001 Device 004: ID 0bda:579f Realtek Semiconductor Corp. 
Bus 001 Device 002: ID 0cf3:e300 Atheros Communications, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

terrycain, 2016-10-25:
Yeah the issue with the blade is that theres a fair few different models all with different keyboard PID's. You got access to razer synapse on it? Tbf I could probs allow 1 line in conf and if you install from PPA all should magically work as long as they havent changed much

Kali is deb based now isnt it?

materangai, 2016-10-25:
I do - Windows 10 dual boot. What's needed from it?

It is - Debian Jessie.

terrycain, 2016-10-25:
Screenshot of all options in synapse like what effects are available etc... A zip of them would be preferred but you can drag them into the issue if needed.

materangai, 2016-10-25:
[Synapse.zip](https://github.com/terrycain/razer-drivers/files/551820/Synapse.zip)
Put 3 in there, can do the whole lot but that will be loads of pics.
The standard effects are:
Breathing
Fire
Reactive
Spectrum Cycling
Starlight
Static
Wave
Ripple
None

terrycain, 2016-10-25:
Dont suppose you can get a video of fire, aint seen that one before. 
And whats the exact model of laptop?

materangai, 2016-10-25:
Breathing video - https://goo.gl/photos/CfG9voALt8xm9SpBA 
Its the one that has just come out - The New Razer Blade - QHD+ 512GB

z3ntu, 2016-10-25:
I think @terrycain meant the "Fire" mode ;)

terrycain, 2016-10-25:
Hmm if the fire effect is hardware (which I have a suspicion it is) we'll need packet captures ;)

materangai, 2016-10-25:
Yeah.... not sure how I read fire as breathing but hey ho
https://1drv.ms/u/s!Anon5pTBlCYehc5O6BoirS56pXX6Dg
Bad video - does look quite good in person!
How does one run a packet capture on a keyboard?!

z3ntu, 2016-10-25:
Looks really nice :)

Normally such a packet capture is run from a Linux host with usbmon (that's a kernel module) with Windows with Synapse running inside a virtual machine and running a packet capture from Linux.

Most stuff should be covered [in the wiki](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol).

terrycain, 2016-10-26:
Nice, another effect I'll need to replicate in software :smile:

Thanks @x3ntu :smile: I like how I mention packet capturing and you appear explaining it :-)

z3ntu, 2016-10-26:
I am happy to help :)

ddark-il, 2016-10-26:
There is a [fork ](https://github.com/openrazer/openrazer-drivers)of drivers that seems to have Blade 14 2016 support, maybe you can lookup some stuff from there.

Waiting for replacement Blade(same model as materangai), and i'll try to go with the packet capture

terrycain, 2016-10-26:
That fork is basically a clone of the driver here with some alterations, no new functionality. Current state of the driver can do all the effects bar fire as that I believe is an extended h/w effect which whilst I have the spec for, I haven't come across it "in the wild" before.

@materangai if you try [these](https://drive.google.com/open?id=0BydYgD2LzDUCeHR4SnJFN25BeFU) debs, install them, reboot then run `tree /sys/bus/hid/drivers`

psusmars, 2016-11-15:
I have the same razer and really appreciate what you're doing. Below is the output of the `tree /sys/bus/hid/drivers` after installing the above debs:
`
/sys/bus/hid/drivers
├── hid-generic
│   ├── 0003:1532:020F.0001 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:1532:020F.0001
│   ├── 0003:1532:020F.0002 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:1532:020F.0002
│   ├── 0003:1532:020F.0003 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:020F.0003
│   ├── bind
│   ├── module -> ../../../../module/hid_generic
│   ├── new_id
│   ├── uevent
│   └── unbind
├── hid-rmi
│   ├── 0018:06CB:5F41.0004 -> ../../../../devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-1532020F:00/0018:06CB:5F41.0004
│   ├── bind
│   ├── module -> ../../../../module/hid_rmi
│   ├── new_id
│   ├── uevent
│   └── unbind
└── razerkbd
    ├── bind
    ├── module -> ../../../../module/razerkbd
    ├── new_id
    ├── uevent
    └── unbind

10 directories, 12 files
`

terrycain, 2016-11-15:
You on 16.10 or higher?

psusmars, 2016-11-15:
16.04, I can upgrade to 16.10 if that'd be useful

terrycain, 2016-11-15:
Nope that's fine. There's a bug on 16.10 lol. Hmm they should have binded to the driver. Is there any logs in /tmp named something like razer.log ?

Can you try manually binding those 3 020F devices to the driver. The wiki here - https://github.com/terrycain/razer-drivers/wiki/Setting-up-the-keyboard-driver#binding-the-chroma-to-the-keyboard-driver should say how but bind all 3 if it lets you not just 1

terrycain, 2016-11-15:
Ahh there's a typo in /lib/udev/rules.d/99-razer.rules

Somewhere near line 8 there will be `020F` needs to be `020f` F being lowercase. Try modifing that and rebooting and see if that binds. Otherwise try the manual bind above

psusmars, 2016-11-15:
Looks like it worked:
```
/sys/bus/hid/drivers
├── hid-generic
│   ├── bind
│   ├── module -> ../../../../module/hid_generic
│   ├── new_id
│   ├── uevent
│   └── unbind
├── hid-rmi
│   ├── 0018:06CB:5F41.0004 -> ../../../../devices/pci0000:00/0000:00:15.0/i2c_designware.0/i2c-5/i2c-1532020F:00/0018:06CB:5F41.0004
│   ├── bind
│   ├── module -> ../../../../module/hid_rmi
│   ├── new_id
│   ├── uevent
│   └── unbind
└── razerkbd
    ├── 0003:1532:020F.0001 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:1532:020F.0001
    ├── 0003:1532:020F.0002 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:1532:020F.0002
    ├── 0003:1532:020F.0003 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:020F.0003
    ├── bind
    ├── module -> ../../../../module/razerkbd
    ├── new_id
    ├── uevent
    └── unbind

10 directories, 12 files
```

terrycain, 2016-11-15:
Nice nice nice. Can you run `ls -l` inside each of those `020F` directories. Then we can get onto the testing part.

psusmars, 2016-11-15:
```
020F.0001$ ls -l
total 0
-r--r--r-- 1 root plugdev 4096 Nov 15 10:49 country
lrwxrwxrwx 1 root plugdev    0 Nov 15 10:49 driver -> ../../../../../../../bus/hid/drivers/razerkbd
drwxr-xr-x 3 root plugdev    0 Nov 15 10:49 hidraw
drwxr-xr-x 3 root plugdev    0 Nov 15 10:49 input
-r--r--r-- 1 root plugdev 4096 Nov 15 10:49 modalias
drwxr-xr-x 2 root plugdev    0 Nov 15 10:49 power
-r--r--r-- 1 root plugdev 4096 Nov 15 10:49 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Nov 15 10:49 subsystem -> ../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Nov 15 10:49 uevent
```

```
020F.0002$ ls -l
total 0
-r--r--r-- 1 root plugdev 4096 Nov 15 10:49 country
lrwxrwxrwx 1 root plugdev    0 Nov 15 10:49 driver -> ../../../../../../../bus/hid/drivers/razerkbd
drwxr-xr-x 3 root plugdev    0 Nov 15 10:49 hidraw
drwxr-xr-x 3 root plugdev    0 Nov 15 10:49 input
-r--r--r-- 1 root plugdev 4096 Nov 15 10:49 modalias
drwxr-xr-x 2 root plugdev    0 Nov 15 10:49 power
-r--r--r-- 1 root plugdev 4096 Nov 15 10:49 report_descriptor
lrwxrwxrwx 1 root plugdev    0 Nov 15 10:49 subsystem -> ../../../../../../../bus/hid
-rw-r--r-- 1 root plugdev 4096 Nov 15 10:49 uevent
```

```
020F.0003$ ls -l
total 0
-r--r--r-- 1 root plugdev 4096 Nov 15 10:49 country
-r--r----- 1 root plugdev 4096 Nov 15 10:49 device_type
lrwxrwxrwx 1 root plugdev    0 Nov 15 10:49 driver -> ../../../../../../../bus/hid/drivers/razerkbd
-r--r----- 1 root plugdev 4096 Nov 15 10:49 get_firmware_version
-r--r----- 1 root plugdev 4096 Nov 15 10:49 get_serial
drwxr-xr-x 3 root plugdev    0 Nov 15 10:49 hidraw
drwxr-xr-x 3 root plugdev    0 Nov 15 10:49 input
-r--r--r-- 1 root plugdev 4096 Nov 15 10:49 modalias
--w--w---- 1 root plugdev 4096 Nov 15 10:49 mode_breath
--w--w---- 1 root plugdev 4096 Nov 15 10:49 mode_custom
-rw-rw---- 1 root plugdev 4096 Nov 15 10:49 mode_game
--w--w---- 1 root plugdev 4096 Nov 15 10:49 mode_none
--w--w---- 1 root plugdev 4096 Nov 15 10:49 mode_reactive
--w--w---- 1 root plugdev 4096 Nov 15 10:49 mode_spectrum
--w--w---- 1 root plugdev 4096 Nov 15 10:49 mode_starlight
--w--w---- 1 root plugdev 4096 Nov 15 10:49 mode_static
--w--w---- 1 root plugdev 4096 Nov 15 10:49 mode_wave
drwxr-xr-x 2 root plugdev    0 Nov 15 10:49 power
-r--r--r-- 1 root plugdev 4096 Nov 15 10:49 report_descriptor
--w--w---- 1 root plugdev 4096 Nov 15 10:49 reset
-rw-rw---- 1 root plugdev 4096 Nov 15 10:49 set_brightness
--w--w---- 1 root plugdev 4096 Nov 15 10:49 set_fn_toggle
--w--w---- 1 root plugdev 4096 Nov 15 10:49 set_key_row
--w--w---- 1 root plugdev 4096 Nov 15 10:49 set_logo
lrwxrwxrwx 1 root plugdev    0 Nov 15 10:49 subsystem -> ../../../../../../../bus/hid
--w--w---- 1 root plugdev 4096 Nov 15 10:49 temp_clear_row
-rw-rw---- 1 root plugdev 4096 Nov 15 10:49 test
-rw-r--r-- 1 root plugdev 4096 Nov 15 10:49 uevent
-r--r----- 1 root plugdev 4096 Nov 15 10:49 version
```

terrycain, 2016-11-15:
As I thought its only the last one you really care about ending in `0003`
So go into directory ending in `0003`
Run
```
cat device_type
cat get_firmware_version
cat get_serial
cat version

echo -n '255' > set_brightness                        # Max brightness
cat set_brightness                                    # should say 255
echo -n '128' > set_brightness                        # half brightness
cat set_brightness                                    # should say 128
echo -n '255' > set_brightness                        # back to max brightness

echo -n '1' > mode_wave                               # Should go right or left
echo -n '2' > mode_wave                               # Should go opposite direction to the last

echo -ne '\x01\xFF\x00\x00' > mode_reactive           # Should be red reactive mode

echo -ne '\x00\xFF\xFF' > mode_static                 # Should be cyan colour

echo -n '1' >mode_starlight                           # Should be starlight effect (dunno if the laptop actually supports it

echo -n '1' > mode_spectrum                           # Spectrum mode

echo -n '1' > mode_none                               # No effect

echo -ne '\x00' > mode_breath                         # Random breathing effect
echo -ne '\xFF\x00\x00' > mode_breath                 # Red breathing effect
echo -ne '\xFF\x00\x00\x00\xFF\x00' > mode_breath     # Red-Green breathing effect
```
Do you have a game mode LED on the laptop?

Get me whole terminal output from that. Including command being entered. Put on pastebin.

psusmars, 2016-11-15:
I can confirm that going through the various commands applied the effect, pastebin here: http://pastebin.com/jnstuvWU
I do not have a game mode led on the laptop

terrycain, 2016-11-15:
All the effects worked as they should yeah?

psusmars, 2016-11-15:
Yep, everything worked as expected when doing the `echo`es

terrycain, 2016-11-15:
Awesome. Ok now for the fun bit. Basically during the development of this I made massive changes to the driver. I will merge them (it'll change the names of some of the files) and then we're test again. Then I can make the daemon part which is a simple bit of copy/paste

We also need to test custom mode but will wait for the merge for that as I'll have to write some quality bash string for that or a simple script

goyox86, 2016-11-23:
Hi guys! I recently got a Razer Blade, let me know if you need help testing this! Cheers!

terrycain, 2016-12-03:
Ok have pushed new code to the branch. Its quite possible that I've broken everything. Have also changed the name of some of the files. Can someone run through this please.
```
cat device_type
cat device_mode
cat device_serial
cat firmware_version
cat version

echo -n '255' > matrix_brightness                              # Max brightness
cat matrix_brightness                                          # should say 255
echo -n '128' > matrix_brightness                              # half brightness
cat matrix_brightness                                          # should say 128
echo -n '255' > matrix_brightness                              # back to max brightness

echo -n '1' > matrix_effect_wave                               # Should go right or left
echo -n '2' > matrix_effect_wave                               # Should go opposite direction to the last

echo -ne '\x01\xFF\x00\x00' > matrix_effect_reactive           # Should be red reactive mode

echo -ne '\x00\xFF\xFF' > matrix_effect_static                 # Should be cyan colour

echo -n '1' >matrix_effect_starlight                           # Should be starlight effect (dunno if the laptop actually supports it

echo -n '1' > matrix_effect_spectrum                           # Spectrum mode

echo -n '1' > matrix_effect_none                               # No effect

echo -ne '\x00' > matrix_effect_breath                         # Random breathing effect
echo -ne '\xFF\x00\x00' > matrix_effect_breath                 # Red breathing effect
echo -ne '\xFF\x00\x00\x00\xFF\x00' > matrix_effect_breath     # Red-Green breathing effec
```

psusmars, 2016-12-06:
I'm trying to use the instructions [here](https://github.com/terrycain/razer-drivers/wiki/Setting-up-the-keyboard-driver#binding-the-chroma-to-the-keyboard-driver) to bind the driver that I built from the code on master, however when trying to bind to the driver I get the following (using the information we've learned from above):
```
/sys/bus/hid/drivers/razerkbd $ echo -n "0003:1532:020F.0003" > bind
bash: echo: write error: No such device
```
Before unbinding:
```
/sys/bus/hid/drivers/hid-generic $ ls -l
total 0
lrwxrwxrwx 1 root root    0 Dec  6 12:42 0003:1532:020F.0001 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:1532:020F.0001
lrwxrwxrwx 1 root root    0 Dec  6 12:42 0003:1532:020F.0002 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:1532:020F.0002
lrwxrwxrwx 1 root root    0 Dec  6 12:54 0003:1532:020F.0003 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.2/0003:1532:020F.0003
--w------- 1 root root 4096 Dec  6 12:53 bind
-rwxrwxrwx 1 root root    0 Dec  6 12:42 module -> ../../../../module/hid_generic
--w------- 1 root root 4096 Dec  6 12:42 new_id
--w------- 1 root root 4096 Dec  6 12:42 uevent
--w------- 1 root root 4096 Dec  6 12:49 unbind
```
After unbinding:
```
/sys/bus/hid/drivers/hid-generic $ ls -l
total 0
lrwxrwxrwx 1 root root    0 Dec  6 12:42 0003:1532:020F.0001 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.0/0003:1532:020F.0001
lrwxrwxrwx 1 root root    0 Dec  6 12:42 0003:1532:020F.0002 -> ../../../../devices/pci0000:00/0000:00:14.0/usb1/1-8/1-8:1.1/0003:1532:020F.0002
--w------- 1 root root 4096 Dec  6 12:53 bind
-rwxrwxrwx 1 root root    0 Dec  6 12:42 module -> ../../../../module/hid_generic
--w------- 1 root root 4096 Dec  6 12:42 new_id
--w------- 1 root root 4096 Dec  6 12:42 uevent
--w------- 1 root root 4096 Dec  6 12:49 unbind
```
I started from the beginning of those instructions on that page, and I did a `make driver` inside the directory. Let me know if you need more information

terrycain, 2016-12-06:
It's on a different branch not master

psusmars, 2016-12-06:
I'm an idiot, I'll grab it and check it out later this evening.

psusmars, 2016-12-07:
Using the "feature...blade..." branch, I'm still getting the same `No such device`

terrycain, 2016-12-07:
hmm I'll have a look after work might have lost something when I merged master into the branch

terrycain, 2016-12-07:
Are you sure the correct driver was loaded from feature_blade? as it looks ok to me

psusmars, 2016-12-08:
So I thought I'd start from scratch and nuked everything, and now I'm in a strange situation. After running `sudo make uninstall; sudo make ubuntu_install` and copying the razerkbd.ko to the appropriate place I get that modprobe can't find it:

```
(feature_blade0x020F) $ modprobe razerkbd
modprobe: FATAL: Module razerkbd not found in directory /lib/modules/4.4.0-53-generic
(feature_blade0x020F) $ ls /lib/modules/4.4.0-53-generic/kernel/drivers/usb/misc/ | grep razer
razerkbd.ko
```

terrycain, 2016-12-09:
Most weird, should of asked for .debs as there probably more up-to-date than the makefile. In theory, you should to be able to run `insmod full_path_to_module.ko`

terrycain, 2016-12-09:
I'd insert the module from the git checkout just to be certain

terrycain, 2016-12-09:
And did you `sudo modprobe` ?

psusmars, 2016-12-09:
I did, well, I mean I was in `sudo -i` I'll try the things suggested later today.

terrycain, 2016-12-09:
Ahh yeah sorry have to mention those things as every now and again someone might forget :wink:

terrycain, 2016-12-09:
Also if you want join the group chat on telegram (add me `@terrycain`) as it might be easier for me to troubleshoot then.

psusmars, 2016-12-11:
Using `insmod` solved the problem, here's the pastebin: http://pastebin.com/xA0q5bFG All the effects worked as expected!!

terrycain, 2016-12-11:
Will go and integrate the changes tomorrow.

terrycain, 2016-12-12:
> Using insmod solved the problem...

Just realised I havent added daemon support yet. Adding now, then will need to test that, then it should be all good.

terrycain, 2016-12-12:
I have created some .deb's [here](https://drive.google.com/open?id=0BydYgD2LzDUCeHR4SnJFN25BeFU). Install them, if the keyboard does not mount then bind it manually. And then run the daemon in foreground. It "should" recognise the device and run happily.

```
pkill razer-service
razer-service --foreground --verbose --run-dir ~/.razer-service/data --log-dir ~/.razer-service/logs --pid-file ~/.razer-service/daemon.pid
```

psusmars, 2016-12-23:
I'm not sure it actually bound...:

```
paul@paul-Blade ~ $ pkill razer-service
paul@paul-Blade ~ $ razer-service --foreground --verbose --run-dir ~/.razer-service/data --log-dir ~/.razer-service/logs --pid-file ~/.razer-service/daemon.pid
2016-12-23 14:46:32 | razer                          | INFO     | Initialising Daemon (v1.1.0). Pid: 4153
2016-12-23 14:46:32 | razer.screensaver              | INFO     | Initialising DBus Screensaver Thread
2016-12-23 14:46:32 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2016-12-23 14:46:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2013 --------------------- (1532:011A)
2016-12-23 14:46:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2016 --------------------- (1532:0214)
2016-12-23 14:46:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2016-12-23 14:46:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2016-12-23 14:46:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2016-12-23 14:46:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChromaTournamentEdition - (1532:021A)
2016-12-23 14:46:32 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2016-12-23 14:46:32 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2016-12-23 14:46:32 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2016-12-23 14:46:32 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderChroma ------------------- (1532:0043)
2016-12-23 14:46:32 | razer                          | DEBUG    | Loaded device specification: RazerFireFly ---------------------------- (1532:0C00)
2016-12-23 14:46:32 | razer                          | DEBUG    | Loaded device specification: RazerImperiator ------------------------- (1532:002F)
2016-12-23 14:46:32 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaTE ---------------------- (1532:0046)
2016-12-23 14:46:32 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
2016-12-23 14:46:32 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
2016-12-23 14:46:32 | razer                          | DEBUG    | Loaded device specification: RazerOrnataChroma ----------------------- (1532:021E)
2016-12-23 14:46:32 | razer                          | DEBUG    | Loaded device specification: RazerOrochiWired ------------------------ (1532:0048)
2016-12-23 14:46:32 | razer                          | DEBUG    | Loaded device specification: RazerTartarus --------------------------- (1532:0208)
2016-12-23 14:46:32 | razer                          | INFO     | Adding razer.devices.getDevices method to DBus
2016-12-23 14:46:32 | razer                          | INFO     | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2016-12-23 14:46:32 | razer                          | INFO     | Adding razer.devices.disableTurnOffOnScreensaver method to DBus
2016-12-23 14:46:32 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
2016-12-23 14:46:32 | razer                          | INFO     | Adding razer.daemon.version method to DBus
2016-12-23 14:46:32 | razer                          | INFO     | Adding razer.daemon.stop method to DBus
2016-12-23 14:46:32 | razer                          | INFO     | Serving DBus
2016-12-23 14:46:34 | razer.screensaver              | INFO     | Initialising DBus screensaver object
```

terrycain, 2016-12-23:
You on the latest package? `ls /sys/bus/hid/drivers/razerkbd` and `sudo cat /var/log/syslog | grep -i binding`

terrycain, 2016-12-31:
@psusmars you installed latest package? / is it still broke

psusmars, 2017-01-03:
Sorry, I have been on vacation for the past couple weeks without my laptop, I'll take a look this evening and see if it's been resolved.

terrycain, 2017-01-03:
👍

psusmars, 2017-01-04:
It looks like somehow I had gotten an out of date version of the debs downloaded... Anyways, it looks like it worked, here's the pastebin of the commands you asked to check (plus a report of the razer packages installed): http://pastebin.com/kKuJ7B4Z

terrycain, 2017-01-04:
Try getting latest from ppa stuff should be released now

psusmars, 2017-01-17:
I didn't forget about this, I had to send my laptop back for repairs. Hopefully will get it back sometime soon

terrycain, 2017-01-18:
Np Np

nickdesaulniers, 2017-01-22:
which blade stealth is 1532:020f?  I have a late 2016 and I have 1532:0220.

terrycain, 2017-01-23:
@nickdesaulniers No clue, they should change the names of each one they release instead of me having to guess. Or people should submit their product and model numbers

psusmars, 2017-01-23:
I had a late 2016 Blade (not stealth) 020F. Although my Blade doesn't have
the QHD screen (touch screen) is that what yours is running?

[8 more replies]

closed 2017-02-16 by terrycain