openrazer/openrazer #69: Issue with Daemon(+auto-detect for python)

GovanifY, 2016-10-27:
Hi,
Btw fwiw before going on with this issue it should be noted that pythondir detection should be done automatically, getting real folder of python for exemple(for me it was /usr/lib/64/python3.4 instead of /usr/lib/python3) and using site-packages on not debian based distros(dist-packages is only used on debian based ones while site-packages is used if python was built from source, Gentoo-like)
Anyways on with the real issue

First of all driver works perfectly:  scripts to retrieve infos on my devices works perfectly, the issue is daemon related. If I try to launch any interface to the daemon(ie polychromatic) in normal user mode it won't detect the driver and if I do it in su mode it will just timeout
The log of all the instances of razer-service has always the same exact issue that I can't seem to understand the reason.

> ThinkPadW500 logs # cat razer.log 
2016-10-26 16:05:58 | razer                          | INFO     | Initialising Daemon (v1.0.17). Pid: 5279
2016-10-26 16:05:58 | razer.screensaver              | INFO     | Initialising DBus Screensaver Thread
2016-10-26 16:06:00 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-10-26 16:06:00 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-10-26 16:06:00 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-10-26 16:06:00 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-10-26 16:06:00 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-10-26 16:06:00 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-10-26 16:06:00 | razer.screensaver              | WARNING  | Giving up on screensaver
2016-10-26 17:10:39 | razer                          | INFO     | Initialising Daemon (v1.0.17). Pid: 8066
2016-10-26 17:10:39 | razer.screensaver              | INFO     | Initialising DBus Screensaver Thread
2016-10-26 17:10:41 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-10-26 17:10:41 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-10-26 17:10:41 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-10-26 17:10:41 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-10-26 17:10:41 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-10-26 17:10:41 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-10-26 17:10:41 | razer.screensaver              | WARNING  | Giving up on screensaver
2016-10-27 14:19:14 | razer                          | INFO     | Initialising Daemon (v1.0.17). Pid: 17998
2016-10-27 14:19:14 | razer.screensaver              | INFO     | Initialising DBus Screensaver Thread
2016-10-27 14:19:16 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-10-27 14:19:16 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-10-27 14:19:16 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-10-27 14:19:16 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-10-27 14:19:16 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-10-27 14:19:16 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-10-27 14:19:16 | razer.screensaver              | WARNING  | Giving up on screensaver
2016-10-27 14:49:56 | razer                          | INFO     | Initialising Daemon (v1.0.17). Pid: 6018
2016-10-27 14:49:56 | razer.screensaver              | INFO     | Initialising DBus Screensaver Thread
2016-10-27 14:49:58 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-10-27 14:49:58 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-10-27 14:49:58 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-10-27 14:49:58 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-10-27 14:49:58 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-10-27 14:49:58 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-10-27 14:49:58 | razer.screensaver              | WARNING  | Giving up on screensaver

Thanks in advance!

terrycain, 2016-10-27:
Feel free to pull request pythondir detection, I'd do it but I have more important stuff to get in with when I have the time :wink:

Can you run the daemon in debug mode?

GovanifY, 2016-10-27:
Will do asap for python, I'm stuck between exams atm so not sure I can do it this week. As for debug mode here goes nothing: I've enabled verbose mode and no more debug message pops up moreso since debug is activated by default in my config file, no debug logs either way.
Tho I got a log from the daemon connection scripts that might help(issue seems related to DBus but I don't really understand why DBus doesn't load the config, since I even tried to manually copy the files to dbus-1/system.d and manually restart with no success(actual makefile doesn't copy DBus files)

> govanify@ThinkPadW500 ~/Documents/PROJECTS/PROGRAMMING/SOFTWARES/razer-drivers/scripts/daemon $ python3 controller.py  -l
usage: controller.py [-h] {help,list,brightness,macro} ...
controller.py: error: the following arguments are required: command
govanify@ThinkPadW500 ~/Documents/PROJECTS/PROGRAMMING/SOFTWARES/razer-drivers/scripts/daemon $ python3 controller.py  list
Traceback (most recent call last):
  File "/usr/lib64/python3.4/site-packages/dbus/bus.py", line 175, in activate_name_owner
    return self.get_name_owner(bus_name)
  File "/usr/lib64/python3.4/site-packages/dbus/bus.py", line 361, in get_name_owner
    's', (bus_name,), **keywords)
  File "/usr/lib64/python3.4/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NameHasNoOwner: Could not get owner of name 'org.razer': no such name

>During handling of the above exception, another exception occurred:

>Traceback (most recent call last):
  File "/usr/lib64/python3.4/site-packages/razer/client/__init__.py", line 17, in __init__
    self._dbus = session_bus.get_object("org.razer", "/org/razer")
  File "/usr/lib64/python3.4/site-packages/dbus/bus.py", line 241, in get_object
    follow_name_owner_changes=follow_name_owner_changes)
  File "/usr/lib64/python3.4/site-packages/dbus/proxies.py", line 248, in __init__
    self._named_service = conn.activate_name_owner(bus_name)
  File "/usr/lib64/python3.4/site-packages/dbus/bus.py", line 180, in activate_name_owner
    self.start_service_by_name(bus_name)
  File "/usr/lib64/python3.4/site-packages/dbus/bus.py", line 278, in start_service_by_name
    'su', (bus_name, flags)))
  File "/usr/lib64/python3.4/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.ServiceUnknown: The name org.razer was not provided by any .service files

>During handling of the above exception, another exception occurred:

>Traceback (most recent call last):
  File "controller.py", line 262, in <module>
    run()
  File "controller.py", line 250, in run
    CMD_MAP[args.command](args)
  File "controller.py", line 149, in list_devices
    devices = _get_devices()
  File "controller.py", line 66, in _get_devices
    device_manager = razer.client.DeviceManager()
  File "/usr/lib64/python3.4/site-packages/razer/client/__init__.py", line 19, in __init__
    raise DaemonNotFound("Could not connect to daemon")
razer.client.DaemonNotFound: Could not connect to daemon

terrycain, 2016-10-30:
ok if you run `ps -A` razer-service or razerdaemon (whatever i set it to should appear) if not the daemon isnt running as it shoud, if it is, kill it.

The razer-daemon shouldn't be ran as root, it was designed to run as a normal user so they could attach with it over dbus (and so I can get info from the screensaver state to turn off devices when screens go black).

Once razer-service isnt running run this `/usr/sbin/razer-service --foreground --verbose --run-dir ~/.razer-service/data --log-dir ~/.razer-service/logs --pid-file ~/.razer-service/daemon.pid`

That should just stay in the terminal and not go to the background. See if that has any debug info

GovanifY, 2016-10-30:
Can you plrease reopen for the time beeing? I was actually writing when I saw you auto closed it due to merging of the other issue :p

terrycain, 2016-10-30:
Ahh didnt even realise github did that

GovanifY, 2016-10-30:
>govanify@ThinkPadW500 ~ $ ps -A | grep razer
>7553 ?        00:00:49 razer-service

Was indeed running as root but fyi sbin folder means all that should be executed as root so I'm not sure about the naming scheme here...

So let's go on:
>2016-10-30 20:17:02 | razer.screensaver              | INFO     | Initialising DBus screensaver object
ERROR:dbus.proxies:Introspect error on :1.25:/org/gnome/ScreenSaver: dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownObject: No such object path '/org/gnome/ScreenSaver'
2016-10-30 20:17:02 | razer.screensaver              | ERROR    | Caught exception in run loop
Traceback (most recent call last):
  File "/usr/lib64/python3.4/site-packages/razer_daemon/misc/screensaver_thread.py", line 122, in run
    screensaver_active = self._dbus_interface.GetActive()
  File "/usr/lib64/python3.4/site-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib64/python3.4/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib64/python3.4/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.UnknownObject: No such object path '/org/gnome/ScreenSaver'

haha issue spotted...wait what?
I'm actually running kde but why would dbus check for a gnome folder in the first place?

GovanifY, 2016-10-30:
Ok that gets even weirder, now Polychromatic detects the daemon even tho the issue occured...the fuck

terrycain, 2016-10-30:
Yeah should be moved to /usr/bin at the time it was going to be ran as root but then I changed my mind.

The idea was look for various (not just gnome, also unity...) screensaver DBus endpoints to add some additional functionality like setting brightness to 0 when screen is locked (as screensaver dbus also handles screen locking)

In theory that should only stall that thread and the main service should continue on

terrycain, 2016-10-30:
#MultiThreadded :wink:

terrycain, 2016-10-30:
Can you install `d-feet`, its a nice DBus browser, you could find out the correct DBus endpoint to use and we can go and patch that routine so that screensavery stuff works. 

Also need to add an exception for that so that it doesn't act so scary, its probably one of the catch all try-excepts in there so that I can tell whats going on. I was under the assumption that if gnome wasnt in use the endpoint wouldnt be there.

GovanifY, 2016-10-30:
sure going to check. Wanna me to also move to /usr/bin since it requires no sbin rights?

terrycain, 2016-10-30:
Sure, just make sure you update all references, running a `grep -r` should help

GovanifY, 2016-10-30:
Will do then! will tell you about the updates here when I'll work on it

GovanifY, 2016-10-30:
Not sure this is a polychromatic or daemon issue but posting it here nevertheless: is it normal polychromatic does not auto detect new devices?
2 exemples
I start the daemon without my equipment plugged in, start poly, nothing found, close it, plug it, and restart poly still nothing found
Other one
I start the daemon with my equipment, I then start poly, detects it, then remove the equipemnt while in poly and then daemon does not answer anymore even if I reattach equipment

GovanifY, 2016-10-30:
ok so it's a driver issue. tl;dr dbus: 
if starts with equipment getdevices return ['PM1548136706XXX'] for exemple, then unplug and ['PM1548136706792'], replug and same, equipment does not auto-bind

GovanifY, 2016-10-30:
ok I'm stupid censor wise. Either way try to use another dbus command after unplug and replug:
>'g-io-error-quark: GDBus.Error:org.freedesktop.DBus.Python.FileNotFoundError: Traceback (most recent call last):\n  File "/usr/lib64/python3.4/site-packages/dbus/service.py", line 707, in _message_cb\n    retval = candidate_method(self, *args, **keywords)\n  File "/usr/lib64/python3.4/site-packages/razer_daemon/dbus_services/dbus_methods/mamba.py", line 17, in get_battery\n    with open(driver_path, \'r\') as driver_file:\nFileNotFoundError: [Errno 2] No such file or directory: \'/sys/bus/hid/devices/0003:1532:0045.0029/get_battery\'\n (36)'

terrycain, 2016-10-30:
Hmm, most likely a udev issue then.

GovanifY, 2016-10-30:
if that can apply I'm using eudev from Gentoo main repo, mostly a modded udev for OpenRC use(got rid off systemd crap) but shouldn't be much different. As for the screensaver I know the issue, gonna fix it and do so for sbin

terrycain, 2016-10-30:
eudev shoudnt matter much if you plug in, it works then replug and it doesnt work then run `tree /sys/bus/hid/drivers` I'm thinking either udev isnt rebinding it, or its rebinding it and the daemon isnt recognising the new one

GovanifY, 2016-10-30:
>govanify@ThinkPadW500 ~ $ ls -R /sys/bus/hid/drivers
/sys/bus/hid/drivers:
a4tech  belkin  chicony  ezkey     hid-generic  logitech   monterey  pantherlord  razermouse  sony     topseed
apple   cherry  cypress  gyration  kensington   microsoft  ntrig     petalynx     samsung     sunplus

>/sys/bus/hid/drivers/a4tech:
bind  new_id  uevent  unbind

>/sys/bus/hid/drivers/apple:
bind  module  new_id  uevent  unbind

>/sys/bus/hid/drivers/belkin:
bind  new_id  uevent  unbind

>/sys/bus/hid/drivers/cherry:
bind  new_id  uevent  unbind

>/sys/bus/hid/drivers/chicony:
bind  new_id  uevent  unbind

>/sys/bus/hid/drivers/cypress:
bind  new_id  uevent  unbind

>/sys/bus/hid/drivers/ezkey:
bind  new_id  uevent  unbind

>/sys/bus/hid/drivers/gyration:
bind  new_id  uevent  unbind

>/sys/bus/hid/drivers/hid-generic:
0003:1532:0045.0033  0003:1532:0045.0034  bind  new_id  uevent  unbind

>/sys/bus/hid/drivers/kensington:
bind  new_id  uevent  unbind

>/sys/bus/hid/drivers/logitech:
bind  module  new_id  uevent  unbind

>/sys/bus/hid/drivers/microsoft:
bind  new_id  uevent  unbind

>/sys/bus/hid/drivers/monterey:
bind  new_id  uevent  unbind

>/sys/bus/hid/drivers/ntrig:
bind  module  new_id  uevent  unbind

>/sys/bus/hid/drivers/pantherlord:
bind  new_id  uevent  unbind

>/sys/bus/hid/drivers/petalynx:
bind  new_id  uevent  unbind

>/sys/bus/hid/drivers/razermouse:
0003:1532:0045.0032  bind  module  new_id  uevent  unbind

>/sys/bus/hid/drivers/samsung:
bind  new_id  uevent  unbind

>/sys/bus/hid/drivers/sony:
bind  new_id  uevent  unbind

>/sys/bus/hid/drivers/sunplus:
bind  new_id  uevent  unbind

>/sys/bus/hid/drivers/topseed:
bind  new_id  uevent  unbind
govanify@ThinkPadW500 ~ $ 

Unplug 

> 
> govanify@ThinkPadW500 ~ $ ls -R /sys/bus/hid/drivers
> /sys/bus/hid/drivers:
> a4tech  belkin  chicony  ezkey     hid-generic  logitech   monterey  pantherlord  razermouse  sony     topseed
> apple   cherry  cypress  gyration  kensington   microsoft  ntrig     petalynx     samsung     sunplus
> 
> /sys/bus/hid/drivers/a4tech:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/apple:
> bind  module  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/belkin:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/cherry:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/chicony:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/cypress:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/ezkey:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/gyration:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/hid-generic:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/kensington:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/logitech:
> bind  module  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/microsoft:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/monterey:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/ntrig:
> bind  module  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/pantherlord:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/petalynx:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/razermouse:
> bind  module  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/samsung:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/sony:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/sunplus:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/topseed:
> bind  new_id  uevent  unbind
> 

Replug

> 
> govanify@ThinkPadW500 ~ $ ls -R /sys/bus/hid/drivers
> /sys/bus/hid/drivers:
> a4tech  belkin  chicony  ezkey     hid-generic  logitech   monterey  pantherlord  razermouse  sony     topseed
> apple   cherry  cypress  gyration  kensington   microsoft  ntrig     petalynx     samsung     sunplus
> 
> /sys/bus/hid/drivers/a4tech:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/apple:
> bind  module  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/belkin:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/cherry:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/chicony:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/cypress:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/ezkey:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/gyration:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/hid-generic:
> 0003:1532:0045.0036  0003:1532:0045.0037  bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/kensington:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/logitech:
> bind  module  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/microsoft:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/monterey:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/ntrig:
> bind  module  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/pantherlord:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/petalynx:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/razermouse:
> 0003:1532:0045.0035  bind  module  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/samsung:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/sony:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/sunplus:
> bind  new_id  uevent  unbind
> 
> /sys/bus/hid/drivers/topseed:
> bind  new_id  uevent  unbind
> govanify@ThinkPadW500 ~ $

terrycain, 2016-10-30:
Ok so udev is rebinding it. So interacting with dbus directly you want to call
`org.razer - /org/razer - razer.devices.getDevices()` it should list serial numbers, unplug, replug, wait 30 seconds and see if running the same command prints out the serial numbers.

GovanifY, 2016-10-30:
Will do. Btw the issue with screensaver is that the exception in some way doesn't work correctly and it tries at infinitum the first dbus options instead to loop on others due to the exception. No idea how this is even happening but still investigating

terrycain, 2016-10-30:
:-( Yeah it should only try 3 times.

GovanifY, 2016-10-30:
it does if I remove the break but it does so on the first option which isn't what it should do lol

GovanifY, 2016-10-30:
['PM1548136706792'] first plug, 10s after cuz I'm lazy to wait 30s ['PM1548136706792'] and after unplugging ['PM1548136706792']. Hurray

terrycain, 2016-10-30:
Yeah the daemon will take a few seconds (2 last time I saw, but not sure if I changed it).

Take it polychromatic still doesnt like it when you replug?

terrycain, 2016-10-30:
Anything useful in the log?

GovanifY, 2016-10-30:
> 2016-10-30 21:35:41 | razer                          | DEBUG    | DBus called get_serial_list
2016-10-30 21:36:30 | razer                          | DEBUG    | DBus called get_serial_list
2016-10-30 21:36:33 | razer                          | DEBUG    | DBus called get_serial_list
2016-10-30 21:36:35 | razer                          | DEBUG    | DBus called get_serial_list
2016-10-30 21:36:39 | razer                          | DEBUG    | DBus called get_serial_list
2016-10-30 21:36:39 | razer.device0                  | DEBUG    | DBus call get_device_type
2016-10-30 21:36:39 | razer.device0                  | DEBUG    | DBus call get_device_name

terrycain, 2016-10-30:
Daemon not responding or polychromatic not responding?

GovanifY, 2016-10-30:
poly says daemon does not respond

terrycain, 2016-10-30:
Ahh, that's probably because, when the device is unplugged, the object polychromatic has would be invalid, and when its plugged in there would be a new instance of "the same" device. I think

GovanifY, 2016-10-30:
totally but the daemon still does not auto update(well I think it has done once but very slowly) so it is a daemon issue definitelly

terrycain, 2016-10-30:
Define does not update?

GovanifY, 2016-10-30:
> 2016-10-30 21:39:47 | razer                          | DEBUG    | DBus called get_serial_list
2016-10-30 21:39:47 | razer.device0                  | DEBUG    | DBus call get_device_type
2016-10-30 21:39:47 | razer.device0                  | DEBUG    | DBus call get_device_name
2016-10-30 21:41:04 | razer                          | DEBUG    | DBus called get_serial_list
2016-10-30 21:41:05 | razer.device0                  | DEBUG    | DBus call get_device_type
2016-10-30 21:41:05 | razer.device0                  | DEBUG    | DBus call get_device_name
2016-10-30 21:43:30 | razer                          | WARNING  | Device 0003:1532:0045.004A is missing. Removing from DBus
Ok it does update but takes 5mins to do so(by update I mean update the status of the device)

terrycain, 2016-10-30:
Ok 5 mins to detect the loss of a device or 5 mins to detect the addition of a device after the loss?

GovanifY, 2016-10-30:
both

terrycain, 2016-10-30:
Ha i was supposed to be 5 seconds, not 5 minutes but it seems I changed some intervals and not others.
Have pushed a change to master to set that interval to 10 seconds, or it at least should do

GovanifY, 2016-10-30:
Just a quick question should razer_bcd go to bin too or keep sbin?

terrycain, 2016-10-30:
That shouldn't even exist anymore

GovanifY, 2016-10-30:
from install_files directory. I'll let you fix it then. Time to check on my side

GovanifY, 2016-10-30:
yup new period works! Gonna push the usr/bin fix

terrycain, 2016-10-30:
:)

terrycain, 2016-10-30:
So are we all good?

GovanifY, 2016-10-30:
Still missing the dbus screensaver fix

GovanifY, 2016-10-30:
tl;dr add org.freedesktop.ScreenSaver for kde and other DEs and make it work CORRECTLY

terrycain, 2016-10-30:
Ahh I see, will to that tomorrow. Am beat for today.

terrycain, 2016-10-30:
In the file screensaver_thread.py
Adding `('org.freedesktop.ScreenSaver', '/org/freedesktop/ScreenSaver', 'org.freedesktop.ScreenSaver'),`
to the beginning of `DBUS_OPTIONS` does that fix things.

GovanifY, 2016-10-30:
It should lemme check. Issue is basically it runs only through first and exception is not catched correctly ending up in a huge fail

terrycain, 2016-10-30:
From the stack trace above it starts on line 122 which should mean its already found the "correct" dbus endpoint. Will probably need to add in a check for if the endpoint exists and contains the correct stuff.

Do you have an `com.canonical.Unity` endpoint?

GovanifY, 2016-10-30:
> 2016-10-30 22:31:32 | razer                          | INFO     | Serving DBus
2016-10-30 22:31:34 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-10-30 22:31:42 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)

Seems to work. Btw another lil' unrelated issue but my console prints uncorrectly the compiling message, seems like the color schemes are not applied correctly:
\n::\033[32m Compiling Razer kernel modules\033[0m

GovanifY, 2016-10-30:
I don't have a unity endpoint, nope, only fdreedesktop aka kde

terrycain, 2016-10-30:
Most odd. Will add the freedesktop in first, as technically should be present on all systems.

And not sure, the colours were from the very original makefile it's possible im missing a -e somewhere or just the colour codes are wrong

terrycain, 2016-11-03:
Have committed change to fix freedesktop, will look into that exception soonish

cyanogen, 2016-12-14:
The freedesktop endpoint isn't functional on all systems either. Fixed this in my branch, it's part of the asyncio patch though.

terrycain, 2016-12-15:
Yeah I figured, some endpoints are, some arnt.

opendata26, 2016-12-26:
I have the same issue (
```
Traceback (most recent call last):
  File "/usr/lib/python3.5/site-packages/razer_daemon/misc/screensaver_thread.py", line 123, in run
    screensaver_active = self._dbus_interface.GetActive()
  File "/usr/lib/python3.5/site-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python3.5/site-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3.5/site-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Error.NotSupported: This method is not implemented
```
with gnome on arch linux

terrycain, 2016-12-26:
But in theory the rest works right?

opendata26, 2016-12-27:
Sort of, i have to manually rebind the keyboard (blackwidow x chroma) to razerkbd after each boot and even then polychromatic fails to detect it so i have to control lightning via the terminal.

terrycain, 2016-12-27:
The rebinding is a separate issue, are you using the latest packages? That shoud have fixed binding issues. And if you manually bind you also have to set permissions to plugdev on the folder after binding otherwise the daemon won't have permissions to use that folder

terrycain, 2016-12-27:
as for that stacktrace. I've fixed it in my naga hexv2 branch. It'll get merged around January.

[3 more replies]

PR #70 (2016-10-30): Auto-detect python3 folder
Partly fix #69 , auto-detect python folder instead of hardcoding it
closed 2016-12-31 by terrycain