openrazer/openrazer #72: Razer BlackWidow 2012 RZ03-0038

ashfallen0, 2016-10-29:
I would like to request support for this keyboard in the daemon per:

https://github.com/lah7/polychromatic/issues/37

Relevant lsusb:
----------------------
Bus 003 Device 005: ID 1532:0015 Razer USA, Ltd Naga Mouse
Bus 003 Device 002: ID 1532:010d Razer USA, Ltd 

ls -l /dev/input/by-id
----------------------
total 0
lrwxrwxrwx 1 root root 9 Oct 29 15:18 usb-Razer_Razer_BlackWidow_Ultimate-event-kbd -> ../event4
lrwxrwxrwx 1 root root 9 Oct 29 15:18 usb-Razer_Razer_BlackWidow_Ultimate-if01-event-kbd -> ../event3
lrwxrwxrwx 1 root root 9 Oct 29 15:18 usb-Razer_Razer_BlackWidow_Ultimate-if02-event-mouse -> ../event2
lrwxrwxrwx 1 root root 9 Oct 29 15:18 usb-Razer_Razer_BlackWidow_Ultimate-if02-mouse -> ../mouse0
lrwxrwxrwx 1 root root 9 Oct 29 15:41 usb-Razer_Razer_Naga-event-mouse -> ../event5
lrwxrwxrwx 1 root root 9 Oct 29 15:41 usb-Razer_Razer_Naga-if01-event-kbd -> ../event6
lrwxrwxrwx 1 root root 9 Oct 29 15:41 usb-Razer_Razer_Naga-mouse -> ../mouse1

terrycain, 2016-10-30:
Naga is a while off so am ignoring that till driver is made (it'll get done eventually but will take a while)

The 2012/2013 shouldn't be too difficult as driver should already exist. Will try and add it later.

terrycain, 2016-10-30:
Can you run `ls -lah /dev/input/by-id` for me

terrycain, 2016-10-30:
What OS you running, am thinking to test we can patch the file manually, if it works then I'll merge what I've got

terrycain, 2016-12-31:
Support is now in master, will be in package form soon, next time I do a release

closed 2016-12-31 by terrycain