openrazer/openrazer #77: DeathAdder Chroma support?

meraklimaymun, 2016-11-03:
Can we please have Razer DeathAdder Chroma support?

z3ntu, 2016-11-03:
Duplicate of #23 (probably)

And please post your `lsusb` output of your mouse (to make sure it's the same product).

And a version which is mostly working (just the kernel driver and not the daemon) is available on the branch [feature_DeathAdder0x0043](https://github.com/terrycain/razer-drivers/tree/feature_DeathAdder0x0043)

artemjackson, 2016-11-11:
@z3ntu is it planned to get full DeathAdder support? And if so, when approximately?

z3ntu, 2016-11-11:
Look at #23 where you can see everything I know.

terrycain, 2016-11-11:
> get full DeathAdder support? And if so, when approximately?

Yes and hopefully before christmas ;)

closed 2016-11-11 by terrycain