openrazer/openrazer #78: issue "daemon not responding"

fonzo255, 2016-11-05:
When I opened Polychromatic he said:

**# Daemon Not Responding
Polychromatic could not initialise the daemon. This could be caused by a bug with the daemon processs.
Exception: org.freedsktop.DBus.Error.NoReply: Did noy receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.**

I don't not know what to do.
Pd: I do not have much knowledge in programming, then I need if you help me as if I knew nothing of this

Thank you

terrycain, 2016-11-05:
Contents of ~/.razer-service/logs/razer.log

fonzo255, 2016-11-07:
I don't understand what do you said.

terrycain, 2016-11-07:
Upload what is in that file.

fonzo255, 2016-11-08:
I do not know what happened, but everything was magically solved. Only one thing, when I apply a change from the keyboard icon of the top bar of Ubuntu, when I restart the machine the changes are not saved there. That's fine?
Sorry if you're bothered by so many questions.

terrycain, 2016-11-08:
Did you reboot by any chance? And yeah setting are not saved after a reboot, for now that's down to whatever app your using to deal with that

fonzo255, 2016-11-09:
I reboot the machine when I install it, several times, but there was no case. I don't understand how it worked out now. The only thing that when restart does not save the configuration, but the other excellent.

You are great, thank you very much for the effort to have made this application. Sorry if I disturbed you with my inquiries. My respects

terrycain, 2016-11-09:
No worries and thanks. Yeah I know it doesnt save config yet. I do plan on having something like that but its a long way off yet, I have more devices and easier features to add first :wink:

closed 2016-11-09 by terrycain