openrazer/openrazer #79: Issue "Daemon Not Responding"

skidrow91, 2016-11-06:
Polychromatic not working with me:

** Daemon Not Responding
Polychromatic could not initialise the daemon. This could be caused by a bug with the daemon processs.
Exception: org.freedsktop.DBus.Error.NoReply: Did noy receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.**

This's from razer.log:

`2016-11-06 13:51:06 | razer                          | INFO     | Initialising Daemon (v1.0.17). Pid: 1857
2016-11-06 13:51:06 | razer.screensaver              | INFO     | Initialising DBus Screensaver Thread
2016-11-06 13:51:07 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2016-11-06 13:51:07 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2013 --------------------- (1532:011A)
2016-11-06 13:51:07 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2016 --------------------- (1532:0214)
2016-11-06 13:51:07 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2016-11-06 13:51:07 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2016-11-06 13:51:07 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2016-11-06 13:51:07 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChromaTournamentEdition - (1532:021A)
2016-11-06 13:51:07 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2016-11-06 13:51:07 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2016-11-06 13:51:07 | razer                          | DEBUG    | Loaded device specification: RazerFireFly ---------------------------- (1532:0C00)
2016-11-06 13:51:07 | razer                          | DEBUG    | Loaded device specification: RazerImperiator ------------------------- (1532:002F)
2016-11-06 13:51:07 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaTE ---------------------- (1532:0046)
2016-11-06 13:51:07 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
2016-11-06 13:51:07 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
2016-11-06 13:51:07 | razer                          | DEBUG    | Loaded device specification: RazerOrochiWired ------------------------ (1532:0048)
2016-11-06 13:51:07 | razer                          | DEBUG    | Loaded device specification: RazerTartarus --------------------------- (1532:0208)
2016-11-06 13:51:07 | razer                          | INFO     | Found device.0: 0003:1532:0214.0004
2016-11-06 13:51:07 | razer.device0                  | INFO     | Initialising device.0 RazerBlackWidow2016
2016-11-06 13:51:08 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-11-06 15:16:44 | razer                          | INFO     | Initialising Daemon (v1.0.17). Pid: 1918
2016-11-06 15:16:44 | razer.screensaver              | INFO     | Initialising DBus Screensaver Thread
2016-11-06 15:16:44 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2016-11-06 15:16:44 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2013 --------------------- (1532:011A)
2016-11-06 15:16:44 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2016 --------------------- (1532:0214)
2016-11-06 15:16:44 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2016-11-06 15:16:44 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2016-11-06 15:16:44 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2016-11-06 15:16:44 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChromaTournamentEdition - (1532:021A)
2016-11-06 15:16:44 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2016-11-06 15:16:44 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2016-11-06 15:16:44 | razer                          | DEBUG    | Loaded device specification: RazerFireFly ---------------------------- (1532:0C00)
2016-11-06 15:16:44 | razer                          | DEBUG    | Loaded device specification: RazerImperiator ------------------------- (1532:002F)
2016-11-06 15:16:44 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaTE ---------------------- (1532:0046)
2016-11-06 15:16:44 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
2016-11-06 15:16:44 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
2016-11-06 15:16:44 | razer                          | DEBUG    | Loaded device specification: RazerOrochiWired ------------------------ (1532:0048)
2016-11-06 15:16:44 | razer                          | DEBUG    | Loaded device specification: RazerTartarus --------------------------- (1532:0208)
2016-11-06 15:16:44 | razer                          | INFO     | Found device.0: 0003:1532:0214.0004
2016-11-06 15:16:44 | razer.device0                  | INFO     | Initialising device.0 RazerBlackWidow2016
2016-11-06 15:16:46 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-11-06 15:16:46 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-11-06 15:16:46 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-11-06 15:16:46 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-11-06 15:16:47 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-11-06 15:16:47 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-11-06 15:16:47 | razer.screensaver              | WARNING  | Giving up on screensaver
2016-11-06 15:25:59 | razer                          | INFO     | Initialising Daemon (v1.0.17). Pid: 2016
2016-11-06 15:25:59 | razer.screensaver              | INFO     | Initialising DBus Screensaver Thread
2016-11-06 15:25:59 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2016-11-06 15:25:59 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2013 --------------------- (1532:011A)
2016-11-06 15:25:59 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2016 --------------------- (1532:0214)
2016-11-06 15:25:59 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2016-11-06 15:25:59 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2016-11-06 15:25:59 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2016-11-06 15:25:59 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChromaTournamentEdition - (1532:021A)
2016-11-06 15:25:59 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2016-11-06 15:25:59 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
[... cut ...]
2016-11-06 15:25:59 | razer                          | DEBUG    | Loaded device specification: RazerOrochiWired ------------------------ (1532:0048)
2016-11-06 15:25:59 | razer                          | DEBUG    | Loaded device specification: RazerTartarus --------------------------- (1532:0208)
2016-11-06 15:25:59 | razer                          | INFO     | Found device.0: 0003:1532:0214.0004
2016-11-06 15:25:59 | razer.device0                  | INFO     | Initialising device.0 RazerBlackWidow2016
2016-11-06 15:26:01 | razer.screensaver              | INFO     | Initialising DBus screensaver object`

z3ntu, 2016-11-06:
Please use triple backticks ` at the start and end of the block because otherwise it's pretty unreadable.

skidrow91, 2016-11-06:
@z3ntu: Edited

terrycain, 2016-11-06:
Stop the daemon (should be able to find the PID its probably called `razer-service`) then run
`id $USER`
And then run
`razer-service --foreground --verbose --run-dir ~/.razer-service/data --log-dir ~/.razer-service/logs --pid-file ~/.razer-service/daemon.pid`
and run
`tree /sys/bus/hid/drivers`
And give output of all of them. (You might need to install `tree`)

skidrow91, 2016-11-06:
After run `tree /sys/bus/hid/drivers` return:

> /sys/bus/hid/drivers
├── hid-generic
│   ├── 0003:04D9:1605.0005 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-10/3-10:1.0/0003:04D9:1605.0005
│   ├── 0003:04D9:1605.0006 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-10/3-10:1.1/0003:04D9:1605.0006
│   ├── 0003:093A:2510.0001 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-5/3-5:1.0/0003:093A:2510.0001
│   ├── 0003:1532:0214.0002 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6:1.0/0003:1532:0214.0002
│   ├── 0003:1532:0214.0003 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6:1.1/0003:1532:0214.0003
│   ├── 0003:1532:0214.0004 -> ../../../../devices/pci0000:00/0000:00:14.0/usb3/3-6/3-6:1.2/0003:1532:0214.0004
│   ├── bind
│   ├── module -> ../../../../module/hid_generic
│   ├── new_id
│   ├── uevent
│   └── unbind
└── razerkbd
    ├── bind
    ├── module -> ../../../../module/razerkbd
    ├── new_id
    ├── uevent
    └── unbind

terrycain, 2016-11-06:
What distro you running? How did you install? and would you mind rebooting and then rerunning the tree command.

skidrow91, 2016-11-06:
I use Ubuntu 16.04 LTS.
It worked after reboot. thanks @terrycain .

terrycain, 2016-11-06:
All good yeah? Close issue?

closed 2016-11-06 by skidrow91

z3ntu, 2016-11-06:
@terrycain You should add a **Troubleshooting** section to the readme where you tell people to reboot and trying it again before reporting an issue :)

terrycain, 2016-11-06:
@z3ntu There are some words somewhere saying to reboot, on readme here
> Before raising an issue saying something doesn't work, read this Wiki page, try not to create new issues if one exists, reopen it.

Needs to be expanded though.

Don't like the github issue templates as they get in the way if you want to raise an issue which isnt "it dont work"