openrazer/openrazer #81: Razer Mamba missing support.

nator666, 2016-11-18:
Hi! Can you add support for wireless Razer Mamba. 

lsusb -d 1532:0025 -v

Bus 001 Device 002: ID 1532:0025 Razer USA, Ltd 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            0 (Defined at Interface level)
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x1532 Razer USA, Ltd
  idProduct          0x0025 
  bcdDevice            2.00
  iManufacturer           1 Razer
  iProduct                2 Razer Mamba Charging dock
  iSerial                 0 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           59
    bNumInterfaces          2
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xa0
      (Bus Powered)
      Remote Wakeup
    MaxPower              100mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      2 Mouse
      iInterface              0 
        HID Device Descriptor:
          bLength                 9
          bDescriptorType        33
          bcdHID               1.11
          bCountryCode            0 Not supported
          bNumDescriptors         1
          bDescriptorType        34 Report
          wDescriptorLength      94
         Report Descriptors: 
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0008  1x 8 bytes
        bInterval               1
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        1
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         3 Human Interface Device
      bInterfaceSubClass      1 Boot Interface Subclass
      bInterfaceProtocol      1 Keyboard
[... 102 lines cut ...]
           ** UNAVAILABLE **
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0009  1x 9 bytes
        bInterval               1
Device Status:     0x0000
  (Bus Powered)

ls -l /dev/input/by-id
total 0
lrwxrwxrwx 1 root root  9 nov   18 21:51 usb-Razer_Razer_Mamba_Charging_dock-event-mouse -> ../event7
lrwxrwxrwx 1 root root  9 nov   18 21:51 usb-Razer_Razer_Mamba_Charging_dock-if01-event-kbd -> ../event8
lrwxrwxrwx 1 root root  9 nov   18 21:51 usb-Razer_Razer_Mamba_Charging_dock-mouse -> ../mouse0

ls -l /dev/input/by-id
total 0
lrwxrwxrwx 1 root root  9 nov   18 21:54 usb-Razer_Razer_Mamba-event-mouse -> ../event7
lrwxrwxrwx 1 root root  9 nov   18 21:54 usb-Razer_Razer_Mamba-if01-event-kbd -> ../event8
lrwxrwxrwx 1 root root  9 nov   18 21:54 usb-Razer_Razer_Mamba-mouse -> ../mouse0

terrycain, 2016-11-19:
> What I normally do is go through the Razer Synapse app and take screenshots of all the various features and buttons you can click on.

Get me that then I can make a start.

nator666, 2016-11-21:
Here are the screenshots. Thank you!

[image]
[image]
[image]
[image]
[image]
[image]

terrycain, 2016-12-31:
Hey, you still after this / able to help work on support?

nator666, 2016-12-31:
I'm still after this, how can I help?

terrycain, 2016-12-31:
Need some packet captures, you capable of doing [this](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol)?

brwyatt, 2017-01-17:
So this is weird. The drivers seem to claim support for the Mamba (wired and wireless), but with IDs 1532:0044 and 1532:0045. However, mine (as well as nator666's) shows up as 1532:0024 and 1532:0025.

I'm guessing this might be because the DEBUG logging seems to indicate support for Mamba Chroma, so it looks like the regular Mamba (non-Chroma) isn't getting picked up.

I might be able to assist with USB debugging, though I really only have a demo install of Windows 10 available (I haven't had a valid Windows license in a long time).

terrycain, 2017-01-17:
@brwyatt Demo install "should" work fine as long as razer synapse works, is it a vm?

And yes your right the chroma one works, it didnt have chroma in the name as that was what it was called on the razer site at the time

brwyatt, 2017-01-18:
@terrycain It is a VM. I only have Ubuntu hosts, but do have a Windows 10 (demo) VM. I can take a look at this this weekend, if I can find time.

terrycain, 2017-01-18:
Yeah it should be fine. And no rush mate

Raboebie, 2017-02-17:
Hi Everyone. I'm facing the exact same problem. PID is listed as 0024. It's a Razor Mamba of that I'm sure :). Any update on this issue?

terrycain, 2017-02-17:
Waiting on packet captures I think

brwyatt, 2017-02-19:
I managed to get time to do this today. Did everything (I hope/think) under lighting and power (that was configurable), and the basics under performance (enable/disable XY sensitivity, sensitivity, and polling rate; Acceleration doesn't seem to be sent to the device, and I did not do independent xy sensitivities or sensitivity stages).

Captures are here: http://home.brwyatt.net/files/RazerMamba2012.zip

I'm hoping this is what you needed, but let me know if there's something not right about them. I can also go ahead and get the separate XY sensitivities and the sensitivity stages as well. Additional note: I didn't capture all 63 different values for the sensitivities, but I'm hoping/assuming they can be extrapolated from the provided data. If not, let me know and I can get the remaining ones for you.

terrycain, 2017-02-20:
Looks good, could of done 1 capture per effect, you could of combined all of the colour ones into one capture :wink:, but the beauty of pcapng is i can just `cat` them all together

I dont suppose I can get a packet capture of the entire startup stopping when synapse is opened? In theory synapse should query the device for a serial.

Also on wireless could I get a capture of like 10 mins of no movement/no clicks, synapse should query the mouse for its battery status and if its plugged in charging or not.

terrycain, 2017-02-20:
Am pretty sure I know what the serial command is but having captures is nice for my records at least in case i ever need to revisit stuff. When I get home should be able to get to work and have it mostly functional. Assuming you are willing to test, what distro are you on?

Raboebie, 2017-02-20:
I'm also available to test if you require more than one participant. I'm running Ubuntu Gnome 16.04 x64.

terrycain, 2017-02-20:
Sure, I'll basically have a list of things to get through so the more people the quicker it gets done

Raboebie, 2017-02-20:
Excellent.

terrycain, 2017-02-20:
Can you change dock colour?

Raboebie, 2017-02-20:
If you mean from within Synapse then no, you can't change the dock colour. It's random

terrycain, 2017-02-20:
Ahh sad times

terrycain, 2017-02-20:
@Raboebie Install debs from here - https://drive.google.com/drive/folders/0BydYgD2LzDUCdE1TSENkY0NGSzg?usp=sharing

We'll test with mamba on wireless first
Then reboot, then 
```
ls -lah /sys/bus/hid/drivers/
ls -lah /sys/bus/hid/drivers/razermouse/
ls -lah /sys/bus/hid/drivers/razermouse/0*/
```

Raboebie, 2017-02-20:
Thank you! I'll give it a bash tomorrow and let you know.

Raboebie, 2017-02-21:
Installed the debs. Attached are the outputs from the commands you asked for.

[commands.zip](https://github.com/terrycain/razer-drivers/files/790531/commands.zip)

Raboebie, 2017-02-21:
For interests sake, I have been using xinput to disable to acceleration of the mounse and change the sensitivity below normal. Is that possible to do with this driver? (If you mouse is supported)

z3ntu, 2017-02-21:
Acceleration should be something with Xorg but dpi is possible with most (if not all) mice

terrycain, 2017-02-21:
Mouse supports DPI so yep (acceleration is a software thing). We'll test some more stuff, then test the daemon, then should be done. The front end apps "should" allow you to change dpi (if they have added the option)

Raboebie, 2017-02-21:
Oh yes I know that I was just wondering if there was an "easy" way to disable it with the daemon. I was hoping for that amazing feeling after you install synapse and the acceleration is all gone... ;). But with xorg there's never really an easy way to do anything without messing up something else in my experience. Thank you for your efforts on this. It's much appreciated!

terrycain, 2017-02-21:
No problem. Thanks for helping test. I've never seen accleration on any mice I have. What xinput stuff are you running?

Raboebie, 2017-02-21:
So what I'm doing currently is using xinput to identify the device and list-props to get the properties. I then use set-prop like every ubuntu thread about mice would suggest but since I updated to 16.04 it doesn't work anymore. Editing the xorg.conf file directly and restarting X also doesn't work. So I'm stuck with a mouse that jumps across the screen when I breathe too heavily :P. Interestingly enough if I use the Mamba in wireless more I can't use xinput to list any of iets properties. It only lists the dock but it lists the dock twice. Since it sees it as the dock now Acceleration and Deceleration aren't "valid" properties anymore. If in wired mode I can see the properties but changing them has no effect.

terrycain, 2017-02-21:
To be fair I think just lowering the dpi should solve that or at least make it more useable ;-)

Raboebie, 2017-02-21:
Lets hope so :)

Raboebie, 2017-02-21:
The good news is I managed to change the DPI of my mouse using the scripts in in the project. Finally, I have a useable desktop.

terrycain, 2017-02-21:
@Raboebie :-)

terrycain, 2017-02-21:
sudo to root
cd to `/sys/bus/hid/drivers/razermouse/0003:1532:0025.0005/` or which ever one contains `dpi`

```
cat charge_level
cat charge_status
cat device_mode
cat device_serial
cat device_type
cat dpi
cat firmware_version
cat poll_Rate
cat scroll_led_brightness
cat scroll_led_effect
cat scroll_led_rgb
cat scroll_led_state
cat version
```

Raboebie, 2017-02-22:
I'll run them for you as soon as I get home.

Raboebie, 2017-02-22:
See attached.
[outputs.zip](https://github.com/terrycain/razer-drivers/files/793798/outputs.zip)

terrycain, 2017-02-22:
@Raboebie Would be easier if you just pasted the output into a code block

Raboebie, 2017-02-22:
Sorry, didn't want to spam up the thread.

terrycain, 2017-02-22:
@Raboebie Ahh thats fine

terrycain, 2017-02-22:
Did cat device_serial not work?

```
# Set brightness of scroll wheel to max
echo -n '255' > scroll_led_brightness
# Half
echo -n '128' > scroll_led_brightness
# Quarter brightness
echo -n '64' > scroll_led_brightness
# Max
echo -n '255' > scroll_led_brightness

#Hopefully red
echo -ne '\xFF\x00\x00' > scroll_led_rgb
echo -ne '0' > scroll_led_effect

# Hopefully specturm
echo -ne '4' > scroll_led_effect
```

Raboebie, 2017-02-22:
Everything that came back blank was blank when I ran it. I'll try again
tomorrow to make sure. I'll run these tomorrow morning and let you know.

terrycain, 2017-02-22:
@Raboebie 👍cool

Raboebie, 2017-02-23:
All of those commands worked as expected, except for the last one. The last one disabled the lighting completely and on the second run did nothing. Something that's pretty cool is that when the device is in wireless more the dock actually changes colour along with the mouse when I run these commands. And yes, cat device_serial does not work.

tildearrow, 2017-02-23:
I have a question. Why "scroll_led_" and not "matrix_effect_"? The Orochi, having 1 LED like this Mamba, uses " matrix_effect_" files.

z3ntu, 2017-02-23:
It's a matter of how the firmware of the device is made afaik

terrycain, 2017-02-23:
@z3ntu Yep exactly. It depends on how things are done in firmware

brwyatt, 2017-02-24:
Just getting a chance to look back at this. Do you still need further captures, @terrycain? It doesn't look like anyone else was able to provide any, but wanted to make sure. Sounds like you had enough to at least get together at least a small test driver.

If I'm not mistaken, the things you are looking for still are:

1. Captures of the startup of synapse
2. 10 minutes of no movement on wireless
3. 10 minutes of no movement while wired (It kinda sounded like this could be useful, too, was hard to tell in context for this one)

terrycain, 2017-02-24:
@brwyatt Yh

hobarrera, 2017-04-14:
Hi!

When I move my mamba's wheel to the side, the event is sends it the same as scrolling the wheel multiple times (checked this closely with `xev`).

Would adding a possible fix for the be part of this issue, or is that out of scope here? If it's part of this issue: how can I help?

I'll go ahead and start fetching a windows ISO so I can set up a VM, since I'm guessing I'll probably need data from there, right?

terrycain, 2017-04-14:
@hobarrera there is code to fix that but its waiting on other stuff but its on my list don't worry :)

hobarrera, 2017-04-14:
Thanks! Do count on my if your need any debug/test data.

terrycain, 2017-04-14:
Sure, can you get telegram and add me `@terrycain`, most of the developers hang out there and I'll have some macro based stuff to test (which will include this bit) soon

brwyatt, 2017-04-14:
Hey guys, apologies for dropping the ball on this one. I'm going to try and get the additional stuff that was requested done today after I get home from work.

terrycain, 2017-04-14:
@brwyatt Mate no rush. Though if your around tomorrow to test could get it in the Sunday release

brwyatt, 2017-04-15:
Synapse startup, 10 minutes wireless with no movement, 10 minutes wired with no movement (little bit of something 40 seconds in, but 10 minutes following that)

[Razer Mamba 2012 - 2.zip](https://github.com/terrycain/razer-drivers/files/923226/Razer.Mamba.2012.-.2.zip)

Hopefully that works for what you need! Let me know if you need anything further!

terrycain, 2017-04-15:
Ok anyone, get debs from here - https://drive.google.com/open?id=0BydYgD2LzDUCdE1TSENkY0NGSzg

Install them, reboot then either respond here, or on telegram if your on there.

hobarrera, 2017-04-15:
Debs don't really work for me, but I can easily work from the source branch if you point me to it. :)

z3ntu, 2017-04-15:
@hobarrera What distro are you using? I can make you easy instructions for RPM and Arch Linux.

hobarrera, 2017-04-15:
I'm using ArchLinux. I don't think I need further instructions (since I can just tweak existing PKGBUILDs), I mostly need a confirmation of which branch this is based of and I can probably work from there.

z3ntu, 2017-04-15:
Great, use https://aur.archlinux.org/pkgbase/razer-drivers-git/ as a base and add a `#branch=branchname` to the end of the source field.
I don't think @terrycain put the branch online yet but hopefully will soon.

[27 more replies]

closed 2017-04-15 by terrycain