openrazer/openrazer #86: fedora_install doesn't support Fedora25

Havilland, 2016-11-28:
Hi,
I just tried to install the razer-drivers on my Blade Stealth with Kaby Lake using Fedora 25. But the make fedora_install just quit with not supported yet. Is there a plan when F25 will be supported?

havilland

terrycain, 2016-12-01:
Tbf it shoudln't be too difficult to add it now. Making proper RPMs and hosting them is another story though as would need somewhere to host them as I dont really want to just keep committing binaries to the repo.

terrycain, 2016-12-01:
Checkout branch `feature_fedorafpm`. then run `make fedora_package` it should result in some RPMs, which may or may not work. I have a feeling the DKMS bit will take some more work.

You'll need `fpm` for it to work so go find that.

Install RPMs, reboot and run `sudo modprobe razerkbd` it should just work, if not then we need to work on dkms

z3ntu, 2016-12-07:
## How to install **manually** (not recommended but probably the only option right now)

```
git clone https://github.com/terrycain/razer-drivers.git
cd razer-drivers
# Install dependencies
sudo dnf install kernel-devel python3-setproctitle python3-numpy
sudo pip3 install notify2
# Compile
sudo make -s all install
# Add yourself to the `plugdev` group
sudo groupadd --system plugdev
# Please replace <USERNAME> with your username!
sudo gpasswd -a <USERNAME> plugdev
# Reboot and everything should be working as expected. Tested on a fresh Fedora 25 install in QEMU.
```

## How to install via dnf/rpm (untested/not fully working)
```
# Repo should already be cloned
# Install fpm
sudo dnf install rubygems ruby-devel libffi-devel redhat-rpm-config
gem install fpm
# Set PATH for fpm (has to be done every time you open the terminal or stick it in your `.bashrc` file)
export PATH=$HOME/.gem/ruby/gems/fpm-1.7.0/bin:$PATH
# Make package
make fedora_package
# Install the resulting rpms
sudo dnf install *.rpm
```

terrycain, 2016-12-31:
We have proper rpms now. see website - https://terrycain.github.io/razer-drivers/

closed 2016-12-31 by terrycain