openrazer/openrazer #87: upower support

z3ntu, 2016-11-28:
> UPower is an abstraction for enumerating power devices, listening to device events and querying history and statistics.

https://upower.freedesktop.org/

terrycain, 2016-11-28:
This'll be good but its going to be low on the priority list

z3ntu, 2018-03-25:
Actually this might really be possible. When doing something with /sys/bus/hid/devices/<gamepad>/power_supply/ , see https://github.com/atar-axis/xpadneo/issues/21 the battery should be registered

Aasikki, 2022-11-12:
This would be really nice to have. It's annoying to have to quess my mouses battery level.

babbaj, 2023-11-10:
implemented this in https://github.com/openrazer/openrazer/pull/2160 with a power_supply. The xpadneo code was a very helpful reference.