openrazer/openrazer #88: NO DEVICE FOUND with AUR install (using yaourt)

doesnotcommute, 2016-11-29:
Tried modprobe and lsmod, can't find the driver.
Manual install fails during make (in path with no white spaces)
Output of make: http://pastebin.com/UCSdryWX

doesnotcommute, 2016-11-29:
PS - Device is Razer Blade Stealth 2016

terrycain, 2016-12-01:
run `lsusb` and paste output. And the make you ran was on code not from this repository.

doesnotcommute, 2016-12-02:
lsusb output: http://pastebin.com/jxmfwfjj
Sorry about that. Must have cloned a different repo (there are a couple of them with similar names). will fix it

terrycain, 2016-12-02:
@GabMus @z3ntu Mind helping troubleshooting the AUR install?

doesnotcommute, 2016-12-02:
Update : ran sudo make -s all install on the code from this repository
Output:

:: Compiling Razer kernel modules
========================================
make: *** [Makefile:25: driver] Error 2

terrycain, 2016-12-02:
Do you have kernel headers installed for the kernel that is currently running

z3ntu, 2016-12-02:
@doesnotcommute I would suggest you to uninstall the manual installation (manual is always bad) and then install the AUR package. If you have Telegram you can contact me @z3ntu , helping there is better ;)

About the kernel headers: If you have the package `linux` installed, please check if `linux-headers` is also installed.

doesnotcommute, 2016-12-02:
Installed the kernel headers. 
Output of lsmod | grep razer* : 
razerkbd               28672  0
hid                       114688  6 i2c_hid,hid_generic,usbhid,hid_rmi,hid_multitouch,razerkbd
usbcore               208896  6 uvcvideo,usbhid,razerkbd,xhci_pci,btusb,xhci_hcd

doesnotcommute, 2016-12-03:
Everything working perfectly. Thanks for the help @terrycain and @z3ntu

doesnotcommute, 2016-12-03:
PS- Let me know if you need me to test anything on the Blade Stealth 2016 with Arch (specifically Antergos OS)

terrycain, 2016-12-03:
Will do, if you join the telegram chat thats where most of the testing goes on as I just spam it asking for anyone to test.

closed 2016-12-03 by terrycain