openrazer/openrazer #89: Razer Diamondback Chroma Support

CorneliaXaos, 2016-11-30:
So.. I couldn't find any open issues for support for this device nor is it listed in the supported devices so I thought I'd open an issue and at least draw some attention to it.  (I'm sure you're busy with lots of other device support issues as well, but I wanted to see it in the list. ;) )

Anyways, I have a Diamondback Chroma and I'd be willing to help out with testing the driver and any other things you might need.  I, however, can't open Synapse to check what options are available there (as I don't have a Windows partition and don't want one :P ), but I do have the guidebook which lists which chroma features it supports (Specturm, Wave, Static, Breathing, and Reactive).

I tried to just see if the driver would recognize the device and just work... but strange things happened when I tried to install it so I'm not sure if the install even worked.  (There was a complaint about python-cryptography not working or something when the dkms ran, I think.)  Before I open an issue on that, though, I'd like to try a bit more troubleshooting and I'll create a separate issue when I've collected more data.

Anywho, to sum it all up I'd love to help out anyway I can with support for the Diamondback Chroma.

CorneliaXaos, 2016-11-30:
And.. I cleared up that strange python-cryptography error.... by setting update-alternatives to use python2 as default...  it would seem that when it compiled the kernel via dkms it was calling pyclean at some point and getting the python3 version and that clashed with python-cryptography and led to the failure there.  Even after getting it "installed" now I still don't see the Diamonback Chroma underneath /sys/bus/hid/drivers/razerkbd/

terrycain, 2016-12-01:
Probably because razerkbd is the keyboard driver, razermouse is the mouse driver :wink:, also diamondback isnt supported yet. Run `lsusb` and `ls -lah /dev/input/by-id` and paste output (in proper code blocks)

CorneliaXaos, 2016-12-01:
Odd...  I didn't see anything in the install for the driver about razermouse in the readme or the wiki..  Anyways...  Text output:

`lsusb`:
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 04f2:b5a7 Chicony Electronics Co., Ltd 
Bus 001 Device 004: ID 8087:0a2b Intel Corp. 
Bus 001 Device 003: ID 1c7a:0603 LighTuning Technology Inc. 
Bus 001 Device 002: ID 1532:004c Razer USA, Ltd 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

`ls -lah /dev/input/by-id`:
```
total 0
drwxr-xr-x 2 root root 140 Dec  1 14:02 .
drwxr-xr-x 4 root root 460 Dec  1 14:02 ..
lrwxrwxrwx 1 root root   9 Dec  1 14:02 usb-Razer_Razer_Diamondback-event-mouse -> ../event6
lrwxrwxrwx 1 root root   9 Dec  1 14:02 usb-Razer_Razer_Diamondback-if01-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 Dec  1 14:02 usb-Razer_Razer_Diamondback-if02-event-kbd -> ../event8
lrwxrwxrwx 1 root root   9 Dec  1 14:02 usb-Razer_Razer_Diamondback-mouse -> ../mouse0
lrwxrwxrwx 1 root root  10 Dec  1 14:02 usb-SunplusIT_Inc_Chicony_USB_2.0_Camera-event-if00 -> ../event15
```

I imagine the ones of interest are the ones with "razer" in them. :P

CorneliaXaos, 2016-12-01:
And... curious..  I have to modprobe razerkbd each time I boot for some reason...  (And I tested a modprobe of razer mouse and that brought that up so I guess it's at least here on the system.)

terrycain, 2016-12-01:
Well, if there was a capable device then udev would load it and assign the device to it.... but udev also seems a bit crap as its fighting with `hid-generic`.

Am just finishing up with some stuff with a DeathAdder chroma. Once I've finished that ill work on this.

CorneliaXaos, 2016-12-01:
Muchas gracias.  And if you need any more information from me I'd love to help..  I would have forked and tried to figure out how to add support myself.. but I'm not exactly proficient with code this low-level. :P  If anything, I can make sure it works with the daemon afterwards and stress check all the options as well as fix any bugs should they exist.  I'm quite good with Python.

terrycain, 2016-12-01:
Nice, few have tried to tackle the daemon (admittingly the daemon is evil), Am yet to see a big contribution to it :wink:

Tbf, your device "should" just be a matter of copy and paste :smile:

CorneliaXaos, 2016-12-01:
I figured it would be a simple copy paste and telling it to recognize the right device code...  and I did peek at the daemon.. it certainly looks evil.. :P

terrycain, 2016-12-01:
Driver shouldn't be too bad :smile:. Yeah I thought I was being awesome making the daemon flexible and super dynamic, which is it... it just got evil and complex on the way :smile:. The good thing about it though is that it can be swapped out with anything else that presents the same dbus interface.

CalcProgrammer1, 2016-12-21:
I just threw the Diamondback device ID into the driver and mimicked the Mamba TE for registering and unregistering files.  All the basic effects work fine.  The matrix from the TE doesn't crash the Diamondback but the positioning is wrong and it doesn't have enough LEDs.  The TE has 16 LEDs and the Diamondback has an additional 5 I think around the bottom, but the LEDs that are on both do not line up.  I'll try increasing the matrix I send to it and see if I can figure out the positioning.

terrycain, 2016-12-21:
Yeah I got the spec for the diamonback, the led layout is a bit different ha

terrycain, 2017-01-16:
In main codebase

closed 2017-01-16 by terrycain