openrazer/openrazer #90: Support for Razer Blade?

rossgoodwin, 2016-12-01:
I have a Razer Blade (not the Stealth), and I just successfully installed Ubuntu 16.10. Tried installing drivers and polychromatic, and the GUI opens but none of the controls do anything to the keyboard lights, which are still stuck on the setting they were in when I set them in Windows before installing Ubuntu.

Are you planning to add support for the non-Stealth 2016 Razer Blade? Thanks!

z3ntu, 2016-12-01:
Can you please run `lsusb | grep '1532:'`?

terrycain, 2016-12-01:
Also run `ls -lah /dev/input/by-id`

rossgoodwin, 2016-12-02:
```
$ lsusb | grep '1532:'
Bus 001 Device 005: ID 1532:020f Razer USA, Ltd 

$ ls -lah /dev/input/by-id
total 0
drwxr-xr-x 2 root root 160 Dec  1 22:04 .
drwxr-xr-x 4 root root 520 Dec  1 22:04 ..
lrwxrwxrwx 1 root root  10 Dec  1 22:04 usb-11121119-000J742ZD_USB_Camera_200901010001-event-if00 -> ../event11
lrwxrwxrwx 1 root root  10 Dec  1 22:04 usb-ELAN_Touchscreen-event-if00 -> ../event10
lrwxrwxrwx 1 root root   9 Dec  1 22:04 usb-Razer_Razer_Blade-event-kbd -> ../event7
lrwxrwxrwx 1 root root   9 Dec  1 22:04 usb-Razer_Razer_Blade-if01-event-kbd -> ../event8
lrwxrwxrwx 1 root root   9 Dec  1 22:04 usb-Razer_Razer_Blade-if02-event-mouse -> ../event9
lrwxrwxrwx 1 root root   9 Dec  1 22:04 usb-Razer_Razer_Blade-if02-mouse -> ../mouse0
```

terrycain, 2016-12-02:
https://github.com/terrycain/razer-drivers/issues/68

Issue open here

closed 2016-12-02 by terrycain