openrazer/openrazer #91: Unable to install from PPA in Ubuntu 16.10 due to missing dependencies.

gaganis, 2016-12-01:
Hello all!

I am trying to install the drivers on my fresh Ubuntu 16.10 installation. I have followed the instructions from here [https://terrycain.github.io/razer-drivers/#ubuntu](https://terrycain.github.io/razer-drivers/#ubuntu). But I get the following error about missing dependencies.

```The following packages have unmet dependencies:
 razer-daemon : Depends: python3-notify2 but it is not installable
                Depends: xautomation but it is not installable
                Depends: xdotool but it is not installable
```

I have tried googling to find how to install these packages but I was unable to find anything. I have also looked at the existing issues here on GitHub but again I was unable to find something that seemed to be similar or related to my problem.

Since it is mentioned on the site that the driver is supported for my version of Ubuntu I suspect that I missing something obvious I just cannot see right now. 

Could you please give me any pointers?

Thanks!
Giorgos

By The Way Great project! Kudos!

z3ntu, 2016-12-01:
I just tried installing it on a completely fresh Ubuntu 16.10 virtual machine and I had no problems installing the packages.

lah7, 2016-12-01:
Those packages come from the `universe` repository. Check that source is enabled in **Software & Updates**, then try refreshing the cache:

    sudo apt update

z3ntu, 2016-12-01:
Apparently main, universe, restricted & multiverse were all enabled by default in my 16.10 install... I even checked not to download the audio codecs pack...

gaganis, 2016-12-01:
Thank you @z3ntu @lah7  for your answers! The problem in my case was exactly the universe repository  not being enabled.

So the problem is solved for me! And the driver has worked quite nicely with my Razer Chroma KB.

Thanks!
Giorgos

closed 2016-12-01 by gaganis