openrazer/openrazer #92: Add note for the requirement of universe repository in Ubuntu installation instructions on the site.

gaganis, 2016-12-02:
I was experiencing the issue [Unable to install from PPA in Ubuntu 16.10 due to missing dependencies.](https://github.com/terrycain/razer-drivers/issues/91). I was quickly answered and it proved a pretty simple solution. Unfortunately it was not obvious to me before and despite searching quite a bit I wasn't able to resolve it without asking here on github.

I suggest that we added a note in the lines: "If you get dependency errors when trying to install the driver packages make sure that you have enabled the ```universe```  repository  in **Software & Updates**" at [https://terrycain.github.io/razer-drivers/#ubuntu](https://terrycain.github.io/razer-drivers/#ubuntu). 

It would have saved me some grief and hopefully others after me will not experience the same problem!

Thanks
Giorgos

terrycain, 2016-12-02:
@lah7 website is your area of expertise, mind added it to that :smile:, also throwing in the troubleshooting section wouldn't hurt either

gaganis, 2016-12-02:
Is the website sources available on github? 
If that is the case I could maybe help by doing a Pull Request with the change.

terrycain, 2016-12-02:
Look at https://github.com/terrycain/razer-drivers/pull/26 and it should be the branch `gh-pages` which contains the website.

gaganis, 2016-12-02:
Submitted Pull Request: https://github.com/terrycain/razer-drivers/pull/94 that adds a note to the site page.

terrycain, 2016-12-02:
Merged

closed 2016-12-02 by terrycain