openrazer/openrazer #93: Razer-Service Crash

Qormix, 2016-12-02:
I recently installed the drivers through the instruction on your page and received the following crash report

[image]
[image]
[image]

terrycain, 2016-12-02:
You installed it from the ppa?

terrycain, 2016-12-02:
Does the group plugdev exist? Is your user part of the plugdev group?

Qormix, 2016-12-02:
thanks for the lighting fast response. yes I did install it through PPA. and yes my user is part of the plugdev group

terrycain, 2016-12-02:
anything named razer in `/tmp` ?

Qormix, 2016-12-02:
tmp_7nnor_c4_razer_daemon

terrycain, 2016-12-02:
Can you run `ls -lah /sys/bus/hid/drivers/razerkbd/00*/`

Qormix, 2016-12-02:
I got the following output
```
qormix@Control-Stealth:~/Documents/Source$ ls -lah /sys/bus/hid/drivers/razerkbd/00*/
total 0
drwxr-xr-x 5 root root    0 Dec  2 11:00 .
drwxr-xr-x 5 root root    0 Dec  2 09:48 ..
-r--r--r-- 1 root root 4.0K Dec  2 11:00 country
-r--r----- 1 root root 4.0K Dec  2 11:00 device_type
lrwxrwxrwx 1 root root    0 Dec  2 09:48 driver -> ../../../../../../../bus/hid/drivers/razerkbd
-r--r----- 1 root root 4.0K Dec  2 11:00 get_firmware_version
-r--r----- 1 root root 4.0K Dec  2 09:48 get_serial
drwxr-xr-x 3 root root    0 Dec  2 09:48 hidraw
drwxr-xr-x 3 root root    0 Dec  2 09:48 input
-r--r--r-- 1 root root 4.0K Dec  2 11:00 modalias
--w--w---- 1 root root 4.0K Dec  2 11:00 mode_breath
--w--w---- 1 root root 4.0K Dec  2 11:00 mode_custom
-rw-rw---- 1 root root 4.0K Dec  2 11:00 mode_game
--w--w---- 1 root root 4.0K Dec  2 11:00 mode_none
--w--w---- 1 root root 4.0K Dec  2 11:00 mode_reactive
--w--w---- 1 root root 4.0K Dec  2 11:00 mode_spectrum
--w--w---- 1 root root 4.0K Dec  2 11:00 mode_static
--w--w---- 1 root root 4.0K Dec  2 11:00 mode_wave
drwxr-xr-x 2 root root    0 Dec  2 11:00 power
-r--r--r-- 1 root root 4.0K Dec  2 11:00 report_descriptor
--w--w---- 1 root root 4.0K Dec  2 11:00 reset
-rw-rw---- 1 root root 4.0K Dec  2 11:00 set_brightness
--w--w---- 1 root root 4.0K Dec  2 11:00 set_fn_toggle
--w--w---- 1 root root 4.0K Dec  2 11:00 set_key_row
--w--w---- 1 root root 4.0K Dec  2 11:00 set_logo
lrwxrwxrwx 1 root root    0 Dec  2 09:48 subsystem -> ../../../../../../../bus/hid
--w--w---- 1 root root 4.0K Dec  2 11:00 temp_clear_row
-rw-rw---- 1 root root 4.0K Dec  2 11:00 test
-rw-r--r-- 1 root root 4.0K Dec  2 09:48 uevent
-r--r----- 1 root root 4.0K Dec  2 11:00 version
```

terrycain, 2016-12-02:
Ok issue is that its not being `chown`'d if you change the group of that folder to plugdev recursively, then run the following 2 commands as your user:
```
pkill razer-service
razer-service --foreground --verbose --run-dir ~/.razer-service/data --log-dir ~/.razer-service/logs --pid-file ~/.razer-service/daemon.pid
```
Then it should start and just run in your terminal.

Dont suppose your running Ubuntu 16.10?

Qormix, 2016-12-02:
Yes I'm running 16.10

Qormix, 2016-12-02:
```
qormix@Control-Stealth:~$ pkill razer-service
qormix@Control-Stealth:~$ razer-service --foreground --verbose --run-dir ~/.razer-service/data --log-dir ~/.razer-service/logs --pid-file ~/.razer-service/daemon.pid
2016-12-02 11:10:48 | razer                          | INFO     | Initialising Daemon (v1.0.17). Pid: 2509
2016-12-02 11:10:48 | razer.screensaver              | INFO     | Initialising DBus Screensaver Thread
2016-12-02 11:10:48 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2016-12-02 11:10:48 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2013 --------------------- (1532:011A)
2016-12-02 11:10:48 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2016 --------------------- (1532:0214)
2016-12-02 11:10:48 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2016-12-02 11:10:48 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2016-12-02 11:10:48 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2016-12-02 11:10:48 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChromaTournamentEdition - (1532:021A)
2016-12-02 11:10:48 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2016-12-02 11:10:48 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2016-12-02 11:10:48 | razer                          | DEBUG    | Loaded device specification: RazerFireFly ---------------------------- (1532:0C00)
2016-12-02 11:10:48 | razer                          | DEBUG    | Loaded device specification: RazerImperiator ------------------------- (1532:002F)
2016-12-02 11:10:48 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaTE ---------------------- (1532:0046)
2016-12-02 11:10:48 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
2016-12-02 11:10:48 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
2016-12-02 11:10:48 | razer                          | DEBUG    | Loaded device specification: RazerOrochiWired ------------------------ (1532:0048)
2016-12-02 11:10:48 | razer                          | DEBUG    | Loaded device specification: RazerTartarus --------------------------- (1532:0208)
2016-12-02 11:10:48 | razer                          | INFO     | Found device.0: 0003:1532:0205.0003
2016-12-02 11:10:48 | razer.device0                  | INFO     | Initialising device.0 RazerBladeStealth
Traceback (most recent call last):
  File "/usr/sbin/razer-service", line 67, in <module>
    run()
  File "/usr/sbin/razer-service", line 61, in run
    daemonize(**daemon_args)
  File "/usr/lib/python3/dist-packages/razer_daemon/daemon.py", line 113, in daemonize
    daemon = RazerDaemon(verbose, log_dir, console_log, run_dir, config_file, test_dir=test_dir)
  File "/usr/lib/python3/dist-packages/razer_daemon/daemon.py", line 205, in __init__
    self._load_devices()
  File "/usr/lib/python3/dist-packages/razer_daemon/daemon.py", line 341, in _load_devices
    razer_device = device_class(device_path, device_number, self._config, testing=self._test_dir is not None)
  File "/usr/lib/python3/dist-packages/razer_daemon/hardware/keyboards.py", line 227, in __init__
    super(RazerBladeStealth, self).__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/razer_daemon/hardware/keyboards.py", line 19, in __init__
    super(_MacroKeyboard, self).__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/razer_daemon/hardware/device_base.py", line 47, in __init__
    self.serial = self.get_serial()
  File "/usr/lib/python3/dist-packages/razer_daemon/hardware/device_base.py", line 176, in get_serial
    with open(serial_path, 'r') as serial_file:
PermissionError: [Errno 13] Permission denied: '/sys/bus/hid/devices/0003:1532:0205.0003/get_serial'
2016-12-02 11:10:50 | razer.screensaver              | INFO     | Initialising DBus screensaver object

```

terrycain, 2016-12-02:
Then that will be Udev being a bellend.

Did you change the group of those driver files?

terrycain, 2016-12-02:
Also am pretty sure its related to this issue https://github.com/terrycain/razer-drivers/issues/67

Qormix, 2016-12-02:
Sorry I don't know why I assumed you added the chown command rerun it after changing group and I got 

```
root@Control-Stealth:/sys/bus/hid/drivers# pkill razer-service
root@Control-Stealth:/sys/bus/hid/drivers# razer-service --foreground --verbose --run-dir ~/.razer-service/data --log-dir ~/.razer-service/logs --pid-file ~/.razer-service/daemon.pid
2016-12-02 11:18:49 | razer                          | INFO     | Initialising Daemon (v1.0.17). Pid: 3459
2016-12-02 11:18:49 | razer.screensaver              | INFO     | Initialising DBus Screensaver Thread
2016-12-02 11:18:49 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2016-12-02 11:18:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2013 --------------------- (1532:011A)
2016-12-02 11:18:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2016 --------------------- (1532:0214)
2016-12-02 11:18:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2016-12-02 11:18:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2016-12-02 11:18:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2016-12-02 11:18:49 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChromaTournamentEdition - (1532:021A)
2016-12-02 11:18:49 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2016-12-02 11:18:49 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2016-12-02 11:18:49 | razer                          | DEBUG    | Loaded device specification: RazerFireFly ---------------------------- (1532:0C00)
2016-12-02 11:18:49 | razer                          | DEBUG    | Loaded device specification: RazerImperiator ------------------------- (1532:002F)
2016-12-02 11:18:49 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaTE ---------------------- (1532:0046)
2016-12-02 11:18:49 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
2016-12-02 11:18:49 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
2016-12-02 11:18:49 | razer                          | DEBUG    | Loaded device specification: RazerOrochiWired ------------------------ (1532:0048)
2016-12-02 11:18:49 | razer                          | DEBUG    | Loaded device specification: RazerTartarus --------------------------- (1532:0208)
2016-12-02 11:18:49 | razer                          | INFO     | Found device.0: 0003:1532:0205.0003
2016-12-02 11:18:49 | razer.device0                  | INFO     | Initialising device.0 RazerBladeStealth
2016-12-02 11:18:49 | razer.device0                  | DEBUG    | Adding getSerial method to DBus
2016-12-02 11:18:49 | razer.device0                  | DEBUG    | Adding razer.device.misc.suspendDevice method to DBus
2016-12-02 11:18:49 | razer.device0                  | DEBUG    | Adding razer.device.misc.resumeDevice method to DBus
2016-12-02 11:18:49 | razer.device0                  | DEBUG    | Adding razer.device.misc.getVidPid method to DBus
2016-12-02 11:18:49 | razer.device0                  | DEBUG    | Adding razer.device.misc.version method to DBus
2016-12-02 11:18:49 | razer.device0                  | DEBUG    | Adding razer.device.misc.hasDedicatedMacroKeys method to DBus
2016-12-02 11:18:49 | razer.device0                  | DEBUG    | Adding razer.device.misc.getFirmware method to DBus
2016-12-02 11:18:49 | razer.device0                  | DEBUG    | Adding razer.device.misc.getMatrixDimensions method to DBus
2016-12-02 11:18:49 | razer.device0                  | DEBUG    | Adding razer.device.misc.hasMatrix method to DBus
2016-12-02 11:18:49 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceName method to DBus
2016-12-02 11:18:49 | razer.device0                  | DEBUG    | Adding razer.device.misc.getDeviceType method to DBus
2016-12-02 11:18:49 | razer.device0                  | DEBUG    | Adding razer.device.lighting.brightness.getBrightness method to DBus
2016-12-02 11:18:49 | razer.device0                  | DEBUG    | Adding razer.device.lighting.brightness.setBrightness method to DBus
2016-12-02 11:18:49 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setWave method to DBus
2016-12-02 11:18:49 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setStatic method to DBus
2016-12-02 11:18:49 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setSpectrum method to DBus
2016-12-02 11:18:49 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setReactive method to DBus
2016-12-02 11:18:49 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setNone method to DBus
2016-12-02 11:18:49 | razer.device0                  | DEBUG    | Adding razer.device.lighting.chroma.setBreathRandom method to DBus
[... cut ...]
2016-12-02 11:18:51 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-12-02 11:18:51 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-12-02 11:18:51 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-12-02 11:18:51 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2016-12-02 11:18:51 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2016-12-02 11:18:51 | razer.screensaver              | WARNING  | Giving up on screensaver

```

terrycain, 2016-12-02:
Yep same as #67, I know its an issue, am busy with releasing support for another device, then I think I decided I was going to just write a simple service that solely deals with binding the device and setting permissions.

Qormix, 2016-12-02:
I'm closing this issue because its directly related to #67. Thanks for the help.

closed 2016-12-02 by Qormix