openrazer/openrazer #96: Ouroboros Support

WhatsCS, 2016-12-02:
Here is a zip of all settings for the mouse, if I missed anything please tell me. 
[2016-12.zip](https://github.com/terrycain/razer-drivers/files/628413/2016-12.zip)

terrycain, 2016-12-02:
Ok we will deal with Wireless mode first, so until otherwise keep the mouse unplugged :wink:.
I need packet captures (each bulletpoint a seperate capture). Thoughout use a seperate mouse as not to pollute the capture with rubbish. Also make sure you set up the correct filter as documented [here](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol#phase-2-setup)

- Start wireshark, then start vm, and open up razer synapse. then stop recording.
- Brightness slider, move to max, half, min, half, max, half, min.
- DPI Min, move it in 1000 increments to max
- Sleep mode idle after, move it from min to max through each increment
- Flash battery indicxator falls below, move from min to max through each increment
- Open synapse and just leave running for 5 mins.

WhatsCS, 2016-12-02:
This should be everything on that list for WIRED, as stated in Telegram I'll let it charge and see where that leads us.
[USB-Inspection-Oro.zip](https://github.com/terrycain/razer-drivers/files/628527/USB-Inspection-Oro.zip)

terrycain, 2016-12-31:
> Start wireshark, then start vm, and open up razer synapse. then stop recording.

did you get this capture?

terrycain, 2016-12-31:
Also what distro you on

terrycain, 2016-12-31:
Also whats the output of `lsusb`

WhatsCS, 2017-01-01:
1. Pretty sure that's idle.zip
2. Arch Linux
3. not sure I'll have to update when I'm on it.

terrycain, 2017-01-01:
Ok I'll double check idle.zip when I get home

WhatsCS, 2017-01-01:
```
Bus 006 Device 004: ID 1532:0032 Razer USA, Ltd 
Bus 006 Device 003: ID 1532:010e Razer USA, Ltd
```

terrycain, 2017-01-01:
What one is ouroboros?

z3ntu, 2017-01-01:
`1532:010e` is either Razer Black Widow or Lachesis

terrycain, 2017-01-01:
Ahh good point it is a BW

terrycain, 2017-01-01:
Idle.zip doesnt contain from startup, (or if it does it doesnt have what im looking for)

WhatsCS, 2017-01-09:
Should include startup.
[idle.zip](https://github.com/terrycain/razer-drivers/files/694462/idle.zip)

terrycain, 2017-01-10:
YES :) that had exactly what I was looking for. Good job

terrycain, 2017-01-10:
Ok @z3ntu as its arch, can you help him here or via telegram install the driver from branch [feature_ouroboros_0x0032](https://github.com/terrycain/razer-drivers/tree/feature_ouroboros_0x0032)

z3ntu, 2017-01-10:
You probably will get some dependency issues because some packages needed are only available in the AUR -> install them first (AUR helper or whatever you like)
**Run the following commands:**
```
git clone https://aur.archlinux.org/razer-drivers-git.git
cd razer-drivers-git
curl https://gist.githubusercontent.com/z3ntu/28cb958ce9a778de0975e54c54b77318/raw/9e21f1473b04afb47951f4000c0e01d0f7442779/PKGBUILD.ouroboros -o PKGBUILD
makepkg -sr
sudo pacman -U python-razer-git-*.tar* razer-daemon-git-*.tar* razer-driver-dkms-git-*.tar*
```

terrycain, 2017-01-16:
PPA updated.

closed 2017-01-16 by terrycain