openrazer/openrazer #97: Razer Blade 2012 support

RabidBurn, 2016-12-03:
Hi fellas,
Unless I've messed something up in the install procedure, I don't think this works on the 2012 Razer Blade.

I'm happy to provide whatever log files you'd like, but I'm not sure what you need.

I'm running Debian 8 (Jessie), Kernel 3.16.0-4, amd64

Here's the procedure I'm following:
- Install Prereqs
- Clone source,
- sudo make -s all install (output: http://pastebin.com/uE48AN8M)
At this stade, all looks good. No errors, at least so far as I can tell.

Next, Razer Commander:
- Clone the repo
- chmod 755 main.py
- ./main.py
- This gives an error (http://pastebin.com/LDPNnSad) saying that it can't connect to the daemon, which leads me to believe it that it has to do with the razer-drivers installation.

As for PolyChromatic:
- Clone the repo
- chmod 755 install.py
- install seems to go fine, but then when I try to run it, (http://pastebin.com/DudNu7is) happens, and it stays frozen. Not really sure where to go from here.

Any suggestions would be appreciated. Also, as I said earlier, I'm happy to provide anything else you guys need.

Thanks!
RabidBurn

terrycain, 2016-12-03:
First run `lsusb`

lah7, 2016-12-03:
> As for PolyChromatic:
>
> install seems to go fine, but then when I try to run it, (http://pastebin.com/DudNu7is) happens, and it stays frozen. Not really sure where to go from here.

Turns out the version of WebKit2 provided in the `jessie` repository is "old" (2.6.2), I didn't realise this. To fix Polychromatic, switch to the `jessie-backports` repository and upgrade the package `gir1.2-webkit2-4.0` to 2.14.

I've updated the README and program to catch this dependency version problem I didn't realise.

Even so, once that program works, it should pop up with an error about the daemon not running.

RabidBurn, 2016-12-03:
Hi, thanks for the reply!

Here's the output of lsusb (http://pastebin.com/cv8mcq3B), though I'm assuming the part you're interested in is: `Bus 003 Device 003: ID 1532:0116 Razer USA, Ltd `

As for Polychromatic, updating that package did made it run! Hitting the "restart daemon" button seems to have worked, but the brightness slider doesn't actually affect keyboard brightness (which I assume is because my device isn't supported). No worries there, don't really care about it.

As for RazerCommander, now it's saying that it requires GTK3+ v3.18, so I'm going to try and figure out how the hell to upgrade that... Will update once I figure it out (or don't).

Thanks!

terrycain, 2016-12-03:
Ugh yet another razer blade variant. This one isn't supported yet

terrycain, 2016-12-03:
Dont suppose you have access to razer synapse? Could do with some screenshots from it

RabidBurn, 2016-12-03:
Don't currently have Windows installed, but I've got a spare hard drive I could do a temp install on... What do you need me to get from Synapse?

z3ntu, 2016-12-03:
Screenshots of all options available there.

RabidBurn, 2016-12-03:
Oh boy. That should be fun... Should I even bother updating GTK?

z3ntu, 2016-12-03:
razercommander can't support your device if the driver doesn't support it. So you can wait ;)

terrycain, 2016-12-03:
Ideally if you could install windows in a vm in virtualbox that would be idea

RabidBurn, 2016-12-03:
Oh, that'll work? Yeah I can do that. 

> razercommander can't support your device if the driver doesn't support it. So you can wait ;)
I don't mind waiting lmao, no rush at all

RabidBurn, 2016-12-03:
Oh, that'll work? I can do that

>razercommander can't support your device if the driver doesn't support it. So you can wait ;)

I don't mind waiting lol, no rush at all

terrycain, 2016-12-03:
Yeah getting driver support would be the best thing to do first. Yeah if you can get windows installed in a VM you can pass through the laptops keyboard to the vm (though it would help to have a second keyboard plugged in to use instead whilst the vm is running. [This](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol) is whats normally done, but seeing as we have support for other blades its probably just a copy-paste job.

RabidBurn, 2016-12-03:
Ok, I'll get started on that.

RabidBurn, 2016-12-03:
Just FYI, this model doesn't have a chroma keyboard. The thing Synapse controls is status (on/off), brightness, and remapping keys. It also controls the switchblade UI, but I'm assuming there's no way to get that working...

terrycain, 2016-12-03:
get me screenshots of synapse and ill have a think

RabidBurn, 2016-12-03:
So any plans I had of doing that today have been cancelled... Somehow managed to totally mess up the graphics drivers, decided to just reinstall Debian from scratch.

RabidBurn, 2016-12-16:
Ok so I keep managing to fuck up my system. There aren't any concrete instructions to get the nVidia drivers set up, and I keep forgetting to write down what the hell I did, so it takes me two to three hours every time.

Everything was working fine last night, I open it up again today, now Gnome doesn't want to run. Which means I'm going to have to reinstall again (since there doesn't seem to be a solution online), which means I'm going to have to waste time figuring out nVidia again...

Have any of you guys managed to find a simple way to get the nVidia drivers to work? I've been doing it using Bumblebee, and I have to stay on kernel 3.16.0-4 since nVidia still haven't released the headers for 4.7.0.

This is rapidly becoming more of a time suck than it's worth for me.

RabidBurn, 2016-12-16:
Alright so I may have panicked a little.
I managed to fix Gnome, I'm going to install a Windows VM now and see if I can get you those screenshots

terrycain, 2016-12-16:
Ha. This is why I avoid nVidia like the plague...

RabidBurn, 2016-12-16:
Ok so I set up a Windows 7 VM in KVM and I redirected only Razer USB device to it, but synapse isn't detecting it... Anything I should try to get it working?

terrycain, 2016-12-16:
What distro you running?

RabidBurn, 2016-12-16:
Debian Jessie, kernel 3.16.0-4

terrycain, 2016-12-16:
Look into selinux or apparmor to see if it's denying anything. Syslog should tell you

RabidBurn, 2016-12-16:
I'll look into this later today

RabidBurn, 2016-12-18:
Ok so I looked at syslog. It loads the Razer device at startup, and then unloads it when I start the Windows 7 VM. Is there anything else I should be looking for?

RabidBurn, 2016-12-18:
Also, in the Device Manager on Windows 7, there's an unknown USB Composite Device, which I assume is the redirected USB device. Is there a driver I can load, or is the driver buried in Synapse somewhere?

RabidBurn, 2016-12-19:
I've got a present for you
[snaps.zip](https://github.com/terrycain/razer-drivers/files/659804/snaps.zip)

terrycain, 2016-12-19:
Ooh nice :smile:. What was wrong with the usb stuff in the end? Will have a proper look when I get home

RabidBurn, 2016-12-20:
Not sure exactly why this problem was happening, but what fixed it was choosing "Redirect USB Device" from the KVM Connection window rather than actually going into the VM Properties and creating a new USB Redirector... Seems strange to me, but maybe I'm missing something.

terrycain, 2016-12-20:
Odd, no clue ;)

terrycain, 2016-12-20:
Ok no clue when it comes to the trackpad. Can you explain more about it, what it does, how it works. And it looks like some buttons above it are part of that and not the kb.

If you follow [this](https://github.com/terrycain/razer-drivers/wiki/Reverse-Engineering-USB-Protocol) link to the wiki I posted before, its packet capture time (preferably plug in a keyboard so that you don't taint the capture with keypresses).

If I'm not mistaken this is one of the first blades?

1 Packet capture of game mode toggling on and off a few times (obviously note down the order)
1 Packet capture of brightness going from min->max and back a few times
1 Packet capture of brightness going from min->max and back a few times for switchblade
1 Packet capture of toggling lighting off when idle
1 Packet capture of lighting off when idle going from min to max stopping at a few options in the middle and back again a few times.
1 Packet capture of toggling switch to trackpad power saving a few times

RabidBurn, 2016-12-21:
Most of this is going to have to wait for a while, I'm a little swamped right now.

As for the trackpad, I can give you a quick run down. It's called the Switchblade UI. The keys above it are programmable macro keys with displays for each key. The trackpad itself is also a display, again, programmable through the use of apps that you can download from Synapse.

This is the second Razer Blade they ever made, back in 2012.

terrycain, 2016-12-21:
Hmm I see.
> The keys above it are programmable macro keys with displays for each key

You mean there's an lcd display per key?

Hmm, once we get the rest of the blade done, we can have a crack at dealing with the switchblade

> Most of this is going to have to wait for a while

Np, I ain't in any rush, I have tons to do too :wink:

taa1, 2017-09-21:
In case it would be helpful, I have access to a Razer Blade 2012 also, plus the Death Stalker Ultimate -- both have the Switchblade. I'd love to have access to the Switchblade on Linux, but like RapidBurn, I'm not in a rush, either.

terrycain, 2017-09-21:
@taa1 I too would like the switchblade working. Have moved house and have no internet. When it's fixed I'll have a look