openrazer/openrazer #98: Razer Blade Stealth (Late 2016) - Device not found

soriat, 2016-12-03:
I'm running Ubuntu Gnome - `Ubuntu 16.10 64-bit`

I feel like the driver  installation went smoothly in every regard:
```
razer-kernel-modules-dkms is already the newest version (1.0.17-0ubuntu1).
razer-doc is already the newest version (1.0.17-0ubuntu1).
razer-daemon is already the newest version (1.0.17-0ubuntu1).
python3-razer is already the newest version (1.0.17-0ubuntu1).
```

I've rebooted already and installed `Polychromatic` and `razerCommander`.

My problem is that both these apps are empty / don't recognize my keyboard. Even the brightness slider in Polychromatic is not working for me. Since both are broken my best guess is that this might be a driver issue.

```
lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 04f3:2356 Elan Microelectronics Corp. 
Bus 001 Device 004: ID 1532:0220 Razer USA, Ltd 
Bus 001 Device 003: ID 0bda:579f Realtek Semiconductor Corp. 
Bus 001 Device 002: ID 0cf3:e300 Atheros Communications, Inc. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

`1532:0220` should be supported right?

Any help or advice would be greatly appreciated.

(Probably not relevant at all, but my keyboard is stuck in the "Starlight" pattern with the function keys lighting up when pressing `fn`. I'd be super happy if they were all just a static color at this point)

terrycain, 2016-12-03:
Hehe its probably a udev issue.

Try
```
pkill razer-service
# Wait 5 seconds
sudo udevadm trigger
# Wait 5 seconds
razer-service --verbose --run-dir ~/.razer-service/data --log-dir ~/.razer-service/logs --pid-file ~/.razer-service/daemon.pid
```
Then try launching one of the apps. The daemon should just sit in the terminal so launch app from another one

terrycain, 2016-12-03:
If that works its probably https://github.com/terrycain/razer-drivers/issues/67

soriat, 2016-12-03:
I ran those commands but no dice.

Both apps have misc errors and there's also this:
```
sudo modprobe razerkbd
modprobe: ERROR: could not insert 'razerkbd': Unknown symbol in module, or unknown parameter (see dmesg)
```
Dmesg contents:
[dmsegLog.txt](https://github.com/terrycain/razer-drivers/files/629037/dmsegLog.txt)

Logfile from your previous comment:
[razerLog.txt](https://github.com/terrycain/razer-drivers/files/629038/razerLog.txt)

Thank you so much for the insanely fast responses btw!

terrycain, 2016-12-03:
How did you install the driver?

soriat, 2016-12-05:
Just the normal Ubuntu way:
```
sudo add-apt-repository ppa:terrz/razerutils
sudo apt update
sudo apt install python3-razer razer-kernel-modules-dkms razer-daemon razer-doc
sudo modprobe razerkbd
```

terrycain, 2016-12-05:
Hmm, as the dmesg suggests the driver didnt get built properly by dkms.

Just to be safe you mind purging all the packages, reinstalling them then rebooting as posting the dmesg again
```
apt-get remove --purge razer-kernel-modules-dkms razer-daemon python3-razer
apt-get install -y razer-kernel-modules-dkms razer-daemon python3-razer

reboot

# post dmesg
```

z3ntu, 2016-12-31:
@soriat Does the issue still exist?

Kequc, 2017-01-08:
I have a razer blade stealth late 2016 and I'm having the same issue. I've tried all the steps listed in this thread and still the device isn't detected.

If I run `lsusb | grep '1532:'` I correctly see:

`Bus 001 Device 004: ID 1532:0220 Razer USA, Ltd`

But in razercommander I see nothing. I've installed, uninstalled, and re-installed the drivers and razercommander in alternating order. I suspect it might have something to do with secure boot. Which forces me to set a password to temporarily disable secure boot, warning me that otherwise 3rd party drivers won't work.

I'm running Ubuntu 16.10. Actually I'd totally remove secure boot, whatever it is if I could. It seems to cause an annoying prompt every time I start my computer that forces me to wait 10 seconds or press any key. I think it was added when I half heartedly selected 'encrypt my home directory' during installation of the os.

That's out of scope of this issue. But maybe it's relevant?

Kequc, 2017-01-08:
Specifically the quirk I'm seeing is called "UEFI Secure Boot" which is apparently firmware, which must have come with the computer.

Kequc, 2017-01-08:
I entered my bios and disabled this security feature and can confirm that things work now. So, that was indeed the issue.

z3ntu, 2017-01-08:
Ubuntu doesn't like dkms modules with Secure Boot enabled unfortunately...

terrycain, 2017-01-08:
@z3but @Kequc I think we should make a wiki page detailing shit about secure boot. There is a way to trust a key in the uefi stuff and use that to sign dkms and leave secure boot on.

terrycain, 2017-01-08:
But turning it off is far easier ;-)

z3ntu, 2017-01-08:
I should try getting Secure Boot to work with Arch, there's some way and then try the dkms stuff ^^

`z3but` :D

terrycain, 2017-01-08:
> z3but

Yep that's autocorrect for you

osum4est, 2017-01-08:
I also have a late 2016 RBS, with ID 1532:0220 Razer USA, Ltd. with the same issue. I have secure boot turned off, and Polychromatic does not find any devices. I'm running Arch Linux, with the latest version on the AUR. Any fixes yet?

z3ntu, 2017-01-09:
@osum4est do you have Telegram so I can help you easier?

osum4est, 2017-01-09:
@z3ntu i do now! just messeged you

osum4est, 2017-01-09:
my issue was i didn't have linux-headers installed so the dkms package couldn't be compiled. once installing the headers it worked!

nickdesaulniers, 2017-01-22:
I have the exact same setup and cannot repro.  This may be a secure boot issue, which is weird, but w/e.

When installing the drivers as instructed, Ubuntu will ask you to disable secure boot via a ncurses menu.  Set a password as it suggests, then reboot.  You'll get a prompt from the uefi about random characters of the password, 1-indexed.  Enter them, then select disable secure boot.

example password: `password`
possible prompt: `enter 5th character of password`
enter: `w`

terrycain, 2017-01-23:
@nickdesaulniers whats up?

nickdesaulniers, 2017-01-24:
OP reported a Late 2016 RBS with Ubuntu 16.10.  I have the same setup, but can't repro.  People were talking about secure boot, so I posted my steps.

terrycain, 2017-01-24:
Ahh ok, nice, yeah I will probably add a mention about secure boot on the readme but until i get something with secureboot on it and it affects me, am not going to devote much effort into it.

Kequc, 2017-01-24:
> When installing the drivers as instructed, Ubuntu will ask you to disable secure boot via a ncurses menu. Set a password as it suggests, then reboot. You'll get a prompt from the uefi about random characters of the password, 1-indexed. Enter them, then select disable secure boot.

Oh I entered a password as instructed and it never asked me for anything again. I was wondering what it was talking about that I needed to choose a password because I never needed it. Was trying to follow instructions as best I could.

Disabling secure boot finally solved the issue for me. But I suspect if the uefi had asked me for random characters of the password that may have worked too.

terrycain, 2017-01-24:
ok ill add secureboot stuff to the troubleshooting page. :)

closed 2017-01-24 by terrycain