openrazer/openrazer #103: Razer Blade Stealth (Late 2016, UK keyboard) fn key weirdness + caps lock crashes laptop

jamtat, 2016-12-06:
Hello there! I've had a couple of fascinating issues with this brand new laptop in the last couple of days. Initially I found that pressing the caps lock key twice crashes the laptop causing the screen to glitch and tear and can only be fixed by a hard reset. I updated to the latest version of your drivers today which fixed that issue (marvellous!) however now when I press the left fn key it types "g" and the right fn key types "h". Running `xev` confirms that those keycodes are indeed being sent. fn+F4 also sends a "p" key. Uninstalling the drivers fixes typing the characters with fn keys but brings back the caps lock crashing issue. I'm a developer myself but have never strayed into driver development, but if there's any information I can send you to help or anything I can contribute please let me know!

Thanks!

terrycain, 2016-12-06:
Yeah another user is finding the G issue. Am working on it

stealthowner, 2016-12-19:
@jamtat have you made any additional tweaks/ installed additional drivers in order to make caps lock issue fix? even after i installed the latest version of these drivers, caps issue still appearing.
thanks

terrycain, 2016-12-19:
By installing the latest version is directly from github or via package?

stealthowner, 2016-12-19:
@terrycain via package.

terrycain, 2016-12-19:
Ahh yeah pretty sure I have a fix for that unreleased

stealthowner, 2016-12-19:
@terrycain are you planning on releasing one if yes then when?

terrycain, 2016-12-19:
Soon :wink: Am working on releasing another razer blade so should come out with that

terrycain, 2016-12-19:
I got some merge conflict fun first

stealthowner, 2016-12-19:
will be waiting on that, thanks!

jamtat, 2016-12-25:
@stealthowner I use this `setxkbmap -option caps:none` now to just disable it. I purged the system of drivers and for the moment just use it sans drivers. looking forward to your fix @terrycain

terrycain, 2016-12-25:
Install latest package, see what that's like.

stealthowner, 2016-12-25:
@terrycain latest package didn't help, do you need anything to help you fix it?

terrycain, 2016-12-25:
Have to ask, did you install and reboot? And anything useful in `~/.razer-service/logs`?

kickthedragon, 2017-01-02:
@terrycain I'm running fedora 25 and I've installed the latest package via the repo you setup. Below is my log of what took place up to the point of hitting CapsLock

`
2017-01-02 14:17:31 | razer                          | INFO     | Initialising Daemon (v1.1.5). Pid: 1902
2017-01-02 14:17:31 | razer.screensaver              | INFO     | Initialising DBus Screensaver Thread
2017-01-02 14:17:32 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
2017-01-02 14:17:32 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
2017-01-02 14:17:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2013 --------------------- (1532:011A)
2017-01-02 14:17:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2016 --------------------- (1532:0214)
2017-01-02 14:17:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
2017-01-02 14:17:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
2017-01-02 14:17:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
2017-01-02 14:17:32 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChromaTournamentEdition - (1532:021A)
2017-01-02 14:17:32 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
2017-01-02 14:17:32 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
2017-01-02 14:17:32 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
2017-01-02 14:17:32 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
2017-01-02 14:17:32 | razer                          | DEBUG    | Loaded device specification: RazerChromaMugHolder -------------------- (1532:0F07)
2017-01-02 14:17:32 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderChroma ------------------- (1532:0043)
2017-01-02 14:17:32 | razer                          | DEBUG    | Loaded device specification: RazerDeathadderElite -------------------- (1532:005C)
2017-01-02 14:17:32 | razer                          | DEBUG    | Loaded device specification: RazerFireFly ---------------------------- (1532:0C00)
2017-01-02 14:17:32 | razer                          | DEBUG    | Loaded device specification: RazerImperiator ------------------------- (1532:002F)
2017-01-02 14:17:32 | razer                          | DEBUG    | Loaded device specification: RazerKraken ----------------------------- (1532:0504)
2017-01-02 14:17:32 | razer                          | DEBUG    | Loaded device specification: RazerKrakenV2 --------------------------- (1532:0510)
2017-01-02 14:17:32 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaTE ---------------------- (1532:0046)
2017-01-02 14:17:32 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWired ------------------- (1532:0044)
2017-01-02 14:17:32 | razer                          | DEBUG    | Loaded device specification: RazerMambaChromaWireless ---------------- (1532:0045)
2017-01-02 14:17:32 | razer                          | DEBUG    | Loaded device specification: RazerNagaHexV2 -------------------------- (1532:0050)
2017-01-02 14:17:32 | razer                          | DEBUG    | Loaded device specification: RazerOrnataChroma ----------------------- (1532:021E)
2017-01-02 14:17:32 | razer                          | DEBUG    | Loaded device specification: RazerOrochiWired ------------------------ (1532:0048)
2017-01-02 14:17:32 | razer                          | DEBUG    | Loaded device specification: RazerTartarus --------------------------- (1532:0208)
2017-01-02 14:17:32 | razer                          | INFO     | Adding razer.devices.getDevices method to DBus
2017-01-02 14:17:32 | razer                          | INFO     | Adding razer.devices.enableTurnOffOnScreensaver method to DBus
2017-01-02 14:17:32 | razer                          | INFO     | Adding razer.devices.disableTurnOffOnScreensaver method to DBus
2017-01-02 14:17:32 | razer                          | INFO     | Adding razer.devices.syncEffects method to DBus
2017-01-02 14:17:32 | razer                          | INFO     | Adding razer.daemon.version method to DBus
2017-01-02 14:17:32 | razer                          | INFO     | Adding razer.daemon.stop method to DBus
2017-01-02 14:17:32 | razer                          | INFO     | Serving DBus
2017-01-02 14:17:33 | razer.screensaver              | INFO     | Initialising DBus screensaver object
2017-01-02 14:17:33 | razer.screensaver              | WARNING  | Could not find screensaver DBus
2017-01-02 14:17:33 | razer.screensaver              | WARNING  | Giving up on screensaver
2017-01-02 14:17:40 | razer                          | INFO     | Stopping daemon on signal 15
[... cut ...] \00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00\00
`

terrycain, 2017-01-02:
@kickthedragon you able to join the telegram group to debug this?

kickthedragon, 2017-01-02:
Just sent you a message, you can go ahead and add me to the group.

terrycain, 2017-01-02:
@jamtat @stealthowner With or without the razer stuff theres still the caps lock issues right?

jamtat, 2017-01-02:
I did not have it when I installed the latest drivers at the time of my
original post. I havent checked with latest drivers since

terrycain, 2017-01-02:
@jamtat what distro you running?

jamtat, 2017-01-02:
Ubuntu 16.04 but with manually updated kernel. I couldn't tell you what
version it was at the time

terrycain, 2017-01-02:
@jamtat Can you test with latest drivers when you get a chance.

jamtat, 2017-01-02:
Sure. I will be without my laptop for long perioids but I'll install when I
get chance

terrycain, 2017-01-02:
Wicked thanks.

kickthedragon, 2017-01-03:
I can confirm CapsLock crashing the entire system happens regardless of whether these drivers are installed or not on both Ubuntu 16.04 LTS and Fedora 25

stealthowner, 2017-01-10:
@kickthedragon i can confirm the same issues on ubuntu 16.04 lts

terrycain, 2017-01-10:
Guys does this stop it `setxkbmap -option caps:none` (obviously disabling caps lock entirely)

stealthowner, 2017-01-10:
@terrycain yes it does but it is not a long term solution

terrycain, 2017-01-10:
No its not, @stealthowner /anyone, you reckon you could enable coredumps and get me a core dump of it crashing, I'd love to help out but I dont really have time to walk people through core-dumping

nickdesaulniers, 2017-01-22:
I don't think this is an issue with these drivers, but with Intel's i915 drm subsystem; I can reproduce without these drivers even installed.  I can work around by adding the kernel command line `i915.enable_rc6=0` which also solves the rapid display flicker.  See [this thread](https://www.reddit.com/r/razer/comments/5orvy5/late_2016_rbs_qhd_linux_damage_report/) for more info.

stealthowner, 2017-01-23:
@nickdesaulniers any chance for detailed explanation on how you solved the caps issue?

terrycain, 2017-01-23:
Its been confirmed its an intel bug. Feel free to chat about it here but am closing the issue to avoid any confusion.

If you guys come up with some good instructions, raise an issue to add it to the wiki, add it yourself if it allows you.

closed 2017-01-23 by terrycain