openrazer/openrazer #105: Fedora 25 installation

Fenwick17, 2016-12-07:
Having some issues with the Fedora installation setup on my PC, as im hoping to stop my blackwidow doing the full rainbow.
When i run make I get http://pastebin.com/6Mg34kXC

terrycain, 2016-12-07:
run `make driver_verbose`

z3ntu, 2016-12-07:
I just tried a fresh Fedora 25 virtual machine and got the same problem:
```
[luca@localhost razer-drivers]$ make driver_verbose

:: Compiling Razer kernel modules
========================================
make -C /lib/modules/4.8.11-300.fc25.x86_64/build SUBDIRS=/home/luca/razer-drivers/driver modules
make[1]: *** /lib/modules/4.8.11-300.fc25.x86_64/build: No such file or directory.  Stop.
Makefile:28: recipe for target 'driver_verbose' failed
make: *** [driver_verbose] Error 2
[luca@localhost razer-drivers]$ ls -al /lib/modules/4.8.11-300.fc25.x86_64/build
lrwxrwxrwx. 1 root root 39 Nov 28 19:47 /lib/modules/4.8.11-300.fc25.x86_64/build -> /usr/src/kernels/4.8.11-300.fc25.x86_64
[luca@localhost razer-drivers]$ ls -al /usr/src/kernels/4.8.11-300.fc25.x86_64
ls: cannot access '/usr/src/kernels/4.8.11-300.fc25.x86_64': No such file or directory
```
But I fixed it by running `sudo dnf install kernel-devel`, then the directory `/usr/src/kernels/4.8.11-300.fc25.x86_64` is populated with files and the compilation works fine.

(I just don't know if I should be using `sudo make install` afterwards 😉 )

terrycain, 2016-12-07:
can you get to the point of modprobe razerkbd?

z3ntu, 2016-12-07:
Duplicate of #86

closed 2016-12-23 by terrycain