openrazer/openrazer #106: Razer Naga Hex v2 Support

artemjackson, 2016-12-07:
First of all thanks for the great work.
Can you please add the Naga Hex v2 support?
I'm ready to give you any info you may need.

terrycain, 2016-12-07:
Yeah I also have one. January time.

terrycain, 2016-12-17:
Started a bit of work on this ;)

terrycain, 2016-12-27:
Done, Haven't made a package yet but its working. Will probs release in Jan

closed 2016-12-31 by terrycain

artemjackson, 2017-01-03:
I've just installed v1.1.5 and it seems that daemon can't recognize the mouse :cry:

Particularly I installed `razer-daemon`, `razer-driver-dkms`, `phyton-razer` all version 1.1.5 from AUR (I'm Arch Linux user).
Tested with `polychromatic` 0.3.5 and `razercommander` 1.0.3

Let me know if I can help you somehow with logs or something like that.

terrycain, 2017-01-03:
~/.razer-service/logs

artemjackson, 2017-01-04:
https://gist.github.com/artemjackson/6863a3eecaac6b7ad0a6da3fa9199c77

Line 39 - Opened Polychromatic
Line 40 - Clicked "Retry"
Line 41 - Clicked "Retry"
Line 42 - Clicked "Restart daemon"

terrycain, 2017-01-04:
Device isnt getting bound by udev. `ls /sys/bus/hid/driver/razermouse` `cat /var/log/syslog | grep -i bind -A3 -B3`

z3ntu, 2017-01-04:
Syslg doesn't exist on Arch ;)

artemjackson, 2017-01-04:
@terrycain 
Well, first of all there is no `/sys/bus/hid/driver`, but `/sys/bus/hid/drivers` is without any razer related files
Secondly, as @z3ntu has already mentioned there is no `syslog` under Arch Linux

terrycain, 2017-01-04:
thats probs what I meant, ok try `modprobe razerkbd` and then `lsmod | grep razer`

artemjackson, 2017-01-04:
For `systemctl` I pulled some logs, but probably not useful
```
...
Jan 04 22:09:46 zenbook razerd[431]: Razer device service daemon
...
Jan 04 22:09:47 zenbook razer_mount[524]: Driver razermouse
Jan 04 22:09:47 zenbook razer_mount[525]: Driver razermouse
Jan 04 22:09:47 zenbook razer_mount[523]: Driver razermouse
Jan 04 22:09:47 zenbook razer_mount[526]: Device_ID 0003:1532:0050.0003
Jan 04 22:09:47 zenbook razer_mount[528]: Device_ID 0003:1532:0050.0002
Jan 04 22:09:47 zenbook razer_mount[527]: Device_ID 0003:1532:0050.0001
Jan 04 22:09:47 zenbook razer_mount[529]: Modprobing razermouse
Jan 04 22:09:47 zenbook razer_mount[530]: Modprobing razermouse
Jan 04 22:09:47 zenbook razer_mount[531]: Modprobing razermouse
...
Jan 04 22:09:47 zenbook kernel: hidraw: raw HID events driver (C) Jiri Kosina
Jan 04 22:09:47 zenbook kernel: usbcore: registered new interface driver usbhid
Jan 04 22:09:47 zenbook kernel: usbhid: USB HID core driver
Jan 04 22:09:47 zenbook kernel: input: Razer Razer Naga Hex V2 as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.0/0003:1532:0050.0001/input/input19
Jan 04 22:09:47 zenbook kernel: hid-generic 0003:1532:0050.0001: input,hidraw0: USB HID v1.11 Mouse [Razer Razer Naga Hex V2] on usb-0000:00:14.0-2/input0
Jan 04 22:09:47 zenbook kernel: input: Razer Razer Naga Hex V2 as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.1/0003:1532:0050.0002/input/input20
Jan 04 22:09:47 zenbook kernel: hid-generic 0003:1532:0050.0002: input,hidraw1: USB HID v1.11 Keyboard [Razer Razer Naga Hex V2] on usb-0000:00:14.0-2/input1
Jan 04 22:09:47 zenbook kernel: input: Razer Razer Naga Hex V2 as /devices/pci0000:00/0000:00:14.0/usb3/3-2/3-2:1.2/0003:1532:0050.0003/input/input21
Jan 04 22:09:47 zenbook kernel: hid-generic 0003:1532:0050.0003: input,hidraw2: USB HID v1.11 Keyboard [Razer Razer Naga Hex V2] on usb-0000:00:14.0-2/input2
...
// lol
...
Jan 04 22:10:03 zenbook polychromatic-tray-applet.desktop[1020]: Traceback (most recent call last):
Jan 04 22:10:03 zenbook polychromatic-tray-applet.desktop[1020]:   File "/usr/bin/polychromatic-tray-applet", line 24, in <module>
Jan 04 22:10:03 zenbook polychromatic-tray-applet.desktop[1020]:     gi.require_version('AppIndicator3', '0.1')
Jan 04 22:10:03 zenbook polychromatic-tray-applet.desktop[1020]:   File "/usr/lib/python3.5/site-packages/gi/__init__.py", line 118, in require_version
Jan 04 22:10:03 zenbook polychromatic-tray-applet.desktop[1020]:     raise ValueError('Namespace %s not available' % namespace)
Jan 04 22:10:03 zenbook polychromatic-tray-applet.desktop[1020]: ValueError: Namespace AppIndicator3 not available
Jan 04 22:10:03 zenbook canberra-gtk-pl[1016]: Theme parsing error: gtk.css:1049:69: Using one color stop with linea
...
Jan 04 22:10:03 zenbook razer-service.desktop[1018]: 2017-01-04 22:10:03 | razer                          | INFO     | Initialising Daemon (v1.1.5). Pid: 1018
Jan 04 22:10:03 zenbook razer-service.desktop[1018]: 2017-01-04 22:10:03 | razer.screensaver              | INFO     | Initialising DBus Screensaver Thread
Jan 04 22:10:03 zenbook razer-service.desktop[1018]: 2017-01-04 22:10:03 | razer                          | DEBUG    | Loaded device specification: RazerAbyssus ---------------------------- (1532:0042)
Jan 04 22:10:03 zenbook razer-service.desktop[1018]: 2017-01-04 22:10:03 | razer                          | DEBUG    | Loaded device specification: RazerAnansi ----------------------------- (1532:010F)
Jan 04 22:10:03 zenbook razer-service.desktop[1018]: 2017-01-04 22:10:03 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2013 --------------------- (1532:011A)
Jan 04 22:10:03 zenbook razer-service.desktop[1018]: 2017-01-04 22:10:03 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidow2016 --------------------- (1532:0214)
Jan 04 22:10:03 zenbook razer-service.desktop[1018]: 2017-01-04 22:10:03 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChroma ------------------- (1532:0203)
Jan 04 22:10:03 zenbook razer-service.desktop[1018]: 2017-01-04 22:10:03 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowChromaTournamentEdition -- (1532:0209)
Jan 04 22:10:03 zenbook razer-service.desktop[1018]: 2017-01-04 22:10:03 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChroma ------------------ (1532:0216)
Jan 04 22:10:03 zenbook razer-service.desktop[1018]: 2017-01-04 22:10:03 | razer                          | DEBUG    | Loaded device specification: RazerBlackWidowXChromaTournamentEdition - (1532:021A)
Jan 04 22:10:03 zenbook razer-service.desktop[1018]: 2017-01-04 22:10:03 | razer                          | DEBUG    | Loaded device specification: RazerBladeProLate2016 ------------------- (1532:0210)
Jan 04 22:10:03 zenbook razer-service.desktop[1018]: 2017-01-04 22:10:03 | razer                          | DEBUG    | Loaded device specification: RazerBladeQHD --------------------------- (1532:020F)
Jan 04 22:10:03 zenbook razer-service.desktop[1018]: 2017-01-04 22:10:03 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealth ----------------------- (1532:0205)
Jan 04 22:10:03 zenbook razer-service.desktop[1018]: 2017-01-04 22:10:03 | razer                          | DEBUG    | Loaded device specification: RazerBladeStealthLate2016 --------------- (1532:0220)
[... cut ...]
Jan 04 22:10:03 zenbook razer-service.desktop[1018]: 2017-01-04 22:10:03 | razer                          | INFO     | Serving DBus
...
Jan 04 22:10:05 zenbook razer-service.desktop[1018]: 2017-01-04 22:10:05 | razer.screensaver              | INFO     | Initialising DBus screensaver object
Jan 04 22:10:05 zenbook razer-service.desktop[1018]: 2017-01-04 22:10:05 | razer.screensaver              | WARNING  | Could not find screensaver DBus
Jan 04 22:10:05 zenbook razer-service.desktop[1018]: 2017-01-04 22:10:05 | razer.screensaver              | WARNING  | Giving up on screensaver
...
```

artemjackson, 2017-01-04:
btw,
```
modprobe razerkbd
modprobe: FATAL: Module razerkbd not found in directory /lib/modules/4.8.13-1-ARCH
```

terrycain, 2017-01-04:
That's the issue. @z3ntu, being an arch issue, your turn :wink:

terrycain, 2017-01-04:
did you try sudo btw?

artemjackson, 2017-01-04:
@terrycain sure

artemjackson, 2017-01-04:
@terrycain @z3ntu should you reopen the issue? ;)

terrycain, 2017-01-04:
Nah, this is a seperate issue, its the driver not binding, my naga is working perfectly infront of me :wink:

artemjackson, 2017-01-04:
Lucky you :)

z3ntu, 2017-01-04:
@artemjackson Maybe this'll help:
```
git clone https://aur.archlinux.org/razer-drivers-git.git
cd razer-drivers-git/
makepkg -sr
sudo pacman -U python-razer-git-*.tar* razer-daemon-git-*.tar* razer-driver-arch-git-*.tar*
```
And then reboot & try again.
If you then still have issues, do you have Telegram? If yes, you can contact me @z3ntu , it's easier to help there ;)

artemjackson, 2017-01-05:
Well, I installed with -Udd to skip dependencies check.
After reboot I tried
```
$ systemctl status razerd.service 
Unit razerd.service could not be found.
$ razer-service 
Traceback (most recent call last):
  File "/usr/bin/razer-service", line 7, in <module>
    from razer_daemon.daemon import daemonize
  File "/usr/lib/python3.5/site-packages/razer_daemon/daemon.py", line 17, in <module>
    import setproctitle
ImportError: No module named 'setproctitle'
```

artemjackson, 2017-01-05:
Well, following @z3ntu instruction I got this working 
But additional packages need to be installed: `python-notify2`, `python-numpy`, `python-setproctitle`, `xautomation`

terrycain, 2017-01-05:
They are dependencies of the Ubuntu RPMs ;-) @GabMus Dependencies need updating?

z3ntu, 2017-01-05:
Nah, out of some reason pacman didn't install them automatically (they ARE dependencies)...

terrycain, 2017-01-05:
Ok cool we all good now?

z3ntu, 2017-01-05:
Seems like it